noble/bleno #90: Clarification on BlueZ-5.23

AlessandroFerri, 2015-01-20:
clarification on BlueZ-5:23

I installed bleno with Bluez-5.23. Of course I found the same error shown in issue # 24. Initially the installation of Bluez-5.23 was made along with PulseAudio-5 for A2DP audio streaming.
I'm going around the problem by running
sudo killall bluetoothd
and subsequently
sudo hciconfig hci0 up.
in this way when I run
sudo node test
my iPad is connected correctly, otherwise the connection is refused.
I do bluetoothd run as a daemon on boot, with an init.d scrip.
So I wanted to have some suggestions:
1) The issue # 24 was passed or still exists and I have to install a previous version, for example BlueZ-4.99, as suggested?
2) is it possible that the issue # 24 is linked to how I run the daemon?
3) the service bleno conflicts with A2DP or they can coexist'
4) I need to intercept and integrate the values of the characteristic to run a my code in C, how can I do or how can I use l2cap-ble.c?

sandeepmistry, 2015-01-24:
@AlessandroFerri everything in #24 is still relevant to BlueZ 5.23.

I would recommend you either:
1) Run BlueZ 4.101 - unless you absolutely need 5.x
2) Run a patched version of BlueZ 5.x bluetoothd, see #87 - remove call to ``btd_adapter_gatt_server_start `` in  [``src/adapter.c``](http://git.kernel.org/cgit/bluetooth/bluez.git/tree/src/adapter.c#n6593)

closed 2015-01-24 by sandeepmistry

AlessandroFerri, 2015-01-25:
Hi Sandeep Mistry,
I have no clear some of your suggestions. you I can confirm the following points?

1) If I install BlueZ 4.101 the conflict with A2DP, which I have described in issue #91 disappears?
2) have I to remove the call btd_adapter_gatt_server_start in src/adapter.c also in the release 4.101?
3) the issue #87 speaks of the example bleno/examples/pizza, what it would add to my issue?

sandeepmistry, 2015-02-03:
@AlessandroFerri 

You have to do either 1 or 2 from my previous comment. I mean't to refer to #56 instead for option 2.