noble/bleno #92: Support for writer/reader pipe file and description of service and characteristic

AlessandroFerri, 2015-01-28:
Hi Sandeep Mistry,
I tried the example /bleno/examples/battery-service and /bleno/test.js. In the first instance, from iOs BLE, I get the description of the service as "Service Battery" and of the characteristic as "Battery Level", while in the second example the description of the service and of the characteristics appear always as "Unknown". I have tried in many ways to change the description "unknow", but did not succeed. Please can you help me?
Also I need to use the data for a WriteOnlyCharacteristic from another program in C code. I have thought of a writer/reader pipe socket file and I modified /beno/test.js in this manner.

var options = { flags: 'w', encoding: 'hex', mode: 0666 };
var wstream = fs.createWriteStream('/tmp/msgfifo', options);

WriteOnlyCharacteristicXXX.prototype.onWriteRequest = function(data, offset, withoutResponse, callback) {
  console.log('WriteOnlyCharacteristicXXX write request: XXX' + data.toString('hex') + ' ' + offset + ' ' + withoutResponse);

  wstream.write('XXX'+data.toString('hex'));

  callback(this.RESULT_SUCCESS);
};

This works, but if the file /tmp/msgfifo is created first by test.js, rather than from the program in C code, the C program goes in loop.
is the right path for you? Can you give me some advice?

sandeepmistry, 2015-02-03:
@AlessandroFerri 

Where is unknown being shown? What is the second example?

Please keep one question/issue per Github issue and specific to bleno - there are better places to ask general node.js + C questions like the one you are asking about files/streams.

sandeepmistry, 2015-02-14:
Closing due to inactivity.

closed 2015-02-14 by sandeepmistry