noble/bleno #93: Clean shutdown

hardillb, 2015-01-29:
Any chance of adding a function to cleanly shut everything down? So I can tear down all the bleno stuff without having to shut the whole node app down.

Thanks

sandeepmistry, 2015-01-30:
@hardillb what's your goal for this? Stop advertising, change services, etc?

hardillb, 2015-01-30:
@sandeepmistry I was thinking of doing another Node-RED node, so the entire capability needs to be able to be removed from the "app" when the node is removed from the flow.

But having thought about it a bit more I think it may be a bit too complicated to build a configuration system to dynamically build services/categories on the fly. It could be done, but I think the UI would just be too messy.

sandeepmistry, 2015-01-30:
@hardillb does sound interesting, but agree UI would be complicated.

Can this issue be closed for now?

hardillb, 2015-01-30:
yeah, we can close this for now, I might have a go at this anyway. If I get something workable I'll come back and we can look at a clean shutdown path

closed 2015-01-30 by hardillb