noble/bleno #95: Not getting any services when connecting on Linux

hardillb, 2015-02-06:
I've been trying to write just a simple HelloWorld iBeacon app to run on my Raspberry Pi and I'm having some issues.

My current code is here: https://gist.github.com/hardillb/9a8b76c325e063b39e45

It seams to advertise fine, but when I connect with the gatttool from my laptop I get no services or characteristics back:

    root@rpi:~# hcitool lescan
    LE Scan ...
    00:15:83:6B:D1:50 (unknown)
    00:15:83:6B:D1:50 pi
    00:15:83:6B:D1:50 (unknown)
    00:15:83:6B:D1:50 pi
    root@rpi:~# gatttool -b 00:15:83:6B:D1:50 -I
    [00:15:83:6B:D1:50][LE]> connect
    Attempting to connect to 00:15:83:6B:D1:50
    Connection successful
    [00:15:83:6B:D1:50][LE]> primary
    [00:15:83:6B:D1:50][LE]>

I'm running with bluez 5.27 and bluetoothd is not running.

Any suggestions?

sandeepmistry, 2015-02-08:
@hardillb which kernel version are you using?

There were issues with 3.18.x, which I've addressed with ff09afe6785a92becf36632bf6f9539aefb7e37a. Try out master, I'll be publishing a new version to npm over the next few days.

hardillb, 2015-02-08:
Thanks, I'm running with 3.18.5 at the moment. I'll give the head stream a go.

sandeepmistry, 2015-02-08:
@hardillb new version of bleno 0.1.10 is in npm now.

Could you also please try with BlueZ 5.x bluetoothd running/started before running your bleno app? They seemed to co-exist on my RPi, but on my Intel Edison bluetoothd's built-in GATT server took precedence over bleno.

sidwarkd, 2015-02-09:
I was experiencing the same issue. Just did an rpi-update which put me on kernel version 3.18.6 and installed 0.1.10 of bleno and the test.js app worked great. My version of BlueZ is 4.99.

sandeepmistry, 2015-02-10:
@sidwarkd thanks for trying it out!

@hardillb I'll close this issue know, please re-open if you experience any issues with the newer version.

closed 2015-02-10 by sandeepmistry