noble/bleno #96: Installing Bleno on Edison fails

danki, 2015-02-16:
Had to re-install everything on my Intel edison but now for some reason im not able to install bleno at all:

[image]

I have tested everything, reinstalling Edison multiple times, re-configuring it but it just dont work any more.

sandeepmistry, 2015-02-16:
It's buried in the screenshot, but have you looked at:
[image]

https://github.com/sandeepmistry/noble#prerequisites

closed 2015-02-16 by sandeepmistry

danki, 2015-02-18:
That was it, thank you ! 
Sorry for that, i thought i had installed bluez, thats why i was confused :)

sandeepmistry, 2015-02-19:
No worries, glad that helped ...