noble/bleno #97: difficulties in creating two primary services

AlessandroFerri, 2015-02-23:
Hi Sandeep Mistry,
how I can create two primary service with two different uuid and different list of characteristics. For example, starting from the example test.js, how can I create another primary service with a different uuid uuid: 'fffffffffffffffffffffffffffffff0' and a list of other characteristics?

zdennis, 2015-02-23:
@AlessandroFerri, take a look at the battery level service example: https://github.com/sandeepmistry/bleno/tree/master/examples/battery-service

That shows an example of setting up a service and a characteristic. You should be able to take what is in that example to add another service and another characteristic.

I'm not sure this needs to be an open issue unless you're requesting that a new example that covers multiple services/characteristics is added to the `examples/` directory. If that's the case, I'm sure @sandeepmistry would love it receive a pull request for that, but I won't presume to know what he wants, because another example means another thing for him to maintain.

AlessandroFerri, 2015-02-23:
Hi Zach Dennis,
I thank you for the answer. I have tried in many ways to modify the example battery-service, but while it was easy to modify and add the characteristics, I was not able to add other services.
I have tried in these ways as follow, but I did not find any results:
First 
function BatteryService() {
  BatteryService.super_.call(this, {
      uuid: '180F',
      characteristics: [
          new BatteryLevelCharacteristic()
      ]
  });
}

util.inherits(BatteryService, BlenoPrimaryService);

function BatteryService2() {
  BatteryService2.super_.call(this, {
      uuid: '180E',
      characteristics: [
          new BatteryLevelCharacteristic()
      ]
  });
}

util.inherits(BatteryService2, BlenoPrimaryService);

and second
function BatteryService() {
  BatteryService.super_.call(this, {
      uuid: '180F',
      characteristics: [
          new BatteryLevelCharacteristic()
      ]
      },{
      uuid: '180E',
      characteristics: [
          new BatteryLevelCharacteristic()
      ]
  });
}

util.inherits(BatteryService, BlenoPrimaryService);

when I run battery-service I have no error but I have only one service. where I'm wrong?

sandeepmistry, 2015-03-01:
@zdennis thanks for helping out! Agree with your comment, if things are unclear in the read me/documentation/examples they should be clarified. That said I think it's pretty straightforward for most people to get multiple services running

@AlessandroFerri more details are required ...

 * What OS are you running bleno on?
 * What are you using as the central (OS, device, app)?
 * Which version of bleno are you using?
 * Could you please share all the code you are running (including unmodified files) as a Github Gist or in a Github repo.
 * Please provide debug output: ```sudo DEBUG=* node <file>.js```

AlessandroFerri, 2015-03-05:
Sandeep Mistry Thanks for the help,
after several attempts I managed to create two primary service. Everything works well and runs. Anyway I will carry information and the code that you have asked:

1) I use my iPad with iOS 8.1.3 Air + LightBlue
2) the version of bleno is the latest downloadable with sudo npm install bleno
3) you find the code on link: https://github.com/AlessandroFerri/Prova/blob/master/Test_TwoSevice.js

As you can see, I have started from test.js. At this point, I would like to know from you, if you have comments to make on the code in general and in particular on how I developed the part of write fifo file, which you can find on characteristics.

sandeepmistry, 2015-03-05:
@AlessandroFerri great to hear you got things working.

Code looks fine.

Is there any changes that code have been made to the read me, that would have clarified how to setup two or more services?

closed 2015-03-05 by sandeepmistry

AlessandroFerri, 2015-03-06:
Hi Sandeep Mistry,
The read me is irrelevant, because it was inserted right to open the repository where I put in the code
The part of the code that reads from the fifo file is linked here:
https://github.com/AlessandroFerri/Prova/blob/master/Read_FifoFile.c
The only issue I have is that if I do run the program Test_TwoService.j before Read_FifoFile.c, the code Read_FifoFile.c enters into a loop. This is why I put the control:

if (readlenbuf == 0) {;
printf ("Read error ... \ n");
sleep (1);
}

Have you any suggestions?

thanks in advance.

sandeepmistry, 2015-03-07:
@AlessandroFerri as mentioned in #92, please leave issue discussion here focused on bleno/BLE.

There are better places like [Stack Overflow](http://stackoverflow.com) for post questions on how to interface C with node.js