noble/bleno #99: Device information service is override by the MAC's data on Yosemite

kapusil, 2015-03-26:
I updated my mac mini to yosemite and recognised that instead of the device information service I configure with bleno, I see the mac's data (e.g. manfufacturer name string characteristics have the value 'Apple inc.' instead of "abc ltd" set by me).

sandeepmistry, 2015-03-28:
@kapusil OS X 10.10 has some built-in services/characteristic for device information etc. now, which you cannot override.

I could make changes to bleno to report a failure when you try to use them.

kapusil, 2015-03-30:
@sandeepmistry 
Thanks for the feedback. I think the error message is a good idea if there is no way to overwrite them.

sandeepmistry, 2015-03-31:
@kapusil no problem. Commit https://github.com/sandeepmistry/bleno/commit/f64c805ee13aeec955e2c418e21f593615bf1c54 will report errors for UUID's which are not allowed.

closed 2015-03-31 by sandeepmistry