noble/bleno #100: Unplugging usb device results in hci-ble 100% cpu

chmorgan, 2015-03-30:
in hci-ble.c a select is performed on both file descriptors, stdin (0) and the hciSocket:

    selectRetval = select(hciSocket + 1, &rfds, NULL, NULL, &tv);

But the hciSocket isn't being checked.

When the bluetooth device is unplugged hciSocket is active, so the select() wakes up, but nothing in the main loop is processing data from hciSocket.

It doesn't look like anything in hci-ble.c is reading from the hci device, just writing to it.

Thoughts on how to detect when the device goes away? Should we re-issue the hci_xxx() calls when the hciSocket is reported via select()?

sandeepmistry, 2015-04-05:
@chmorgan is there a particular use case around unplugging the adapter or is this something you noticed?

The select and not reading is probably a remanent of porting from noble. However, events are triggered when the adapter changes states (powered on/off). ```ioctl(hciSocket, HCIGETDEVINFO, (void *)&hciDevInfo);``` is used to get that state.

Maybe the ``ioctl`` return value + ``errno`` could be used to determine when the adapter.

chmorgan, 2015-04-05:
Noticed the issue when unplugging the bluetooth module and started looking into how it might be fixed. Let me try checking the return value of the ioctl tomorrow. I'll send a PR if it pans out.

I'm also thinking about removing the hciSocket from the list of descriptors being selected if the code isn't going to process that descriptor anyway.

sandeepmistry, 2015-04-26:
@chmorgan any progress on this?

chmorgan, 2015-04-26:
Hey man.

Haven't had a chance to go back to look at it but using that ioctl looks like it has potential to resolve the issue. I'll probably get back to it in May sometime.

sandeepmistry, 2015-04-27:
Cool, thanks for the update!

sandeepmistry, 2015-08-10:
This issue no longer applies to bleno, ```hci-ble``` has been removed in master.
[node-bluetooth-hci-socket](https://github.com/sandeepmistry/node-bluetooth-hci-socket) is now used for the HCI layer.

closed 2015-08-10 by sandeepmistry