noble/bleno #103: example for communicate with iOS

fishkingsin, 2015-04-21:
anyone have such experience?
thanks

MylesCaley, 2015-04-21:
> <reply+00562f5c21877bec2486ee3bfb804e089f469b6042fa8b2092cf00000001114e2c6b9
> 2a169ce042a31a7@reply.github.com>

Its part of a bigger iOS app, but here is a view controller that uses the
basic BLE stuff in iOS.

https://github.com/FirstBuild/FirstBuild-Mobile/blob/master/MobiUs/ParagonEn
gineeringViewController.m

Talking to this hack3ed up device (running on a pi):

https://github.com/FirstBuild/SousVideBlueberryPi/blob/master/index.js

Its very sub-par but was something I needed to get running rather quickly.
Maybe there is enough there to get you started.

<https://github.com/sandeepmistry/bleno/issues/103> .

sandeepmistry, 2015-04-21:
@MylesCaley thanks!

@fishkingsin could you clarify your question - are you look for help on the iOS or bleno side?

fishkingsin, 2015-04-22:
@sandeepmistry thanks for the reply ,
I am try to catch the bleno advertising with some iOS BLE utilities , and then read the service and characteristic as usual 
bleno  can be scanned  by iOS app , however can not read the rest.
I am try to do the test as @MylesCaley suggested, and see how it works

fishkingsin, 2015-04-22:
thanks dudes,
I have successfully made it work with my Mac book Pro MacBook Pro (Retina, 15-inch, Mid 2014)
with 
https://github.com/OpenBLE/OpenBLE
by change the LE-UUIDs.plist 
let me clean the code and share it later on

sandeepmistry, 2015-04-22:
@fishkingsin no problem

closed 2015-04-22 by sandeepmistry

fishkingsin, 2015-04-23:
for other guys who are very beginner 

https://github.com/fishkingsin/Pizza