noble/bleno #104: NOT returning string data from a custom characteristic if offset is greater than or equal 157

AlessandroFerri, 2015-04-22:
Hello,
and thank you again for the work you have done, Bleno is fantastic.
I have a problem with the buffer of a custom characteristic. My problem is similar to the issue #86. I need to read from LightBlue a very long string, even more than 256 characters. Here is the code:

CharacteristicWIFI.prototype.onReadRequest = function(offset, callback) {
	if (offset) {
		console.log('RESULT_INVALID_OFFSET>'+offset+'<');
		callback(this.RESULT_INVALID_OFFSET, null);
	}
	else {
		var data = new Buffer(sendmesgBLE.length);
		data.write(sendmesgBLE, 0, sendmesgBLE.length);
		console.log('RESULT_SUCCESS_OFFSET>'+offset+'<');
		console.log('data.length>'+data.length+'<data>\n'+data+'<');
		callback(this.RESULT_SUCCESS, data);
	}
};

I am using the LightBlue iOS app to read the characteristics.
If the string has a length less than 156 characters, the result I get is the following:

Bleno
on -> stateChange: poweredOn
on -> advertisingStart: success
on -> servicesSet
on -> accept, client: 46:44:35:4A:76:69
on -> rssiUpdate: 127
RESULT_SUCCESS_OFFSET>0<
data.lenght>156<data>
0	on	Network di Alessandro & Daniela
1	on	Network di Alessandro & Daniela
2	on	Network ospiti di A & D
3	on	Network ospiti di A & D
4	on	Vodafone-3
6	on	123<

I have no issue, and I read data from LightBlue iOS app. But if the string of data is 157 characters or greater returns RESULT_INVALID_OFFSET

Bleno
on -> stateChange: poweredOn
on -> advertisingStart: success
on -> servicesSet
on -> accept, client: 76:01:C8:D8:3A:CE
on -> rssiUpdate: 127
RESULT_SUCCESS_OFFSET>0<
data.length>157<data>
0	on	Network di Alessandro & Daniela
1	on	Network di Alessandro & Daniela
2	on	Network ospiti di A & D
3	on	Network ospiti di A & D
4	on	Vodafone-3
6	on	1234<
RESULT_INVALID_OFFSET>157<

and I NOT I read data from LightBlue iOS app
I do not know if the problem is how to prepare data for BLE or is different.

sandeepmistry, 2015-04-26:
@AlessandroFerri you shouldn't be returning invalid offset if there is one, as your characteristic is "long".
iOS currently negotiates an MTU of 158 bytes, so for data lengths longer than that, it will call read multiple times with an offset.

I've updated test.js with a "long dynamic read only" characteristic: https://github.com/sandeepmistry/bleno/commit/9fc7b0c97ff05f518399adb4aeffe8a47cea7a7e

closed 2015-04-26 by sandeepmistry

AlessandroFerri, 2015-04-28:
thanks for the information and updates. I wanted to ask you further information. On LightBlue, raspberry appears with the same name as the hostname. How can I change it and where do I change it? You keep in mind that I'm using your test.js

sandeepmistry, 2015-04-28:
@AlessandroFerri I'd appreciate it if you keep issues focused on one topic.

I'd be happy to discuss your question on gitter: https://gitter.im/sandeepmistry/noble