noble/bleno #105: Bleno peripheral not being detected

jagreenwood, 2015-04-22:
I'm having trouble finding a Bleno peripheral. Centrals I'm using atm are LightBlue for iOS and the Noble module.  Both centrals detect other BLE devices but nothing from Bleno,  For discussions sake I logged debug info from the battery-service example.  Any insight in what I'm doing wrong would be awesome, thanks!

```
helios-2:battery-service jeremy$ DEBUG=* node main
  bindings sendCBMsg: 1, {
  "kCBMsgArgName": "node-1429737123064",
  "kCBMsgArgOptions": {
    "kCBInitOptionShowPowerAlert": 1
  },
  "kCBMsgArgType": 1
} +0ms
  bindings xpcEvent: {
  "kCBMsgId": 6,
  "kCBMsgArgs": {
    "kCBMsgArgState": 5
  }
} +12ms
  bindings state change poweredOn +1ms
  bleno stateChange poweredOn +0ms
on -> stateChange: poweredOn
  bindings sendCBMsg: 8, {
  "kCBAdvDataLocalName": "Battery",
  "kCBAdvDataServiceUUIDs": [
    {
      "type": "Buffer",
      "data": [
        24,
        15
      ]
    }
  ]
} +2ms
  bindings xpcEvent: {
  "kCBMsgId": 16,
  "kCBMsgArgs": {
    "kCBMsgArgResult": 0
  }
} +1ms
  bleno advertisingStart: null +3ms
on -> advertisingStart: success
  bindings sendCBMsg: 12, null +0ms
  bindings sendCBMsg: 10, {
  "kCBMsgArgAttributeID": 1,
  "kCBMsgArgAttributeIDs": [],
  "kCBMsgArgCharacteristics": [
    {
      "kCBMsgArgAttributeID": 2,
      "kCBMsgArgAttributePermissions": 1,
      "kCBMsgArgCharacteristicProperties": 2,
      "kCBMsgArgData": null,
      "kCBMsgArgDescriptors": [
        {
          "kCBMsgArgData": "Battery level between 0 and 100 percent",
          "kCBMsgArgUUID": {
            "type": "Buffer",
            "data": [
              41,
              1
            ]
          }
        },
        {
          "kCBMsgArgData": {
            "type": "Buffer",
            "data": [
              4,
              1,
              39,
              173,
              1,
              0,
              0
            ]
          },
          "kCBMsgArgUUID": {
            "type": "Buffer",
            "data": [
              41,
              4
            ]
          }
        }
      ],
      "kCBMsgArgUUID": {
        "type": "Buffer",
        "data": [
          42,
          25
        ]
      }
    }
  ],
  "kCBMsgArgType": 1,
  "kCBMsgArgUUID": {
    "type": "Buffer",
    "data": [
      24,
      15
    ]
  }
} +1ms
  bindings xpcEvent: {
  "kCBMsgId": 18,
  "kCBMsgArgs": {
    "kCBMsgArgAttributeID": 1,
    "kCBMsgArgResult": 27
  }
} +0ms
  bleno servicesSet +1ms
```

sandeepmistry, 2015-04-22:
@jagreenwood is this on OS X 10.9 or 10.10?

Is there anything connected? Click on Bluetooth menu bar item to see connected devices ....

jagreenwood, 2015-04-22:
I'm running 10.10.3, no other devices are connected.

jagreenwood, 2015-04-23:
Turns out there isn't an issue.  Noble on an intel edison detects bleno on OSX just fine.  I was expecting LightBlue to show a name of `Battery` but instead the battery service was getting buried in the advertisement data section under my machine name.
Noble still doesn't detect the peripheral when it is being run on the same machine, but maybe that makes sense if there is a conflict in the modes?

closed 2015-04-23 by jagreenwood

sandeepmistry, 2015-04-23:
@jagreenwood yes, same running noble and bleno on the same machine is not supported. You cannot advertise and scan for your own peripheral unless you have more than one Bluetooth Adapter.