noble/bleno #107: node-gyp build fails for node-gyp version 2.0.1 on OS X

pfafman, 2015-06-12:
*This might be a node-gyp issue* 

node-gyp@2.0.1 is trying to compile the C sources meant for linux.

I do a clean git clone then

```
$ node-gyp -v
v2.0.1

$ node-gyp clean
...

$ node-gyp configure
...

$ node-gyp build
gyp info it worked if it ends with ok
gyp info using node-gyp@2.0.1
gyp info using node@0.12.4 | darwin | x64
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
  LINK(target) Release/hci-ble
Undefined symbols for architecture x86_64:
  "_main", referenced from:
      start in crt1.10.5.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [Release/hci-ble] Error 1
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/node-gyp/lib/build.js:269:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1074:12)
gyp ERR! System Darwin 14.3.0
gyp ERR! command "node" "/usr/local/bin/node-gyp" "build"
gyp ERR! cwd /Users/tep/Projects/Bluetooth/bleno
gyp ERR! node -v v0.12.4
gyp ERR! node-gyp -v v2.0.1
gyp ERR! not ok
```

If I downgrade back to node-gyp@1.0.3 the build succeeds.

This also means the `npm install bleno` fails for node v0.12.4/ npm v2.11.2 which is using node-gyp v2.0.1 

```
$ npm -v
2.11.2

$ npm install bleno
\
> xpc-connection@0.1.3 install /Users/tep/Projects/Bluetooth/fakeTag/node_modules/bleno/node_modules/xpc-connection
> node-gyp rebuild

  CXX(target) Release/obj.target/binding/src/XpcConnection.o
  SOLINK_MODULE(target) Release/binding.node

> bleno@0.1.13 install /Users/tep/Projects/Bluetooth/fakeTag/node_modules/bleno
> node-gyp rebuild

  LINK(target) Release/hci-ble
Undefined symbols for architecture x86_64:
  "_main", referenced from:
      start in crt1.10.5.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [Release/hci-ble] Error 1
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:269:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1074:12)
gyp ERR! System Darwin 14.3.0
gyp ERR! command "node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/tep/Projects/Bluetooth/fakeTag/node_modules/bleno
gyp ERR! node -v v0.12.4
gyp ERR! node-gyp -v v2.0.1
gyp ERR! not ok
npm ERR! Darwin 14.3.0
npm ERR! argv "node" "/usr/local/bin/npm" "install" "bleno"
npm ERR! node v0.12.4
npm ERR! npm  v2.11.2
npm ERR! code ELIFECYCLE

npm ERR! bleno@0.1.13 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the bleno@0.1.13 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the bleno package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls bleno
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /Users/tep/Projects/Bluetooth/fakeTag/npm-debug.log
```

sandeepmistry, 2015-06-24:
@pfafman this is very similar to https://github.com/sandeepmistry/noble/issues/200

Would you have a chance to try out what I suggested in: https://github.com/sandeepmistry/noble/issues/200#issuecomment-115027744 ?

peterdemartini, 2015-06-24:
I found a temporary solution since I was getting the same error. If I downgrade to version 2.10.1 of npm it works.

````
npm i npm@2.10.1 -g
`````

jpv, 2015-06-30:
npm i npm@2.10.1 -g works for me too. Thanks.

pfafman, 2015-07-06:
Sorry for the delay.  Just tried the suggested fix from noble#200 and that looks like it works.

sandeepmistry, 2015-07-07:
Fixed via #111.

closed 2015-07-07 by sandeepmistry