noble/bleno #108: Can NOT use "npm install bleno" in WRTnode, which is a kind of openwrt system.

kaiser-ren, 2015-06-20:
my platform is WRTnnode, a kind of openwrt syste. www.wrtnode.com
I had installed nodejs.
Then when I use npm install bleno, the installer reported an error:
npm ERR! bleno@0.1.13 install: `node-gyp rebuild`
npm ERR! Exit status 137
npm ERR!
npm ERR! Failed at the bleno@0.1.13 install script.
npm ERR! This is most likely a problem with the bleno package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls bleno
npm ERR! There is likely additional logging output above.
npm ERR! System Linux 3.10.49
npm ERR! command "node" "/usr/bin/npm" "install" "bleno"
npm ERR! cwd /tmp
npm ERR! node -v v0.10.33
npm ERR! npm -v 1.4.28
npm ERR! code ELIFECYCLE

what's wrong with that?

sandeepmistry, 2015-06-21:
Can you please share the npm debuf log? Also, do you have the libbluetooth headers installed?

kaiser-ren, 2015-06-21:
Hi Weekly Status Report--20150615

kaiser-ren, 2015-06-21:
I have tried to install "libbluetooth-dev", but it's unavailable in my platform, WRTnode. I can install "libbluetooth-dev" in Raspberry Pi.
As per npm debug log, how can I get it from my board?

sandeepmistry, 2015-06-21:
You'll have to find the equivalent to install.

You could just use ```cat npm-debug.log``` and copy/paste the output from the console.

kaiser-ren, 2015-06-23:
I can use "npm list" in /root to list the install packages, I found bleno was installed in my WRTnode.
-bleno@0.1.13
--bplist-parser@0.0.6
--debug@0.7.4

kaiser-ren, 2015-06-23:
After that, I copy example/battery_service into my WRTnode, and run "npm install" in folder "battery_service", but below message pop-up.
npm WARN package.json battery-service@0.1.0 No repository field.
npm WARN optional dep failed, continuing xpc-connection@0.1.3

> bleno@0.1.13 install /kai/battery-service/node_modules/bleno
> node-gyp rebuild

#
# Fatal error in ../deps/v8/src/incremental-marking.cc, line 458
# CHECK(success) failed
#

==== Stack trace ============================================

Security context: 0x35f09619 <JS Object>#0#
    1: emit [events.js:53] (this=0x2b1f7d71 <a BlockStream>#1#,type=0x2b70bf6d <String[4]: data>)
    2: arguments adaptor frame: 2->1
    3: _emitChunk [/usr/lib/node_modules/npm/node_modules/block-stream/block-stream.js:145] (this=0x2b1f7d71 <a BlockStream>#1#,flush=0x35f08091 <undefined>)
    4: arguments adaptor frame: 0->1
    5: resume [/usr/lib/node_modules/npm/node_modules/block-stream/block-stream.js:58] (this=0x2b1f7d71 <a BlockStream>#1#)
    6: arguments adaptor frame: 1->0
    7: resume [/usr/lib/node_modules/npm/node_modules/fstream/lib/reader.js:255] (this=0x2b1f7dc1 <an Extract>#2#,who=0x2b1f7dc1 <an Extract>#2#)
    8: arguments adaptor frame: 0->1
    9: ondrain [stream.js:61] (this=0x2b1a6be1 <a DirWriter>#3#)
   10: emit [events.js:92] (this=0x2b1a6be1 <a DirWriter>#3#,type=0x3f4a5455 <String[5]: drain>)
   11: _process [/usr/lib/node_modules/npm/node_modules/fstream/lib/dir-writer.js:88] (this=0x2b1a6be1 <a DirWriter>#3#)
   12: onend [/usr/lib/node_modules/npm/node_modules/fstream/lib/dir-writer.js:169] (this=0x3470819d <a FileWriter>#4#)
   13: emit [events.js:92] (this=0x3470819d <a FileWriter>#4#,type=0x2b7193b9 <String[5]: close>)
   14: statCb [/usr/lib/node_modules/npm/node_modules/fstream/lib/writer.js:135] (this=0x35f096c1 <JS Global Object>#5#,er=0x34708245 <an Error>#6#,current=0x35f08091 <undefined>)
   15: arguments adaptor frame: 1->2
   17: oncomplete [evalmachine.<anonymous>:107] (this=0x347082a5 <an Object>#7#)
   18: arguments adaptor frame: 1->0

==== Details ================================================

[1]: emit [events.js:53] (this=0x2b1f7d71 <a BlockStream>#1#,type=0x2b70bf6d <String[4]: data>) {
  // stack-allocated locals
  var arguments = 0x35f08091 <undefined>
  var er = 0x35f08091 <undefined>
  var handler = 0x35f08091 <undefined>
  var len = 0x35f08091 <undefined>
  var args = 0x35f08091 <undefined>
  var i = 0x35f08091 <undefined>
  var listeners = 0x35f08091 <undefined>
  // heap-allocated locals
  var type = 0x2b70bf6d <String[4]: data>
  // expression stack (top to bottom)
  [10] : 2
  [09] : 1069774640
  [08] : 0x35f24431 <JS Function emit>#8#
  [07] : 0x35f08091 <undefined>
--------- s o u r c e   c o d e ---------
function emit(type) {?  var er, handler, len, args, i, listeners;??  if (!this._events)?    this._events = {};??  // If there is no 'error' event listener then throw.?  if (type === 'error') {?    if (!this._events.error ||?        (typeof this._events.error === 'object' &&?         !this._events.error.length)) ...

-----------------------------------------
}

[2]: arguments adaptor frame: 2->1 {
  // actual arguments
  [00] : 0x2b70bf6d <String[4]: data>
  [01] : 0x3470809d <a Buffer>#9#  // not passed to callee
}

[3]: _emitChunk [/usr/lib/node_modules/npm/node_modules/block-stream/block-stream.js:145] (this=0x2b1f7d71 <a BlockStream>#1#,flush=0x35f08091 <undefined>) {
  // stack-allocated locals
  var padBytes = 0x35f08091 <undefined>
  var bufferIndex = 4
  var out = 0x3470809d <a Buffer>#9#
  var outOffset = 0
  var outHas = 0
[... 540 lines cut ...]
#94# 0x47ab1685: 0x47ab1685 <JS Function>
#95# 0x50a6e959: 0x50a6e959 <JS Function onclose>
#96# 0x3473457d: 0x3473457d <JS Function>
=====================

Trace/breakpoint trap

npm ERR! bleno@0.1.13 install: `node-gyp rebuild`
npm ERR! Exit status 133
npm ERR!
npm ERR! Failed at the bleno@0.1.13 install script.
npm ERR! This is most likely a problem with the bleno package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls bleno
npm ERR! There is likely additional logging output above.
npm ERR! System Linux 3.10.49
npm ERR! command "node" "/usr/bin/npm" "install"
npm ERR! cwd /kai/battery-service
npm ERR! node -v v0.10.33
npm ERR! npm -v 1.4.28
npm ERR! code ELIFECYCLE
npm ERR! not ok code 0

sandeepmistry, 2015-06-24:
@kaiser-ren this is starting to become out of scope of bleno. Your either ```fstream`` module`,```node-gyp``` or ```npm``` is very broken on your system. I suggest you raise an issue with one of those ...

closed 2015-06-24 by sandeepmistry

soonsebii, 2016-08-16:
@kaiser-ren You can use Noble on a WRTnode. 
https://github.com/soonsebii/custom-package/tree/master/lang/noble