noble/bleno #109: Problem with Mac

MeMyselfI, 2015-06-27:
I get this exception on a Mac. I tried to install everything with and without -g flag but it didn't help me. I use bleno und noble at the same time in the program - is this the reason for it?

module.js:340
    throw err;
          ^
Error: Cannot find module 'xpc-connection'
    at Function.Module._resolveFilename (module.js:338:15)
    at Function.Module._load (module.js:280:25)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at Object.<anonymous> (/Users/joern/praktikum/ss15/trunk/rpi/code/bluetooth/rpi-receive/node_modules/bleno/lib/mac/bindings.js:10:21)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)

Although I tried to install it with

sandeepmistry, 2015-07-05:
@MeMyselfI can you please provide the console output when you run: ```npm install xpc-connection```

MeMyselfI, 2015-07-05:
> xpc-connection@0.1.3 install /Users/joern/Desktop/tTIME-5.1.1/node_modules/xpc-connection
> node-gyp rebuild

  CXX(target) Release/obj.target/binding/src/XpcConnection.o
  SOLINK_MODULE(target) Release/binding.node
  SOLINK_MODULE(target) Release/binding.node: Finished
xpc-connection@0.1.3 node_modules/xpc-connection
└── nan@1.8.4

I helped me by changing the references of XpcConnection to the full path at several locations.

sandeepmistry, 2015-07-05:
@MeMyselfI do things work now when you install bleno from npm?

> I helped me by changing the references of XpcConnection to the full path at several locations.

Could you please explain this a little more?

MeMyselfI, 2015-07-05:
For instance in the top error message, I changed

var XpcConnection = require('xpc-connection');

to 

var XpcConnection = require('/...full.path.here...../node_modules/bleno/node_modules/xpc-connection');

sandeepmistry, 2015-07-05:
@MeMyselfI what version of node are you using? Your folder structure looks ok to me, for relative requires to work.

This is looking more like a environment issue than an issue with bleno itself ...

thierrybucco, 2015-07-06:
I got this on OS X when uuid does not contain -

sandeepmistry, 2015-07-11:
@MeMyselfI any updates?

sandeepmistry, 2015-08-02:
Closing due to lack of activity ...

closed 2015-08-02 by sandeepmistry

jesshas, 2018-10-10:
When i try running npm install xpc-connection i get Error: EACCES: permission denied, mkdir ...even though i use sudo.. any help on this?

tomasero, 2018-11-16:
Any updates on this?

tpuza, 2019-02-15:
My index.js file is just that simple:
```javascript
var noble = require('noble');
noble.startScanning();
```
After running this I have same issue with the *`Error: Cannot find module 'xpc-connection'`*

When I do `npm install xpc-connection`, the important part of the log is this:

```
../src/XpcConnection.cpp:143:89: error: too few arguments to function call, expected 2, have 1
      Local<Value> propertyValue = object->GetRealNamedProperty(propertyName->ToString());
```
MacOS: 10.14.3
Node: 10.15.1 (LTE)
node-gyp: 3.8.0
Python: 2.7.15

andreas-bergstrom, 2021-04-01:
Getting this on Node 15.11, Macbook M1, Mac Big Sur:

module.js:550
    throw err;
    ^

Error: Cannot find module 'xpc-connection'
    at Function.Module._resolveFilename (module.js:548:15)
    at Function.Module._load (module.js:475:25)
    at Module.require (module.js:597:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/Users/andreasbergstrom/Projekt/test/js/node_modules/bleno/lib/mac/highsierra.js:10:21)
    at Module._compile (module.js:653:30)
    at Object.Module._extensions..js (module.js:664:10)
    at Module.load (module.js:566:32)
    at tryModuleLoad (module.js:506:12)
    at Function.Module._load (module.js:498:3)

lockieluke, 2022-07-19:
no i don't think i ended up writing a separate native extension to it in Swift

smohanty92, 2022-07-19:
Thanks. Fyi, the https://github.com/notjosh/bleno-mac fork works for Mac!