noble/bleno #110: How to set OOB (Out of Band) Pairing on Raspberry Pi?

shreyashirday, 2015-06-30:
Hi,

Great library! I'm using bleno on a Raspberry Pi for testing purposes and using an Android phone as the central device. How would I go about setting OOB functionality on the Pi? Or must OOB data only be set on the Android phone?

Thanks!

sandeepmistry, 2015-07-05:
@shreyashirday good question! Unfortunately, I don't have an answer.

You could look into how to do this with BlueZ/bluetoothd. Please let me know if you this working.

ostrichjones, 2015-08-13:
This is exactly the issue I am looking for an answer for. Any progress?

shreyashirday, 2015-08-13:
No luck :/ I think the best thing to do is buy a 4.2 qualified chip/module and read the manafacturer SDK for it

ostrichjones, 2015-08-13:
I want to use an Android phone to control a BLE peripheral running on a Linux SBC (RPi). I don't really want anyone to be able to connect and read/write attributes but I also don't want to create a roll-your-own security solution. Looks like I'll have to ...

shreyashirday, 2015-08-13:
@ostrichjones what do you mean by you don't want "anyone" to connect? I think you might be able to control who connects based on mac address. i.e. accept connection if they are in whitelist, otherwise don't

ostrichjones, 2015-08-13:
I'm making a gate opener so I want to control which devices can open it. The MAC address idea sounds fine. It does fall into the roll-your-own category - can I be sure it can't be spoofed? I'm not too concerned but like the idea of at least finding out how to do security properly.

zdennis, 2015-08-14:
A MAC address can be easily spoofed.

shreyashirday, 2015-08-14:
Yes, a MAC address can be easily spoofed. Instead, a better a solution would be to use the whitelist idea and on top of that, when your user wants open the gate, have them bond with the opener, send a specific key that only that phone has that only your app generates to the peripheral, determine if they key is valid, and if it is, open the garage. A spoofer might be able to get past the first step (the whitelist), but it's very unlikely they'll be able to past the key your app generates. On top of that, if you bond, your phone and the peripheral will exchange a Long Term Key (LTK) so your peripheral will only respond to bonded devices (assuming the peripheral isn't constantly advertising, even if it is the app layer key will still be a safeguard)

sandeepmistry, 2015-08-14:
OOB pairing should be much easier to implement now. There is a minimal SMP layer built into bleno now. See: https://github.com/sandeepmistry/bleno/blob/master/lib/hci-socket/smp.js

The current implementation only supports "Just Works" security.

However, if you follow this guide: https://community.freescale.com/thread/332191 you should be able to prototype OOB pairing.

Let me know if you get it working, then we can discuss API's ...

ostrichjones, 2016-01-17:
I've found myself wanted to implement pairing using a PIN (I am trying to replicate a BLE device - the Android app expects to pair this way). I found myself back at this thread in which I asked about OOB pairing. I really can't figure out how to use the SMP layer in smp.js with bleno - is there any example code available or could a simple example be posted?

sandeepmistry, 2016-01-23:
@ostrichjones 

>  is there any example code available or could a simple example be posted?

Not that I'm aware of at this time.