noble/bleno #114: Uncaught Exception: Error: Module did not self-register on electron

ragingwind, 2015-07-19:
I've got an error when I was trying to use node-eddystone-beacon(bleno) with electron. But It works well on terminal with node

node v0.10.29
electron: 0.29.2
os: mac 10.10.3
here is the error message:

App threw an error when running [Error: Module did not self-register.]
2015-07-17 01:27:37.122 Electron[14999:2357677] Communications error: { count = 1, contents =
"XPCErrorDescription" => { length = 18, contents = "Connection invalid" }
}>
2015-07-17 01:27:37.122 Electron[14999:2357587] --error: [ShareKit] error calling +[NSExtension extensionsWithMatchingAttributes:error:]: Error Domain=PlugInKit Code=4 "Connection invalid" UserInfo=0x7fc1385b07f0 {NSLocalizedDescription=Connection invalid}

sandeepmistry, 2015-07-19:
@ragingwind have you looked into what might be causing the issue?

Supporting ```electron``` is not high on my list ...

ragingwind, 2015-07-19:
Nop. I tested with electron just once. I wished It might be easy an issue to you. I will look into it more

ragingwind, 2015-07-22:
FYI. I've tested on node v0.12.0. I've got same error that xpc-connectiion failed to register. Did you have test on node v0.12.0?

sandeepmistry, 2015-07-26:
@ragingwind I don't use once v0.12.0 on my mac yet, but I'm sure others have got it working.

Did you do a fresh install of bleno after installing node 0.12.0? The xpc-connection module needs to be rebuilt using node-gyp, when switching between releases.

latentflip, 2015-08-10:
You need to be running the same version of iojs that electron uses when you install the module. You can see the version of iojs that electron is running using `process.version` from the devtools of the electron app. Do that, install the right version of iojs with nvm or whatever, and reinstall bleno/noble, and you should be in business.

sandeepmistry, 2015-08-12:
@latentflip thanks for the suggestion!

@ragingwind any updates? Have you tried my suggestion or @latentflip's yet?

ragingwind, 2015-08-15:
I've test with iojs 1.0/2.0/3.0. 1.0 and 3.0 have some issue for build. 2.0 has no problem for starting electron apps. I'll close the issue after testing of advertising. @latentflip thanks! I've forgotten about that

sandeepmistry, 2015-08-17:
@ragingwind closing this, as your original module loading issue has been resolved.

closed 2015-08-17 by sandeepmistry