noble/bleno #115: Raised `xpcError: connection interrupted` when use  advertise EIRDdata over 29 bytes

ragingwind, 2015-07-22:
It seems that xpc-connection has some issues for MacOS 10.10.x but I'm not sure that my issue like being related to the others.

`xpcError: connection interrupted` error occurred when I was trying to write data over at 29 bytes by `startAdvertisingWithEIRData`. Actually, I tried to make Eddystone UID frame which has bytes over 28 bytes (31bytes).  Here is my simple test code please let me know what are problems.

- OS: Mac, Yosemite, 10.10.3
- bleno: 0.1.14
- node: 0.10.29

```
var bleno = require('bleno');
var data = new Buffer(29);
data.fill(1);

function start(data) {
	if (bleno.state === 'poweredOn') {
		bleno.startAdvertisingWithEIRData(data);
	} else {
		bleno.once('stateChange', function() {
			start(data);
		});
	}
}

bleno.once('advertisingStart', function (err) {
	console.log('advertisingStart', err);
});

start(data);

```

sandeepmistry, 2015-07-26:
@ragingwind you are crashing ```blued```, ```bleno.startAdvertisingWithEIRData``` is an advanced API with no error checking on OS X.

It looks like OS X will auto include the GAP flags EIR data in the advertisement data, so only supports upto 28 bytes (instead of 31).

Closing this as not much can be done on the bleno side.

closed 2015-07-26 by sandeepmistry

ragingwind, 2015-07-27:
I'm afraid of that I'm not familiar with BLE(bluetooth). Let me be clear please. Could you get solve the problem at the bleno side after  for a while even OS X is including GAP flags EIR data automatically to advertisement data? I mean, I would like to know that should I get a linux device to use the `bleno. startAdvertisingWithEIRData`.

sandeepmistry, 2015-08-02:
@ragingwind I'm not planning to support this OS X specific oddity at this time. Just don't include the first 3 bytes of the Eddystone frame spec for now when running on OS X: https://github.com/google/eddystone/blob/master/protocol-specification.md

ragingwind, 2015-08-02:
@sandeepmistry I saw the implementation at `node-eddystone-beacon'. I'm using the node module now. Thanks for guide