noble/bleno #116: node-gyp build error

HmidiSlim, 2015-08-01:
I run this command:node-gyp build but I got these errors:
gyp info it worked if it ends with ok
gyp info using node-gyp@2.0.2
gyp info using node@0.12.7 | linux | ia32
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
make: Entering directory `/home/slim/test/build'
make: *** No rule to make target `Release/obj.target/example/example.o', needed by `Release/obj.target/example.node'.  Stop.
make: Leaving directory `/home/slim/test/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/node-gyp/lib/build.js:269:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1074:12)
gyp ERR! System Linux 3.13.0-32-generic
gyp ERR! command "node" "/usr/bin/node-gyp" "build"
gyp ERR! cwd /home/slim/test
gyp ERR! node -v v0.12.7
gyp ERR! node-gyp -v v2.0.2
gyp ERR! not ok

sandeepmistry, 2015-08-10:
@HmidiSlim unfortunately this info. is not enough for me to help you ...

What happens when you run ```npm install bleno```? Anymore output? Have you installed the prerequisites? https://github.com/sandeepmistry/bleno#linux-ubuntu

Btw, I don't see ```bleno``` anywhere in the path for the snippet you provided ...

sandeepmistry, 2015-08-18:
Closing due to lack of activity ...

closed 2015-08-18 by sandeepmistry