noble/bleno #117: Unable to "npm install bleno"

sokolware, 2015-08-08:
When i try to install bleno I get the following stack trace:

```
npm install bleno
npm WARN optional dep failed, continuing xpc-connection@0.1.3

> bleno@0.1.14 install /home/pi/workspace/node_modules/bleno
> node-gyp rebuild

gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/0.12.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/workspace/node_modules/bleno/.node-gyp"

gyp WARN install got an error, rolling back install
gyp WARN install got an error, rolling back install
gyp ERR! configure error 
gyp ERR! stack Error: node-v0.12.0.tar.gz local checksum ee8203eecbf053b32001307fb65ec14ef2d3042dfb73f16e9adef7ce5675e1ce not match remote 9700e23af4e9b3643af48cef5f2ad20a1331ff531a12154eef2bfb0bb1682e32
gyp ERR! stack     at deref (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/install.js:299:20)
gyp ERR! stack     at IncomingMessage.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/install.js:340:13)
gyp ERR! stack     at IncomingMessage.emit (events.js:129:20)
gyp ERR! stack     at _stream_readable.js:908:16
gyp ERR! stack     at process._tickCallback (node.js:355:11)
gyp ERR! System Linux 3.12.22+
gyp ERR! command "node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/workspace/node_modules/bleno
gyp ERR! node -v v0.12.0
gyp ERR! node-gyp -v v1.0.2
gyp ERR! not ok 
npm ERR! Linux 3.12.22+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "bleno"
npm ERR! node v0.12.0
npm ERR! npm  v2.5.1
npm ERR! code ELIFECYCLE

npm ERR! bleno@0.1.14 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the bleno@0.1.14 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the bleno package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls bleno
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/workspace/npm-debug.log

```

Please help resolve this issue

sokolware, 2015-08-08:
Unable to attached the debug log, here is the content:

```
0 info it worked if it ends with ok
1 verbose cli [ '/usr/local/bin/node',
1 verbose cli   '/usr/local/bin/npm',
1 verbose cli   'install',
1 verbose cli   'bleno' ]
2 info using npm@2.5.1
3 info using node@v0.12.0
4 silly cache add args [ 'bleno', null ]
5 verbose cache add spec bleno
6 silly cache add parsed spec { raw: 'bleno',
6 silly cache add   scope: null,
6 silly cache add   name: 'bleno',
6 silly cache add   rawSpec: '',
6 silly cache add   spec: '*',
6 silly cache add   type: 'range' }
7 verbose addNamed bleno@*
8 silly addNamed semver.valid null
9 silly addNamed semver.validRange *
10 silly addNameRange { name: 'bleno', range: '*', hasData: false }
11 silly mapToRegistry name bleno
12 silly mapToRegistry using default registry
13 silly mapToRegistry registry https://registry.npmjs.org/
14 silly mapToRegistry uri https://registry.npmjs.org/bleno
15 verbose addNameRange registry:https://registry.npmjs.org/bleno not in flight; fetching
16 verbose request uri https://registry.npmjs.org/bleno
17 verbose request no auth needed
18 info attempt registry request try #1 at 17:37:24
19 verbose request id 828c93b051f43b9d
20 verbose etag "6INOP4GZRAG90S3V3QKDAEGLF"
21 http request GET https://registry.npmjs.org/bleno
22 http 304 https://registry.npmjs.org/bleno
23 silly get cb [ 304,
23 silly get   { date: 'Sat, 08 Aug 2015 17:37:26 GMT',
23 silly get     via: '1.1 varnish',
23 silly get     'cache-control': 'max-age=60',
23 silly get     etag: '"6INOP4GZRAG90S3V3QKDAEGLF"',
23 silly get     age: '0',
23 silly get     connection: 'keep-alive',
23 silly get     'x-served-by': 'cache-lax1434-LAX',
23 silly get     'x-cache': 'MISS',
23 silly get     'x-cache-hits': '0',
23 silly get     'x-timer': 'S1439055446.051122,VS0,VE185',
23 silly get     vary: 'Accept' } ]
24 verbose etag https://registry.npmjs.org/bleno from cache
25 verbose get saving bleno to /root/.npm/registry.npmjs.org/bleno/.cache.json
26 silly addNameRange number 2 { name: 'bleno', range: '*', hasData: true }
27 silly addNameRange versions [ 'bleno',
27 silly addNameRange   [ '0.0.1',
27 silly addNameRange     '0.0.2',
27 silly addNameRange     '0.0.3',
27 silly addNameRange     '0.1.0',
27 silly addNameRange     '0.1.1',
27 silly addNameRange     '0.1.2',
27 silly addNameRange     '0.1.3',
27 silly addNameRange     '0.1.4',
27 silly addNameRange     '0.1.5',
27 silly addNameRange     '0.1.6',
27 silly addNameRange     '0.1.7',
27 silly addNameRange     '0.1.8',
27 silly addNameRange     '0.1.9',
27 silly addNameRange     '0.1.10',
27 silly addNameRange     '0.1.11',
27 silly addNameRange     '0.1.12',
27 silly addNameRange     '0.1.13',
27 silly addNameRange     '0.1.14' ] ]
28 verbose addNamed bleno@0.1.14
29 silly addNamed semver.valid 0.1.14
[... 472 lines cut ...]
271 error node v0.12.0
272 error npm  v2.5.1
273 error code ELIFECYCLE
274 error bleno@0.1.14 install: `node-gyp rebuild`
274 error Exit status 1
275 error Failed at the bleno@0.1.14 install script 'node-gyp rebuild'.
275 error This is most likely a problem with the bleno package,
275 error not with npm itself.
275 error Tell the author that this fails on your system:
275 error     node-gyp rebuild
275 error You can get their info via:
275 error     npm owner ls bleno
275 error There is likely additional logging output above.
276 verbose exit [ 1, true ]
277 verbose unbuild node_modules/bleno
278 info preuninstall bleno@0.1.14
279 info uninstall bleno@0.1.14
280 verbose unbuild rmStuff bleno@0.1.14 from /home/pi/workspace/node_modules
281 info postuninstall bleno@0.1.14
282 verbose gentlyRm vacuuming /home/pi/workspace/node_modules/bleno
283 silly gentlyRm purging /home/pi/workspace/node_modules/bleno
284 silly gentlyRm removing /home/pi/workspace/node_modules
285 silly gentlyRm finished vacuuming up to /home/pi/workspace

```

sandeepmistry, 2015-08-08:
@sokolware please see: https://github.com/TooTallNate/node-gyp/issues/454

This is a ```node-gyp``` issue not related to bleno ...

closed 2015-08-08 by sandeepmistry