noble/bleno #118: Client Characteristic Descriptor (2902) value for notify

francip, 2015-08-11:
Hi, I am looking at https://github.com/sandeepmistry/bleno/blob/master/lib/hci-socket/gatt.js#L213 and was wondering - shouldn't the value for this descriptor be `new Buffer([0x01, 0x00])` for notify, `new Buffer([0x02, 0x00])` for indicate, and `new Buffer([0x03, 0x00])` for both? At least, as I read https://developer.bluetooth.org/gatt/descriptors/Pages/DescriptorViewer.aspx?u=org.bluetooth.descriptor.gatt.client_characteristic_configuration.xml, value `00` means both notify and indicate are disabled for the characteristic.

sandeepmistry, 2015-08-13:
@francip most peripheral's I've seen set the value to 0x0000 initially, so notifications + indications are disabled. Then the central writes 0x0001 or 0x0002 to enable notifications or indications.

Is there an issue with this or is this a general question?

closed 2015-08-13 by sandeepmistry

ilker-aktuna, 2020-05-27:
> @francip most peripheral's I've seen set the value to 0x0000 initially, so notifications + indications are disabled. Then the central writes 0x0001 or 0x0002 to enable notifications or indications.
> 
> Is there an issue with this or is this a general question?

with which command can we set 0x0002 to a specific characteristic client configuration ?