noble/bleno #119: working with android

jonalumot, 2015-08-12:
hello,

First of all, thank you for your amazing project.

I installed the bleno on a raspberry pi and set my own service. 
Also, i made an android app that scans and looks for ble devices.
I succeed to connect to the pi using gatt but when i called the getServices method, i didn't get the service i set before.

Do you have any suggestions for me?
Thanks!

sandeepmistry, 2015-08-13:
@jonalumot I suggest you try running [test.js](https://github.com/sandeepmistry/bleno/blob/master/test.js) with [nRF Master Control Panel](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en). I have no issues with both interacting on my Raspberry Pi.

What version of BlueZ do you have installed? If you are running 5.14 or later you will need to stop ```bluetoothd``` as it interferes with bleno.

If that doesn't work, please provide debug logs: ```sudo DEBUG=* node test.js```

sandeepmistry, 2015-08-18:
@jonalumot any updates?

sandeepmistry, 2015-08-30:
Closing due to lack of activity.

closed 2015-08-30 by sandeepmistry

py2, 2015-10-01:
hi sandeepmistry, thank for this code. 
I am currently facing the same problem, I am running the test.js bleno code with the  nRF Master Control Panel app and I only discover the Generic Access Service which does not have any characteristics. It seems like the advertising is stopped somewhere. 
Thank you for your help

py2, 2015-10-01:
Ok, sorry, it's working now, I had an old version of bluez running bluetoothd. I removed it with only bluez 5 left.