noble/bleno #120: Windows 8.1 can't pare when advert with startAdvertisingWithEIRData

PeterKow, 2015-08-17:
Hi, 

Anyone have a problem when running advertisement in startAdvertisingWithEIRData mode on raspberry?

In my case Windows is picking my device, but can't pair with it. Can anyone share some working advertisement for Windows? 

Currently I am using this:
<code>var advert = new Buffer('060950686F6E65050315111F110105010700','hex');
this._hciBle.stdin.write(advert.toString('hex') + '\n');</code>

and advert value I got from 27 page of this e.g. https://www.bluetooth.org/docman/handlers/downloaddoc.ashx?doc_id=245130

Thanks!

sandeepmistry, 2015-08-18:
@PeterKow next steps:

* can you please try again with test.js from bleno ```0.2.0```?
* share the code you are running, via something like Github gist
* also share debug logs: ```sudo DEBUG=* node <file>.js```
* expand on what "doesn't work" means. What app are you using on Android etc.

PeterKow, 2015-08-25:
Our main problem was how to build advertisement for BLE EIR, finally we manage to create one which works on Windows and Android.

## BLE Advertisement

This document helps to clear up any Bluetooth LE questions.
For more documentation

* [Supplement to the Bluetooth Core Specification](https://www.bluetooth.org/docman/handlers/downloaddoc.ashx?doc_id=245130)
* [Assigned numbers](https://www.bluetooth.org/en-us/specification/assigned-numbers)

## BLE Advertisement

The BLE advertisement is broken down into:

| 1. Length and flags | 2. Pre-manufacturer data | 3. Manufacturer data | 4. Local name | 5. Ending flag |
| --- | --- | --- | --- | --- |
| 020106 | 03190000 | 07FFXXXX00000000 | 0D09536f6d65206578616d706c65 | 00 |

That is, the entire advertisement of the device with "Some example" local name:

    0201060319000007FFXXXX000000000D09536f6d65206578616d706c6500

See page 27 in the specification for an example of advertising data

### 1. Length and flags

| Length of this Data | Length of Flag | Flags^ |
| --- | --- | --- |
| 02  | 01 | 06 |

^ The format of the flags are defined on page 12 of the spec. It is `06` because in binary it is `00000110` where the 1st bit (counting from 0) is for "LE General Discoverable Mode", and the 2nd bit is for "BR/EDR Not Supported. Bit 37 of LMP Feature Mask Definitions"

### 2. Pre-manufacturer data

After the discovery flags - I don't know why it's there but works
Value: `03190000`

### 3. Manufacturer data

| Length of data | Manufacturer data start | Your company ID | Additional data |
| --- | --- | --- | --- |
| 07 | FF | XXXX | 00000000 |

* Length of data - length of data including `Manufacturer data start` - fixed
* Manufacturer data start - BLE id to indicate that manufacturer data is going to follow
* Your company manufacturer ID -(https://www.bluetooth.org/en-us/specification/assigned-numbers/company-identifiers) (no. 460). You need to paid to register for this ID.
* Additional data

eg, `07FFXXXX0000000000 `

### 4. Local name

| Length of local name | Local name start | `Some example` local name |
| --- | --- | --- |
| 0D | 09 | 536f6d65206578616d706c65 |

* Length of local name - `D` in hex = 13 characters (including the `Local name start`, where each character is 16 bits/2 octets)
* Local name start - BLE id to indicate that local name is going to follow (see page 27)
* `Some example` local name - the hex representation of `Some example` as a local name

### 5. Ending flag

To end the advertising packet, you need to add this

Value: `00`

## Glossary

* EIR = Extended Inquiry Response
* AD = Advertising Data
* OOB = Out of Band

PeterKow, 2015-08-25:
Also regards to Bleno you need to create advertisement as follows 

```bleno.startAdvertisingWithEIRData('0201060319000007FFXXXX000000000D09536f6d65206578616d706c6500', null, callback)```;

@sandeepmistry I don't know why when I want to split advertisement into advertisementData and scanData 

E.g.
```var advertisementData =  "0201060319000007FFXXXX000000000D09"```
```var scanData = "536f6d65206578616d706c6500"```
```bleno.startAdvertisingWithEIRData(advertisementData, scanData, callback)```

It does't work, I mange pinpoint that it is caused by additional " " (space) between advertisementData.toString and scanData.toString.

```this._hciBle.stdin.write(advertisementData.toString('hex') + ' ' + scanData.toString('hex') + '\n');```
in ```bleno/lib/linux/Hci-ble.js```

Any ideas why?

Thanks!

sandeepmistry, 2015-08-29:
@PeterKow Please try with the latest 0.2.0 bleno release.

Also, you need to pass buffer's in, not strings:

```javascript
var advertisementData = new Buffer("0201060319000007FFXXXX000000000D09", "hex");
var scanData = new Buffer("536f6d65206578616d706c6500");
bleno.startAdvertisingWithEIRData(advertisementData, scanData, callback)
```

PeterKow, 2015-09-02:
Thanks, my mistake it should be new Buffer and this is what we have in our code.

I also tried with belno 0.2.0 but it doesn't work. For now I am closing this issue, because it might be just our advertisement which is specific ti our solution. Hope our previous post about advertisement will help someone!

Thanks!

closed 2015-09-02 by PeterKow