noble/bleno #123: Linux, Still get unauthorized error after running setcap,

AdaRoseEdwards, 2015-08-26:
```
~ sudo setcap cap_net_raw+eip /home/ada/bin/iojs/bin/iojs
~ iojs .
bleno warning: adapter state unauthorized, please run as root or with sudo
               or see README for information on running without root/sudo:
               https://github.com/sandeepmistry/bleno#running-on-linux
 ```

sandeepmistry, 2015-08-29:
@AdaRoseEdwards could you please provide debug logs:

```sh
DEBUG=* iojs <file>.js
```

AdaRoseEdwards, 2015-08-31:
```
~ DEBUG=* iojs ../physical-web/index.js 
  hci-ble hciBle = /home/ada/gitWorkingDir/physical-web/node_modules/eddystone-beacon/node_modules/bleno/lib/linux/../../build/Release/hci-ble +0ms
  l2cap-ble l2capBle = /home/ada/gitWorkingDir/physical-web/node_modules/eddystone-beacon/node_modules/bleno/lib/linux/../../build/Release/l2cap-ble +0ms
  l2cap-ble handles = [
  {},
  {
    "type": "service",
    "uuid": "1800",
    "startHandle": 1,
    "endHandle": 5
  },
  {
    "type": "characteristic",
    "uuid": "2a00",
    "properties": 2,
    "secure": 0,
    "startHandle": 2,
    "valueHandle": 3
  },
  {
    "type": "characteristicValue",
    "handle": 3,
    "value": "Buffer('4e6f6d69', 'hex')"
  },
  {
    "type": "characteristic",
    "uuid": "2a01",
    "properties": 2,
    "secure": 0,
    "startHandle": 4,
    "valueHandle": 5
  },
  {
    "type": "characteristicValue",
    "handle": 5,
    "value": "Buffer('8000', 'hex')"
  },
  {
    "type": "service",
    "uuid": "1801",
    "startHandle": 6,
    "endHandle": 9
  },
  {
    "type": "characteristic",
    "uuid": "2a05",
    "properties": 32,
    "secure": 0,
    "startHandle": 7,
    "valueHandle": 8
  },
  {
    "type": "characteristicValue",
    "handle": 8,
    "value": "Buffer('00000000', 'hex')"
  },
  {
    "type": "descriptor",
    "handle": 9,
    "uuid": "2902",
    "properties": 14,
    "secure": 0,
    "value": "Buffer('0000', 'hex')"
  }
] +4ms
  hci-ble buffer = "address 6C:29:95:71:29:55\nadapterState unauthorized\n" +29ms
  hci-ble line = address 6C:29:95:71:29:55 +0ms
  hci-ble adapterAddress = 6c:29:95:71:29:55 +0ms
  bleno addressChange 6c:29:95:71:29:55 +0ms
  hci-ble line = adapterState unauthorized +0ms
  hci-ble adapterState = unauthorized +0ms
bleno warning: adapter state unauthorized, please run as root or with sudo
               or see README for information on running without root/sudo:
               https://github.com/sandeepmistry/bleno#running-on-linux
  bleno stateChange unauthorized +1ms
  l2cap-ble buffer = "bind success\nlisten success\n" +78ms
  l2cap-ble line = bind success +0ms
  l2cap-ble line = listen success +0ms

```

AdaRoseEdwards, 2015-08-31:
Whoops wrong button.

AdaRoseEdwards, 2015-08-31:
I've confirmed It does work if run with sudo.

sandeepmistry, 2015-09-05:
@AdaRoseEdwards got it, it's because you are using an older version of bleno, but the read me is only for the current version. Two options:

1) Upgrade to the latest bleno version (preferred)
2) Follow old instructions: https://github.com/sandeepmistry/bleno/tree/1a57b7d8a4a9616138dbcd806f1949df99bc1d96#running-without-rootsudo

closed 2015-09-05 by sandeepmistry