noble/bleno #125: Advertising power

carlonluca, 2015-08-30:
I see BLENO_ADVERTISING_INTERVAL can be used to set the frequency of the advertising packets. But is it possible also to set the tx power of the advertising by any chance? Is there a way to do it? What is the default power?

sandeepmistry, 2015-08-30:
@carlonluca nothing yet, a pull request would be great :)

Default power is whatever the adapter defaults to. Which I think is ```sudo hciconfig hci0 inqtpl```, but not 100% sure.

carlonluca, 2015-08-30:
Yes, I found that too. Although trying to modify it had no result. I'll investigate further. Thanks for answering.

sandeepmistry, 2015-08-30:
Yes, I had a similar observation here: https://github.com/don/node-eddystone-beacon/issues/19

carlonluca, 2015-08-30:
Oh, it means I'm not alone... :-) maybe not possible to do that with the hardware... it may be confirmed if there was a way to get the range of available values but I couldn't find it.

sandeepmistry, 2015-09-05:
@carlonluca any progress or should I close?

carlonluca, 2015-09-05:
It depends on you.

sandeepmistry, 2015-09-07:
Closing this for now, since we haven't been able to find a HCI command to set the TX power level.

closed 2015-09-07 by sandeepmistry