noble/bleno #126: processCmdCompleteEvent callback stopping LE Advertising

thegeebe, 2015-09-02:
I am advertising an iBeacon, but this goes for any LE advertising. The advertising and scan response data is set and then the command to enable LE advertising is sent to the controller. Soon after, `processCmdCompleteEvent` is called with the `READ_LOCAL_VERSION_CMD` command and  enters into the block that first disables LE advertising and then sets the advertising parameters.

This behavior seems deliberate and so I am wondering if I set something up wrong that is causing this command to come back and disable LE advertising right after the HCI issues a response.

Output from debug [here](https://gist.github.com/thegeebe/dff201102da766009a31), output from hcidump [here](https://gist.github.com/thegeebe/153fb8fefcef8ad13f2c)

Note that on this run I commented out lines 155 and 156 of [gap.js](https://github.com/sandeepmistry/bleno/blob/master/lib/hci-socket/gap.js#L155) because I thought they were causing the error, but it is irrelevant. 

Using Ubuntu 14.10, Node v0.10.25. My script is:

    var bleno = require('bleno')

    var uuid = 'e2c56db5dffb48d2b060d0f5a71096e0';
    var major = 15; // 0x0000 - 0xffff
    var minor = 147; // 0x0000 - 0xffff
    var measuredPower = -59; // -128 - 127

    bleno.startAdvertisingIBeacon(uuid, major, minor, measuredPower, function(error){
      console.log(error);
    });

Any insights?

sandeepmistry, 2015-09-05:
@thegeebe 

This is unclear in the read me, but you need to wait for a ```stateChange``` event of ```poweredOn``` before starting advertisement. See: https://github.com/sandeepmistry/bleno/blob/master/test-ibeacon.js

I'll update the read me and change the code to handle incorrect usage.

closed 2015-09-05 by sandeepmistry

thegeebe, 2015-09-05:
Thanks! @sandeepmistry