noble/bleno #127: GATT Service Changed Characteristic

popasquat89, 2015-09-14:
Would there be any willingness to accept a PR that adds support for the Service Changed Characteristic? 
(Ref Vol 3. Part G. 7.1)

I've started some initial work on a fork of mine, basically modifying the gatt.js and abstracting some functions. I've only started work on Linux, and can look into Mac compatibility as well if this would be wanted. It would use the same characteristic (uuid - 1801) as before, just need to modify the value when we have some services that have changed.

The goal would be to add the following prototype functions in gatt.js:

- addService
- removeService

This would enable attribute caching on the central side of things, and enable services to be taken out without causing problems when there is a connection. 

I'm looking to help out if at all possible, to support the project and use it for our needs. 

cc/ @sandeepmistry 
Thanks, 
Bryce

popasquat89, 2015-09-14:
Ref #128

sandeepmistry, 2015-09-15:
@popasquat89 for sure, it would be great to have this feature! However, I'm trying to keep the API for both OS X and Linux in sync, so it would be great to have both in at the same time.

popasquat89, 2015-09-16:
I'll try to make some time today to patch the Mac bindings as well!

popasquat89, 2015-09-17:
@sandeepmistry What's your process for discovering xpc core bluetooth message id's? I'm specifically interested in the XPC message id for the removeService (assuming it's different from removeAllServices, which is this.sendCBMsg(12, null)).

I can't find a reference to the message id's anywhere on the internet, so I'm assuming there has to be some way of doing it by sniffing them out.

Link Reference: 
https://developer.apple.com/library/ios/documentation/CoreBluetooth/Reference/CBPeripheralManager_Class/#//apple_ref/occ/instm/CBPeripheralManager/removeService:

I'm also curious if they automatically update the Service Changed Characteristic. Thinking out loud here, let me know if you have any insight. Getting late here, going to finish this up tomorrow.

Thanks,
Bryce

sandeepmistry, 2015-09-17:
@popasquat89 to sniff out see: https://github.com/sandeepmistry/osx-ble-peripheral/blob/master/BLEPeripheral/BPAppDelegate.m#L83

and comment the ``// #define XPC_SPY 1`` line.

I'll probably be swapping out the XPC layer with [node-core-bluetooth](https://github.com/sandeepmistry/node-core-bluetooth) at a later date, but I just started on it.

popasquat89, 2015-09-18:
@sandeepmistry Take a look when you have a chance, updated with Mac compatibility and squashed down the commit.

Let me know if you need anything.

rzr, 2020-11-04:
Please check if this ticket is still relevant for this release:

https://www.npmjs.com/package/@abandonware/bleno/v/0.5.1-3