noble/bleno #131: Intermittently working on Edison

mikaye, 2015-09-24:
When running test.js on the edison, sometimes the android scanner sees it advertise as 'test' and I'm able to connect and all works as I would expect. Mostly, it sees it as "BCM433..." and I'm unable to connect

mikaye, 2015-09-25:
Well it works fine with iOS. Seems to be something with any of the off the shelf BLE scanners on Android. Any ideas welcome

sandeepmistry, 2015-09-28:
@mikaye hm, strange, I'm wondering if this is driver related.

Could you please capture an hcidump capture, while running the test.js:
```sh
hcidump -t -x
```

mikaye, 2015-09-28:
Hi Sandeep,

This is what I see.

HCI sniffer - Bluetooth packet analyzer ver 5.24

device: hci0 snap_len: 1500 filter: 0xffffffff

*From iOS lightblue application I see*

*edisonNo Services*

*after connect*

edison

uuid: ...  (and all expected services)

2015-09-28 10:58:12.122561 > HCI Event: LE Meta Event (0x3e) plen 19

    LE Connection Complete

      status 0x00 handle 64, role slave

      bdaddr 66:C1:D7:AA:C3:1F (Random)

on -> accept, client: 66:c1:d7:aa:c3:1f

2015-09-28 10:58:12.134761 < HCI Command: Read RSSI (0x05|0x0005) plen 2

    handle 64

2015-09-28 10:58:12.135492 > HCI Event: Command Complete (0x0e) plen 7

    Read RSSI (0x05|0x0005) ncmd 1

    status 0x00 handle 64 rssi 127

on -> rssiUpdate: 127

2015-09-28 10:58:12.200699 > ACL data: handle 64 flags 0x02 dlen 7

    ATT: MTU req (0x02)

      client rx mtu 158

2015-09-28 10:58:12.205808 < ACL data: handle 64 flags 0x00 dlen 7

    ATT: MTU resp (0x03)

      server rx mtu 158

2015-09-28 10:58:12.290769 > ACL data: handle 64 flags 0x02 dlen 11

    ATT: Read By Group req (0x10)

      start 0x0001, end 0xffff

      type-uuid 0x2800

2015-09-28 10:58:12.296682 < ACL data: handle 64 flags 0x00 dlen 18

    ATT: Read By Group resp (0x11)

      attr handle 0x0001, end group handle 0x0005

      value 0x00 0x18

      attr handle 0x0006, end group handle 0x0009

[... 319 lines cut ...]
*On Android (5.1.1 on Nexus 6)*

application BLE Scanner

Peripheral: BCM4334080..

hcidump -t -x

HCI sniffer - Bluetooth packet analyzer ver 5.24
device: hci0 snap_len: 1500 filter: 0xffffffff

When I tap on connect, nothing. This is true for BLE Scanner, BLE Tool and
my own app.

Strange, because the first time I tried it, it worked flawlessly from
Android. Since then, I've been unable to get it to work.

Also tried a Nexus 5 running 5.1.1. Same behavior.

Both phones show the Edison as "dual mode" device. However, I doubt that
should matter.

Thanks,

Mike

sandeepmistry, 2015-09-29:
@mikaye we really need an hcidump capture running before you start bleno, and trying to connect from Android. I want to see the advertising commands in the hcidump capture.

Maybe you could reboot your phone?

Btw, you don't have ```bluetoothd``` running right? http://rexstjohn.com/configure-intel-edison-for-bluetooth-le-smart-development/

mikaye, 2015-09-29:
Hi Sandeep,

The advertising data is below. No success with either Android phone even
though I run BLE Scanner after boot. All I can say is it worked from the
build I pulled several weeks ago before your latest checkin. I don't
remember if it was consistent though. And to confirm, yes, I followed the
instructions and bluetoothd is not running. iOS connects and disconnects
flawlessly.

I noticed that if android BLE scanner starts to connect, I do not see any
dump from hcidump but if I then try connecting from iOS (lightblue app) it
first says something like it is disconnected from the peripheral. All
subsequent iOS attempts work fine.

see below

Thx,
Mike

Advertising data
---------------------

HCI sniffer - Bluetooth packet analyzer ver 5.24

device: hci0 snap_len: 1500 filter: 0xffffffff

2015-09-29 08:44:38.990421 < HCI Command: Set Event Mask (0x03|0x0001) plen
8

    Mask: 0xfffffbff07f8bf3d

2015-09-29 08:44:38.997108 > HCI Event: Command Complete (0x0e) plen 4

    Set Event Mask (0x03|0x0001) ncmd 1

    status 0x00

2015-09-29 08:44:38.997223 < HCI Command: LE Set Event Mask (0x08|0x0001)
plen 8

    mask 0x1f00000000000000 (Reserved)

2015-09-29 08:44:38.997970 > HCI Event: Command Complete (0x0e) plen 4

    LE Set Event Mask (0x08|0x0001) ncmd 1

    status 0x00

2015-09-29 08:44:38.998065 < HCI Command: Read Local Version Information
(0x04|0x0001) plen 0

2015-09-29 08:44:38.998731 > HCI Event: Command Complete (0x0e) plen 12

    Read Local Version Information (0x04|0x0001) ncmd 1

    status 0x00

    HCI Version: 4.0 (0x6) HCI Revision: 0xa6

    LMP Version: 4.0 (0x6) LMP Subversion: 0x410e

    Manufacturer: Broadcom Corporation (15)

2015-09-29 08:44:38.998826 < HCI Command: Read BD ADDR (0x04|0x0009) plen 0

2015-09-29 08:44:38.999562 > HCI Event: Command Complete (0x0e) plen 10

    Read BD ADDR (0x04|0x0009) ncmd 1

    status 0x00 bdaddr 98:4F:EE:03:26:2B

2015-09-29 08:44:39.038835 < HCI Command: LE Set Advertise Enable
(0x08|0x000a) plen 1

  00

2015-09-29 08:44:39.039970 > HCI Event: Command Complete (0x0e) plen 4

    LE Set Advertise Enable (0x08|0x000a) ncmd 1

    status 0x0c

    Error: Command Disallowed

2015-09-29 08:44:39.041078 < HCI Command: LE Set Advertising Parameters
(0x08|0x0006) plen 15

    min 100.000ms, max 100.000ms

    type 0x00 (ADV_IND - Connectable undirected advertising) ownbdaddr 0x00
(Public)

    directbdaddr 0x00 (Public) 00:00:00:00:00:00

    channelmap 0x07 filterpolicy 0x00 (Allow scan from any, connection from
any)

2015-09-29 08:44:39.041895 > HCI Event: Command Complete (0x0e) plen 4

    LE Set Advertising Parameters (0x08|0x0006) ncmd 1

    status 0x00

2015-09-29 08:44:39.058031 < HCI Command: LE Set Scan Response Data
(0x08|0x0009) plen 32

  06 05 08 74 65 73 74 00 00 00 00 00 00 00 00 00 00 00 00 00

  00 00 00 00 00 00 00 00 00 00 00 00

2015-09-29 08:44:39.059281 > HCI Event: Command Complete (0x0e) plen 4

    LE Set Scan Response Data (0x08|0x0009) ncmd 1

    status 0x00

2015-09-29 08:44:39.059391 < HCI Command: LE Set Advertising Data
(0x08|0x0008) plen 32

  15 02 01 06 11 06 F0 FF FF FF FF FF FF FF FF FF FF FF FF FF

  FF FF 00 00 00 00 00 00 00 00 00 00

2015-09-29 08:44:39.060256 > HCI Event: Command Complete (0x0e) plen 4

    LE Set Advertising Data (0x08|0x0008) ncmd 1

    status 0x00

2015-09-29 08:44:39.060347 < HCI Command: LE Set Advertise Enable
(0x08|0x000a) plen 1

  01

2015-09-29 08:44:39.061798 > HCI Event: Command Complete (0x0e) plen 4

    LE Set Advertise Enable (0x08|0x000a) ncmd 1

    status 0x00

2015-09-29 08:44:39.061908 < HCI Command: LE Set Scan Response Data
(0x08|0x0009) plen 32

  06 05 08 74 65 73 74 00 00 00 00 00 00 00 00 00 00 00 00 00

  00 00 00 00 00 00 00 00 00 00 00 00

2015-09-29 08:44:39.062905 > HCI Event: Command Complete (0x0e) plen 4

    LE Set Scan Response Data (0x08|0x0009) ncmd 1

    status 0x00

2015-09-29 08:44:39.064513 < HCI Command: LE Set Advertising Data
(0x08|0x0008) plen 32

  15 02 01 06 11 06 F0 FF FF FF FF FF FF FF FF FF FF FF FF FF

  FF FF 00 00 00 00 00 00 00 00 00 00

2015-09-29 08:44:39.065442 > HCI Event: Command Complete (0x0e) plen 4

    LE Set Advertising Data (0x08|0x0008) ncmd 1

    status 0x00

2015-09-29 08:44:39.065535 < HCI Command: LE Set Advertising Data
(0x08|0x0008) plen 32

  1F 02 01 0A 02 0A 04 18 08 42 43 4D 34 33 33 34 30 42 30 20

  33 37 2E 34 20 4D 48 7A 20 57 4C 42

2015-09-29 08:44:39.066642 > HCI Event: Command Complete (0x0e) plen 4

    LE Set Advertising Data (0x08|0x0008) ncmd 1

    status 0x00

sandeepmistry, 2015-10-04:
I'm not sure how to help you next, if nothing is showing on the hcidump for Android, it doesn't look likes it trying to connect.

Btw, what app are you using on the Android side? I highly recommend nRF Master Control panel, I've had no issues connecting to a bleno Linux peripheral using it on my Nexus 7 running Android 5.1.1.

danki, 2015-10-05:
Having this issue too now, have had it before.. like 6+ months ago and i found back then a post where someone said bluez 5.24 wasnt the best one to use on edison with bleno.
As i recall, when i ran Bluez5.18 everything worked smoothly, but now with 5.24 nothing works anymore :/

mikaye, 2015-10-05:
I have confirmed that running nRF Master Control Panel on both Nexus 7,
Nexus 6 and Nexus 5 (all running 5.1.1) have the same problem. iOS on the
other hand, no problem.

Should I try installing an older version of bluez?

As I said, I have it seen this run at least for a short period on Android
before the latest npm install but can't seem to get back their even when
using the older dependencies.

Thanks,
Mike

danki, 2015-10-05:
@mikaye pas weeks ive had loads of problems with bleno, node, npm and node-gyp
Every time i install bleno (as ive tried to re-install it) i get all kinds of errors from node, bleno, xpc-connection, node-gyp and npm.

As of 5.18, ive tried to find that but seems its gone from the web, dunno tho if i could take it directly from bluez web and install from the tar. (ill make a try in a bit).
Few months ago people wanted to use gatttool which seemed to be available in 5.24 but there was just too much issues and i think Alex_T form intel forums even said that 5.18 is best one for edison.

Now, theres tho something else im thinking of (could be the actual issue with bleno).
when ever i clear my edison back to factory, i run all the setups thats needed and install later bleno. In the installation i get errors and warnings. I think it could be nodejs version.
If im correct, bleno is "compiled" on the device  while installed. Tho i cant confirm that theory either cause i tried to update my nodejs to the latest one, same for the npm.. and node-gyp too but then my other nodejs applications stopped working.
The only thing im sure of is that when i used Bluez5.18 and nodejs 0.10.28 and some older bleno, everything worked as expected.

Im currently trying to figure out whats going on and  testing all kinds of theories, when ever i find something, ill uppdate here.
Sidenote... also my rfkill unblock bluetooth stopped working... now after every reboot, i need to unblock that again :/

danki, 2015-10-05:
Here's  output from my edison.
Ive now tried literally everything, flashed eddy, setup, reinstalled, tested multiple version, diabled bluetooth, kille bluetoothd, tested node 0.10.28, tested latest releases...

No idea whats going on, getting these errors on bleno install all the time
http://pastebin.com/0bLLr0WE

danki, 2015-10-05:
Another update...
Previously when node app was running using bleno, in ps i could grep l2cap-ble and hci-ble.
Could that be a bleno issue or more of a bluez issue ?

danki, 2015-10-06:
Ive now updated poky to ww25 (17) and tried with same results.

One thing that i noticed now is that any app using bleno, broadcasts themselves as DUAL (Classic and BLE). Ealier  this year when i had similar issue, i found that unless my phone doesnt see bleno app as BLE, it wont connect.

I have disabled bluetooth, killed the process but still, something is catching my connection attempts so bleno never gets any events.

Really really frustrating not to even have at least some success :/

sandeepmistry, 2015-10-07:
Here's my setup:

```sh
root@edison:~# configure_edison --latest-version
146

root@edison:~/test# uname -a
Linux edison 3.10.17-poky-edison+ #1 SMP PREEMPT Wed Apr 29 03:54:01 CEST 2015 i686 GNU/Linux

root@edison:~/test# node -v
v0.10.35
```

> Previously when node app was running using bleno, in ps i could grep l2cap-ble and hci-ble.

Those are no longer used, node-bluetooth-hci-socket is used instead now.

hcidump from test.js:
```
device: hci0 snap_len: 1500 filter: 0xffffffff
2015-10-07 23:49:29.480250 < HCI Command: Set Event Mask (0x03|0x0001) plen 8
    Mask: 0xfffffbff07f8bf3d
2015-10-07 23:49:29.486881 > HCI Event: Command Complete (0x0e) plen 4
    Set Event Mask (0x03|0x0001) ncmd 1
    status 0x00
2015-10-07 23:49:29.487009 < HCI Command: LE Set Event Mask (0x08|0x0001) plen 8
    mask 0x1f00000000000000 (Reserved)
2015-10-07 23:49:29.487724 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Event Mask (0x08|0x0001) ncmd 1
    status 0x00
2015-10-07 23:49:29.487813 < HCI Command: Read Local Version Information (0x04|0x0001) plen 0
2015-10-07 23:49:29.489511 > HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.0 (0x6) HCI Revision: 0xa6
    LMP Version: 4.0 (0x6) LMP Subversion: 0x410e
    Manufacturer: Broadcom Corporation (15)
2015-10-07 23:49:29.490194 < HCI Command: Read BD ADDR (0x04|0x0009) plen 0
2015-10-07 23:49:29.491539 > HCI Event: Command Complete (0x0e) plen 10
    Read BD ADDR (0x04|0x0009) ncmd 1
    status 0x00 bdaddr 98:4F:EE:03:5A:E9
2015-10-07 23:49:29.518509 < HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1
  00 
2015-10-07 23:49:29.519600 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertise Enable (0x08|0x000a) ncmd 1
    status 0x0c
    Error: Command Disallowed
2015-10-07 23:49:29.519815 < HCI Command: LE Set Advertising Parameters (0x08|0x0006) plen 15
    min 100.000ms, max 100.000ms
    type 0x00 (ADV_IND - Connectable undirected advertising) ownbdaddr 0x00 (Public)
    directbdaddr 0x00 (Public) 00:00:00:00:00:00
    channelmap 0x07 filterpolicy 0x00 (Allow scan from any, connection from any)
2015-10-07 23:49:29.520621 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertising Parameters (0x08|0x0006) ncmd 1
    status 0x00
2015-10-07 23:49:29.535513 < HCI Command: LE Set Scan Response Data (0x08|0x0009) plen 32
  06 05 08 74 65 73 74 00 00 00 00 00 00 00 00 00 00 00 00 00 
  00 00 00 00 00 00 00 00 00 00 00 00 
2015-10-07 23:49:29.536780 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Response Data (0x08|0x0009) ncmd 1
    status 0x00
2015-10-07 23:49:29.536884 < HCI Command: LE Set Advertising Data (0x08|0x0008) plen 32
  15 02 01 06 11 06 F0 FF FF FF FF FF FF FF FF FF FF FF FF FF 
  FF FF 00 00 00 00 00 00 00 00 00 00 
2015-10-07 23:49:29.537693 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertising Data (0x08|0x0008) ncmd 1
    status 0x00
2015-10-07 23:49:29.537781 < HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1
  01 
2015-10-07 23:49:29.538485 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertise Enable (0x08|0x000a) ncmd 1
    status 0x00
2015-10-07 23:49:29.539191 < HCI Command: LE Set Scan Response Data (0x08|0x0009) plen 32
  06 05 08 74 65 73 74 00 00 00 00 00 00 00 00 00 00 00 00 00 
  00 00 00 00 00 00 00 00 00 00 00 00 
2015-10-07 23:49:29.539714 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Response Data (0x08|0x0009) ncmd 1
    status 0x00
2015-10-07 23:49:29.539819 < HCI Command: LE Set Advertising Data (0x08|0x0008) plen 32
  15 02 01 06 11 06 F0 FF FF FF FF FF FF FF FF FF FF FF FF FF 
  FF FF 00 00 00 00 00 00 00 00 00 00 
2015-10-07 23:49:29.541395 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertising Data (0x08|0x0008) ncmd 1
    status 0x00
2015-10-07 23:49:29.541479 < HCI Command: LE Set Advertising Data (0x08|0x0008) plen 32
  1F 02 01 0A 02 0A 04 18 08 42 43 4D 34 33 33 34 30 42 30 20 
  33 37 2E 34 20 4D 48 7A 20 57 4C 42 
2015-10-07 23:49:29.542405 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertising Data (0x08|0x0008) ncmd 1
    status 0x00
```

nRF master control panel:

[image]

Also showing CLASSIC and BLE, flags also show BeErd enable when bleno uses 0x6. I think the kernel or driver is doing something weird .... also having issues connecting from my Nexus 7 running 5.1.1 ...

sandeepmistry, 2015-10-08:
@danki @mikaye 

If I modify: https://github.com/sandeepmistry/bleno/blob/master/lib/hci-socket/gap.js#L152 to:

```javascript
    // this._hci.setScanResponseData(scanData);
    // this._hci.setAdvertisingData(advertisementData);
    this._hci.setAdvertiseEnable(true);
    this._hci.setScanResponseData(scanData);
    this._hci.setAdvertisingData(advertisementData);
```

The Nexus 7 + nRF Master Control Panel is much happier and can connect. The edison peripheral shows up as BLE only and flags GeneralDiscoverable,BrEdrNotSupported as expected as well.

Can you please try? I need to make sure the change doesn't break newer kernels used in the Raspberry Pi etc.

danki, 2015-10-08:
@sandeep thank you, ill confirm that for you in few hrs after some sleep.

danki, 2015-10-08:
@sandeepmistry ive tested it now, and unfortunately its not working as well for me.
Im using Nexus 4 and BLE Scanner (sometimes nrf master as well) tho ive found that blescanner is most reliable to use to detect if connection is possible or not. 
If Ble scanner says DUAL MODE, i will never be able to connect (experience from allmost a years dev on app with bleno) but when it says BLE then its all good for me. Currently i get DUAL

Here is my steps:

rfkill of course and the rest
```
root@seddec:~# node -v
v0.10.38
root@seddec:~# configure_edison --version
159
root@seddec:~# uname -a
Linux seddec 3.10.17-poky-edison+ #1 SMP PREEMPT Fri Jun 19 12:06:40 CEST 2015 i686 GNU/Linux
root@seddec:~# 
root@seddec:~# hciconfig hci -a
hci0:	Type: BR/EDR  Bus: UART
	BD Address: 98:4F:EE:02:E5:69  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING 
	RX bytes:3055 acl:0 sco:0 events:210 errors:0
	TX bytes:2855 acl:0 sco:0 commands:210 errors:0
	Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH SNIFF 
	Link mode: SLAVE ACCEPT 
	Name: 'BCM43340B0 37.4 MHz WLBGA_iTR Intel Edison-0122-N'
	Class: 0x000000
	Service Classes: Unspecified
	Device Class: Miscellaneous, 
	HCI Version: 4.0 (0x6)  Revision: 0xa6
	LMP Version: 4.0 (0x6)  Subversion: 0x410e
	Manufacturer: Broadcom Corporation (15)

root@seddec:~# systemctl status bluetooth
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; disabled)
   Active: inactive (dead)
     Docs: man:bluetoothd(8)

Oct 07 22:55:54 seddec systemd[1]: Starting Bluetooth service...
Oct 07 22:55:54 seddec bluetoothd[229]: Bluetooth daemon 5.24
Oct 07 22:55:54 seddec bluetoothd[229]: Starting SDP server
Oct 07 22:55:54 seddec bluetoothd[229]: Bluetooth management interface 1.3 i...d
Oct 07 22:55:55 seddec systemd[1]: Started Bluetooth service.
Oct 08 01:09:24 seddec systemd[1]: Stopping Bluetooth service...
Oct 08 01:09:24 seddec systemd[1]: Stopped Bluetooth service.
Hint: Some lines were ellipsized, use -l to show in full.
root@seddec:~# ps  | grep bluetooth
  192 root      3228 S    /usr/sbin/bluetooth_rfkill_event
 1609 root      2664 S    grep bluetooth
root@seddec:~# 
´´´

sandeepmistry, 2015-10-08:
@danki how did you get version 159?
```sh
root@edison:~# configure_edison --version
146
root@edison:~# configure_edison  --latest-version
146
root@edison:~# configure_edison  --upgrade       
The latest version is already installed.
```

With version 146, I no longer have to run rfkill ...

```sh
root@edison:~# hciconfig -a
hci0:	Type: BR/EDR  Bus: UART
	BD Address: 98:4F:EE:03:5A:E9  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING 
	RX bytes:1774 acl:33 sco:0 events:107 errors:0
	TX bytes:2195 acl:33 sco:0 commands:83 errors:0
	Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH SNIFF 
	Link mode: SLAVE ACCEPT 
	Name: 'BCM43340B0 37.4 MHz WLBGA_iTR Intel Edison-0122-N'
	Class: 0x000000
	Service Classes: Unspecified
	Device Class: Miscellaneous, 
	HCI Version: 4.0 (0x6)  Revision: 0xa6
	LMP Version: 4.0 (0x6)  Subversion: 0x410e
	Manufacturer: Broadcom Corporation (15)
```

danki, 2015-10-08:
@sandeepmistry i rarely use -latest-version, have seen multiple times it being "not updated", the image is otherwise available on  intels web for downloading, which i did 2 days ago.

I just tested NRF master, and yes it finds the bleno advertising, but is tho unable to connect, android logcat does not reveal anything either why its not connecting.

danki, 2015-10-08:
@sandeepmistry Ive tried now a bit more, with your fix.
Im able to connect now, tho i had to pair phone with edison  in bt settings first. It is now connected, but not getting all the services and characteristics defined in Bleno app, is there any changes in the bleno that needs to be adapted in app ?

mikaye, 2015-10-08:
Here's what I'm seeing with Nexus 6 (5.1.1). My edison configuration is
below the image. The changes I made to gap.js are,

  // this._hci.setScanResponseData(scanData);
  // this._hci.setAdvertisingData(advertisementData);
  this._hci.setAdvertiseEnable(true);
  this._hci.setScanResponseData(scanData);
  this._hci.setAdvertisingData(advertisementData);
}

From nRF Master

​

My edison is pretty much like yours except for node.

>node -v
v0.10.38
After pairing in Bluetooth, I am now able to connect and I see the
following. Of course, I shouldn't have to pair first. And it is missing
services.

​
One last thing I tried. I reverted gap.js and as long as I'm paired, I'm
still able to connect.

--Mike

mikaye, 2015-10-08:
I forgot to mention that in all cases, iOS continues to behave well. I can
connect/disconnect and it shows all services

danki, 2015-10-10:
@sandeepmistry you mentioned  GeneralDiscoverable,BrEdrNotSupported flags, i tried to find info on what exactly those flag but couldnt find  anything "useful", could you explain what these do (in Bleno).

With Edison default BLE (non bleno app) i det BleBrEdrSupported, which kind of feels should be what we look for :P

Yesterday i had everything working, but over the night my edison was powered off and now when i try to run bleno app again and command it with NRFMaster / Blescanner, i get no response from Edison.

sandeepmistry, 2015-10-13:
@mikaye @danki I haven't had a chance to update to the latest firmware but here's my current setup:

 * edison version 146
 * ```bluetoothd``` disabled vis ```systemctl disable bluetooth```
 * _NO_ ```rfkill unblock bluetooth``` run after reboot
 * bleno patch mentioned above

I did not need to pair for it to work with Android.

> @sandeepmistry you mentioned GeneralDiscoverable,BrEdrNotSupported flags, i tried to find info on what exactly those flag but couldnt find anything "useful", could you explain what these do (in Bleno).

> With Edison default BLE (non bleno app) i det BleBrEdrSupported, which kind of feels should be what we look for :P

bleno uses an advertisement type of GeneralDiscoverable (0x00 - [see this line in hci.js](https://github.com/sandeepmistry/bleno/blob/master/lib/hci-socket/hci.js#L202)) and does not support GATT over BrEdr. So if you ever see type "Classic and BLE" instead of BLE only and/or LEandBrEdrCapable in nRF Master Control panel something is wrong. Then I think Android will try to use GATT over BrEdr (non low energy) and have issues connecting.

Btw, @rexstjohn also mentioned a workaround for Android was to turn Bluetooth off and on again, as he and someone from Amazon suspects a bug in the BLE stack. I still need to look into incorporating my suggested patch into master ... maybe it's not needed if it's an Android OS issue ...

mikaye, 2015-10-26:
Hi,

Just wondering if there are any updates planned to address the problems
we've experienced? Otherwise I have two options. The first is to switch to
ios where it works fine and hope for a future updates to android. The
second is to look more into the possibility of using WiFi direct

Thanks,
Mike

danki, 2015-10-26:
@mikaye sorry for hi-jacking you post, the original issue you described, iven seen that behaviour as well in my project. Tho ive found a mini workaround to BCM... advertising.

First, in bleno app, onconnect, add a timeoutlike this:
```javascript
global.g_timeout = setTimeout(function(){
    if(bleno){
        bleno.disconnect();
        bleno.stopAdv(); }
}, 10000); // Give it 10 sec to establish connection
```

Then create a characteristics, where you set read or write properties, which when executed, clears the timeout
```javascript
clearTimeout( global.g_timeout );
```

1) First when you scan for Bleno, lets say you find it as BCM...
2) Make a connection, which eventually will be disconnected, you will need to scan and connect again afterwards, where hopefully your phone sees bleno as advertised name and as BLE (in bluetooth low energy it would be type 2). BCM usually is type 3 and when your phone sees that type, your node app has failed initiate properly i think.
3) When you see that advertising name in scannings, make a connection which hopefully is success, and right after that write/read the characteristics which clears the timeout.

You may also do some pre-initiation in bleno app (ive had to do that few times)
When ever the main.js is started in nodejs, you use exec to dovarious things:

 Run this as first thing in bleno app, just to be sure
```javascript
// You may want to change hciconfig to run up command instead of reset, test around to see what works
exec("rfkill unblick bluetooth && hciconfig hci0 reset && sleep 5 && echo 'OK'",function(err, in, ou){
	if(err){
		console.log(err);

		if (ou.indexOf("No such device") > -1 || ou.indexOf("(110)") > -1) {
			console.log("This is big problem, HCI is completely dead!!!!")
			console.log("Need to reboot the system...")
			setTimeout(function () {
				exec("reboot", null);
				return;
			}, 3000);
		}

		// Use this ONLY if you have a service running, which is setup to restart app if it exits
		setTimeout(function () {
			console.log("Exiting APP!");
			process.exit(2);
		}, 5000);
	}else{
		// Just to clear any cache
		for ( var i in require.cache ){
			if(i.indexOf("/bleno/") > -1 ){
				delete require.cache[i];
			}
		}
		if(in.indexOf("OK") > -1){
        // Initiate Bleno library like
			bleno = require("bleno");
        // add event listeners as no on
		}
	}
});
```

Another tip, set a timeout to exit the app after like 5 seconds.
Now when Bleno initiated and you get event  stateChanged poweredOff or unsupported, exit the app so service can re-start it again. If statechanged gives you poweredOn OR your startAdvertising is triggered, you can kill this timeout so app would not exit when everything is correct.

Ive tested around all kinds of solutions, and (cant share code tho) what i do is that my bleno app runs as service, (boots up with edison). The service is set to restart the app whenever the exit code is non clean. In my Case clean exit is 0, so any other exit are non clean ones.

Also, you could tie bleno lib event stateChanged  poweredOff and unsupported to some kind of behaviour to re-enable your app and its advertising.

I hope this might help you a little and you'll find a good way to get it back working.

PS. these above examples were what ive tried and tested, at some, my current setup is similar, and so far only "way" that assures the right conditions for bleno to work and i have not has any issues so far.

sandeepmistry, 2015-10-27:
@mikaye I'm surprised my suggested patch to gap.js hasn't fixed things.

I plan to have fix for it in eventually. Have you had a chance to recreate your edison setup as per my notes in https://github.com/sandeepmistry/bleno/issues/131#issuecomment-147560198?

sandeepmistry, 2015-11-14:
Based on @danki's comment from https://github.com/sandeepmistry/bleno/issues/84#issuecomment-156472920 it looks like version 146 is the best version to use with bleno.

If you can confirm my suggested patch from https://github.com/sandeepmistry/bleno/issues/131#issuecomment-146371117, I can update bleno with it.

danki, 2015-11-15:
I can confirm #131 to solve, every reinstall of bleno stops working and applying this fix, enables BLE behaviours again on Edison for Android.

sandeepmistry, 2015-11-17:
I've implemented the workaround in master via https://github.com/sandeepmistry/bleno/commit/ceb2151ac9656ce96234e58eb5b880be9d120a38.

You can test it out via ```npm install sandeepmistry/bleno``` before a new version is published to npm.

closed 2015-11-17 by sandeepmistry

mikaye, 2015-11-23:
I'm just able to do some additional testing with the following configuration and everything appears to be working quite well between edison bleno and android 5.1.1 (with or without the patch #131)

edison - 159.devkit
3.10.17-poky-edison+

hci0: hci0:	Type: BR/EDR  Bus: UART
	BD Address: 98:4F:EE:02:E3:AE  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING 
	RX bytes:23856 acl:127 sco:0 events:1193 errors:0
	TX bytes:18556 acl:126 sco:0 commands:1103 errors:0
	Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH SNIFF 
	Link mode: SLAVE ACCEPT 
	Name: 'BCM43340B0 37.4 MHz WLBGA_iTR Intel Edison-0122-N'
	Class: 0x000000
	Service Classes: Unspecified
	Device Class: Miscellaneous, 
	HCI Version: 4.0 (0x6)  Revision: 0xa6
	LMP Version: 4.0 (0x6)  Subversion: 0x410e
	Manufacturer: Broadcom Corporation (15)

I had to do a fresh install of 159.devkit due to some issue I had. Furthermore I installed Node v0.12.7 in order to get Promises.

I also added @danki exec("rfkill unblick bluetooth && hciconfig hci0 reset && sleep 5 && echo 'OK'",function(err, stdin, ou){  to bleno app but have noticed that as long as rfkill unblick bluetooth is run at startup it appears to have the same result. 

with these changes, bleno is running like a champ.

Thanks,
Mike

sandeepmistry, 2015-11-25:
@mikaye great, thanks for the trying it out, the change has also been published to npm.

mikaye, 2015-12-04:
Well, I thought I had this completely resolved. Seems I'm back to square one. I'm using several androids (5.1.1 and 6.0) and randomly, an android sees the advertised device as "Classic and BLE." I've even seen one reporting as BLE and the other as both. Always works fine in iOS though. Certainly seems to be an Android issue. FYI, > npm ls bleno -> bleno@0.3.3. Once I'm seeing "Classic and BLE", no connection can be made.

sandeepmistry, 2015-12-05:
@mikaye please try with firmware version 146 as @danki has suggested.