noble/bleno #133: Battery Service on OSX (Yosemite) broken?

jacobrosenthal, 2015-09-28:
On linux I appear to have a connectable battery service, but not in yosemite.
[image]
[image]

jacobrosenthal, 2015-09-28:
[osx log](https://gist.github.com/jacobrosenthal/0cff7544c46346573e56)
[linux log](https://gist.github.com/jacobrosenthal/b6ab2205da5bad6ad0f5)

jacobrosenthal, 2015-09-28:
Aha, https://gist.github.com/jacobrosenthal/0cff7544c46346573e56#file-gistfile1-txt-L101 is an error, but the example doesnt catch it. (think il PR that)

[Error: failed to set service 180F, UUID not allowed!]

sandeepmistry, 2015-09-29:
@jacobrosenthal unfortunately OS X 10.10 stopped allowing that UUID. It used to work in 10.9, then they blocked it along with some other UUID's.

Does bleno report an error on ```setServices``` callback?

jacobrosenthal, 2015-10-01:
Yes, The error comes through, the example just doesnt catch it. Probably for brevity's sake. So just PR a note on battery example that it doesnt work on osx starting at 10.10?

jacobrosenthal, 2015-10-01:
Do you have a link to anything with those UUIDs? A simple search hasnt pulled them up yet

sandeepmistry, 2015-10-01:
@jacobrosenthal Apple docs a very vague: https://developer.apple.com/library/mac/documentation/CoreBluetooth/Reference/CoreBluetooth_Constants/index.html#//apple_ref/c/tdef/CBError

```
CBErrorUUIDNotAllowed
The specified UUID is not permitted.

Available in OS X v10.9 and later.
```

sandeepmistry, 2015-10-01:
> Yes, The error comes through, the example just doesnt catch it. Probably for brevity's sake. So just PR a note on battery example that it doesnt work on osx starting at 10.10?

That would be great!

The error should be reported in main.js, with something like this:

```javascript
// ...

bleno.on('servicesSet', function(error) {
    console.log(error);
});
```

jacobrosenthal, 2015-10-01:
Do you want to pollute the example with the error logic? I assumed it was
left off brevity.

On Thu, Oct 1, 2015 at 12:32 PM, Sandeep Mistry <notifications@github.com>
wrote:

> bleno.on('servicesSet', function(error) {
>     console.log(error);
> });
>
>

jacobrosenthal, 2015-10-01:
Wow. CBErrorUUIDNotAllowed doesnt even google really. A huge amount of peripherals have a battery as mandatory.. Can you even properly create those without a battery service?

jacobrosenthal, 2015-10-01:
Ran through the current list on bluetooth sig. Heres the ones that error.
180F Battery
1812 Human Interface Device
1805 Current Time Service
180A Device Information
1800 Generic Access
1801 Generic Attribute

sandeepmistry, 2015-10-01:
From the [CBMutableService](https://developer.apple.com/library/mac/documentation/CoreBluetooth/Reference/CBMutableService_Class/index.html#//apple_ref/occ/instm/CBMutableService/initWithType:primary:) docs, only 128-bit UUIDS should be used:

```
UUID	
A 128-bit UUID that identifies the service.
```

However, if I try 0xffff with bleno it's fine ...

closed 2015-10-01 by sandeepmistry