noble/bleno #136: startadvertisingwitheirdata from node-eddystone-beacon gives Command Disallowed on Linux

jacobrosenthal, 2015-09-29:
Moving over from https://github.com/don/node-eddystone-beacon/pull/25 as I would think this would be a bleno issue

Using node-eddystone-beacons's uid/with-tlm.js example which repeatedly calls startadvertisingwitheirdata seems to work in linux.. but gives an error back every time "Command Disallowed"

[hcidump](https://gist.github.com/jacobrosenthal/4959a66a5fa5984113eb)

jacobrosenthal, 2015-09-29:
Seeing the dump, you can see why it works. Its only every 1 error every 5 calls that fails. And since its pi only maybe a resource issue? i tried to set the startadvertisingwitheirdata  (TICK_INTERVAL) from default 100ms to 1000ms and its still 1/5 seemingly consistently..

danki, 2015-10-08:
ive seen that issue too, what i found out to kind of solve temporarily, is to close app that uses bleno (and sometimes hci0 down.
mostly I've had to reboot bleno app to allow changes, hope this helps out a bit

danki, 2015-11-06:
Actually, think i posted wrong workaround... just had this issue

This is what works for me:

```
bleno.on('advertisingStart', function (error) {
    if(error){ console.log("Adv error",error); }
    if (!error) {
        bleno.setServices(services);
    }else{
        bleno.stopAdvertising();
        bleno.startAdvertising(advName, serviceUUIDS);
    }
});
```