noble/bleno #139: Raspberry PI does not allow my Nexus 5 to connect to the test.py BLE

davidmann4, 2015-10-03:
maybe similar to #119 - tried everything what you mentioned there

I would like to connect to my Raspberry PI 2 from my Nexus 5 (Android)

my setup: 

Raspberrypi 2 with 
Bus 001 Device 006: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)

```shell
pi@raspberrypi ~ $ uname -a
Linux raspberrypi 4.1.7-v7+ #817 SMP PREEMPT Sat Sep 19 15:32:00 BST 2015 armv7l GNU/Linux
pi@raspberrypi ~ $ sudo hciconfig
hci0:   Type: BR/EDR  Bus: USB
        BD Address: 00:1A:7D:DA:71:13  ACL MTU: 310:10  SCO MTU: 64:8
        UP RUNNING
        RX bytes:2188 acl:0 sco:0 events:153 errors:0
        TX bytes:3326 acl:0 sco:0 commands:151 errors:0
pi@raspberrypi ~ $ sudo killall bluetoothd
bluetoothd: no process found
```

```shell
pi@raspberrypi ~/bleno/bleno-master $ sudo DEBUG=* node test.js
  bleno platform linux +0ms
  hci setting filter to: 1400000020c10000000000400000 +33ms
  hci set event mask - writing: 01010c08fffffbff07f8bf3d +20ms
  hci set le event mask - writing: 010120081f00000000000000 +3ms
  hci read local version - writing: 01011000 +2ms
  hci read bd addr - writing: 01091000 +1ms
bleno
  hci onSocketData: 040e0401010c00 +38ms
  hci   event type = 4 +2ms
  hci   sub event type = 14 +1ms
  hci           cmd = 3073 +1ms
  hci           status = 0 +1ms
  hci           result =  +0ms
  hci onSocketData: 040e0401012000 +3ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +1ms
  hci           cmd = 8193 +0ms
  hci           status = 0 +0ms
  hci           result =  +1ms
  hci onSocketData: 040e0c0101100006861d060a00861d +0ms
  hci   event type = 4 +1ms
  hci   sub event type = 14 +0ms
  hci           cmd = 4097 +0ms
  hci           status = 0 +0ms
  hci           result = 06861d060a00861d +1ms
  hci set advertise enable - writing: 010a200100 +1ms
  hci set advertisement parameters - writing: 0106200fa000a0000000000000000000000700 +3ms
  hci onSocketData: 040e0a010910001371da7d1a00 +1ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
  hci           cmd = 4105 +1ms
  hci           status = 0 +0ms
  hci           result = 1371da7d1a00 +0ms
  hci address = 00:1a:7d:da:71:13 +4ms
  bleno addressChange 00:1a:7d:da:71:13 +1ms
  hci onSocketData: 040e04010a200c +1ms
  hci   event type = 4 +1ms
  hci   sub event type = 14 +0ms
  hci           cmd = 8202 +0ms
  hci           status = 12 +1ms
  hci           result =  +0ms
  hci onSocketData: 040e0401062000 +0ms
  hci   event type = 4 +1ms
  hci   sub event type = 14 +0ms
  hci           cmd = 8198 +0ms
  hci           status = 0 +1ms
  hci           result =  +0ms
  bleno stateChange poweredOn +0ms
on -> stateChange: poweredOn, address = 00:1a:7d:da:71:13
  gap startAdvertising: name = test, serviceUuids = [
  "fffffffffffffffffffffffffffffff0"
] +9ms
  gap startAdvertisingWithEIRData: advertisement data = 0201061106f0ffffffffffffffffffffffffffffff, scan data = 050874657374 +5ms
  hci set scan response data - writing: 010920200605087465737400000000000000000000000000000000000000000000000000 +3ms
  hci set advertisement data - writing: 01082020150201061106f0ffffffffffffffffffffffffffffff00000000000000000000 +1ms
  hci set advertise enable - writing: 010a200101 +1ms
  hci set scan response data - writing: 010920200605087465737400000000000000000000000000000000000000000000000000 +1ms
  hci set advertisement data - writing: 01082020150201061106f0ffffffffffffffffffffffffffffff00000000000000000000 +1ms
  hci onSocketData: 040e0401092000 +1ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
  hci           cmd = 8201 +1ms
  hci           status = 0 +0ms
  hci           result =  +0ms
  hci onSocketData: 040e0401082000 +1ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
  hci           cmd = 8200 +1ms
  hci           status = 0 +0ms
  hci           result =  +0ms
  hci onSocketData: 040e04010a2000 +1ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
  hci           cmd = 8202 +2ms
  hci           status = 0 +1ms
  hci           result =  +0ms
  bleno advertisingStart: null +1ms
on -> advertisingStart: success
  gatt handles = [
  {},
  {
    "type": "service",
    "uuid": "1800",
    "startHandle": 1,
    "endHandle": 5
  },
  {
    "type": "characteristic",
    "uuid": "2a00",
    "properties": 2,
    "secure": 0,
    "startHandle": 2,
    "valueHandle": 3
  },
  {
    "type": "characteristicValue",
    "handle": 3,
    "value": "Buffer('7261737062657272797069', 'hex')"
  },
  {
    "type": "characteristic",
    "uuid": "2a01",
    "properties": 2,
    "secure": 0,
    "startHandle": 4,
    "valueHandle": 5
  },
  {
    "type": "characteristicValue",
    "handle": 5,
    "value": "Buffer('8000', 'hex')"
  },
  {
    "type": "service",
    "uuid": "1801",
    "startHandle": 6,
    "endHandle": 9
  },
  {
    "type": "characteristic",
    "uuid": "2a05",
    "properties": 32,
    "secure": 0,
    "startHandle": 7,
    "valueHandle": 8
  },
  {
    "type": "characteristicValue",
    "handle": 8,
    "value": "Buffer('00000000', 'hex')"
  },
  {
    "type": "descriptor",
    "handle": 9,
    "uuid": "2902",
    "properties": 14,
    "secure": 0,
    "value": "Buffer('0000', 'hex')"
  },
  {
    "type": "service",
    "uuid": "fffffffffffffffffffffffffffffff0",
    "startHandle": 10,
    "endHandle": 25
  },
  {
    "type": "characteristic",
    "uuid": "fffffffffffffffffffffffffffffff1",
    "properties": 2,
    "secure": 0,
    "startHandle": 11,
    "valueHandle": 12
  },
  {
    "type": "characteristicValue",
    "handle": 12,
    "value": "Buffer('76616c7565', 'hex')"
  },
  {
    "type": "descriptor",
    "handle": 13,
    "uuid": "2901",
    "properties": 2,
    "secure": 0,
    "value": "user description"
  },
  {
    "type": "characteristic",
    "uuid": "fffffffffffffffffffffffffffffff2",
    "properties": 2,
    "secure": 0,
    "startHandle": 14,
    "valueHandle": 15
  },
  {
    "type": "characteristicValue",
    "handle": 15,
    "value": null
  },
  {
    "type": "characteristic",
    "uuid": "fffffffffffffffffffffffffffffff3",
    "properties": 2,
    "secure": 0,
    "startHandle": 16,
    "valueHandle": 17
  },
  {
    "type": "characteristicValue",
    "handle": 17,
    "value": null
  },
  {
    "type": "characteristic",
    "uuid": "fffffffffffffffffffffffffffffff4",
    "properties": 12,
    "secure": 0,
    "startHandle": 18,
    "valueHandle": 19
  },
  {
    "type": "characteristicValue",
    "handle": 19,
    "value": null
  },
  {
    "type": "characteristic",
    "uuid": "fffffffffffffffffffffffffffffff5",
    "properties": 16,
    "secure": 0,
    "startHandle": 20,
    "valueHandle": 21
  },
  {
    "type": "characteristicValue",
    "handle": 21,
    "value": null
  },
  {
    "type": "descriptor",
    "handle": 22,
    "uuid": "2902",
    "properties": 14,
    "secure": 0,
    "value": "Buffer('0000', 'hex')"
  },
  {
    "type": "characteristic",
    "uuid": "fffffffffffffffffffffffffffffff6",
    "properties": 32,
    "secure": 0,
    "startHandle": 23,
    "valueHandle": 24
  },
  {
    "type": "characteristicValue",
    "handle": 24,
    "value": null
  },
  {
    "type": "descriptor",
    "handle": 25,
    "uuid": "2902",
    "properties": 14,
    "secure": 0,
    "value": "Buffer('0000', 'hex')"
  }
] +41ms
  bleno servicesSet +3ms
on -> servicesSet: success
  hci onSocketData: 040e0401092000 +2ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
  hci           cmd = 8201 +1ms
  hci           status = 0 +0ms
  hci           result =  +0ms
  hci onSocketData: 040e0401082000 +1ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
  hci           cmd = 8200 +0ms
  hci           status = 0 +1ms
  hci           result =  +0ms
```

when I try to connect I hang at connecting... 
[image]

thx for your effort - I already tried bluepy and pyBluez - they had no  good BLE support ... your lib looks promissing :+1:

sandeepmistry, 2015-10-03:
@davidmann4 interesting, I don't have any issues with my Nexus 7 running 5.1.1with nRF master control panel and Raspberry Pi kernel 4.0.9+.

Could you also please provide an hcidump capture? ```sudo hcidump -t -x```

One thing to try is changing the GAP flags, maybe try 0x05 instead of 0x06: https://github.com/sandeepmistry/bleno/blob/master/lib/hci-socket/gap.js#L120

davidmann4, 2015-10-03:
BLUEZ version - maybe also interessting
```shell
pi@raspberrypi ~ $ sudo apt-cache policy bluez
bluez:
  Installed: 5.23-2+b1
  Candidate: 5.23-2+b1
  Version table:
 *** 5.23-2+b1 0
        500 http://mirrordirector.raspbian.org/raspbian/ jessie/main armhf Packages
        100 /var/lib/dpkg/status
```

```shell
pi@raspberrypi ~ $ sudo hcidump -t -x
HCI sniffer - Bluetooth packet analyzer ver 5.23
device: hci0 snap_len: 1500 filter: 0xffffffff
2015-10-03 21:25:56.957443 < HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1
  00
2015-10-03 21:25:56.959661 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertise Enable (0x08|0x000a) ncmd 1
    status 0x00
2015-10-03 21:26:02.464363 < HCI Command: Set Event Mask (0x03|0x0001) plen 8
    Mask: 0xfffffbff07f8bf3d
2015-10-03 21:26:02.465440 > HCI Event: Command Complete (0x0e) plen 4
    Set Event Mask (0x03|0x0001) ncmd 1
    status 0x00
2015-10-03 21:26:02.465540 < HCI Command: LE Set Event Mask (0x08|0x0001) plen 8
    mask 0x1f00000000000000 (Reserved)
2015-10-03 21:26:02.467443 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Event Mask (0x08|0x0001) ncmd 1
    status 0x00
2015-10-03 21:26:02.467526 < HCI Command: Read Local Version Information (0x04|0x0001) plen 0
2015-10-03 21:26:02.480454 > HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.0 (0x6) HCI Revision: 0x1d86
    LMP Version: 4.0 (0x6) LMP Subversion: 0x1d86
    Manufacturer: Cambridge Silicon Radio (10)
2015-10-03 21:26:02.480647 < HCI Command: Read BD ADDR (0x04|0x0009) plen 0
2015-10-03 21:26:02.482435 > HCI Event: Command Complete (0x0e) plen 10
    Read BD ADDR (0x04|0x0009) ncmd 1
    status 0x00 bdaddr 00:1A:7D:DA:71:13
2015-10-03 21:26:02.519313 < HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1
  00
2015-10-03 21:26:02.520445 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertise Enable (0x08|0x000a) ncmd 1
    status 0x0c
    Error: Command Disallowed
2015-10-03 21:26:02.521130 < HCI Command: LE Set Advertising Parameters (0x08|0x0006) plen 15
    min 100.000ms, max 100.000ms
    type 0x00 (ADV_IND - Connectable undirected advertising) ownbdaddr 0x00 (Public)
    directbdaddr 0x00 (Public) 00:00:00:00:00:00
    channelmap 0x07 filterpolicy 0x00 (Allow scan from any, connection from any)
2015-10-03 21:26:02.522484 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertising Parameters (0x08|0x0006) ncmd 1
    status 0x00
2015-10-03 21:26:02.542100 < HCI Command: LE Set Scan Response Data (0x08|0x0009) plen 32
  06 05 08 74 65 73 74 00 00 00 00 00 00 00 00 00 00 00 00 00
  00 00 00 00 00 00 00 00 00 00 00 00
2015-10-03 21:26:02.543459 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Response Data (0x08|0x0009) ncmd 1
    status 0x00
2015-10-03 21:26:02.543635 < HCI Command: LE Set Advertising Data (0x08|0x0008) plen 32
  15 02 01 06 11 06 F0 FF FF FF FF FF FF FF FF FF FF FF FF FF
  FF FF 00 00 00 00 00 00 00 00 00 00
2015-10-03 21:26:02.545459 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertising Data (0x08|0x0008) ncmd 1
    status 0x00
2015-10-03 21:26:02.545555 < HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1
  01
2015-10-03 21:26:02.547437 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertise Enable (0x08|0x000a) ncmd 1
    status 0x00
2015-10-03 21:26:02.547512 < HCI Command: LE Set Scan Response Data (0x08|0x0009) plen 32
  06 05 08 74 65 73 74 00 00 00 00 00 00 00 00 00 00 00 00 00
  00 00 00 00 00 00 00 00 00 00 00 00
2015-10-03 21:26:02.549446 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Response Data (0x08|0x0009) ncmd 1
    status 0x00
2015-10-03 21:26:02.549565 < HCI Command: LE Set Advertising Data (0x08|0x0008) plen 32
  15 02 01 06 11 06 F0 FF FF FF FF FF FF FF FF FF FF FF FF FF
  FF FF 00 00 00 00 00 00 00 00 00 00
2015-10-03 21:26:02.551443 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertising Data (0x08|0x0008) ncmd 1
    status 0x00
```

hmm also funny is that he does not output anything when I start the script first and then run the dump command and take my nexus5 and connect - seems like I get no connection at all!

```console
Cpi@raspberrypi ~ $ sudo hcidump -t -x
HCI sniffer - Bluetooth packet analyzer ver 5.23
device: hci0 snap_len: 1500 filter: 0xffffffff
```

changed gap flags without any success - hmm maybe change BLUEZ ?

do you have any good guide how to switch it to another version?

davidmann4, 2015-10-03:
solved the issue - i was still paired with the MAC of my dongl from pybluez 

thank you for your input - have to read about the GAP stuff ... what did it changed?(pretty sure it was not the issue anyways ... )

closed 2015-10-03 by davidmann4

sandeepmistry, 2015-10-04:
@davidmann4 cool, thanks for the follow up.

I don't think you need the GAP flag change I suggested, you can read up on in in the Core Specification ...

davidmann4, 2015-10-04:
do you use some kind of stratup script on your pi to disable bluetoothd?

sandeepmistry, 2015-10-04:
@davidmann4 I just run the stock BlueZ 4.99 bluetoothd, then there's no need to disable it.

Some options to disable:
1) ```sudo chmod -x /usr/sbin/bluetoothd```
2) ```update-rc.d bluetooth disable```

P.S.: It's best to ask general questions like this on Gitter: https://gitter.im/sandeepmistry/bleno