noble/bleno #140: Cannot find module '../build/Release/binding.node'

aaronjessen, 2015-10-04:
I'm encountering an issue when trying to run a NodeJS app from an Intel Edison Arduino board. The error I receive is `Cannot find module '../build/Release/binding.node`, stemming from the `bluetooth-hci-socket` library, which I see that you also maintain. The `binding.node` file in question does not appear under the Release directory. Can I do anything to create the file, or is this a bug? 

ERROR: module.js:340                                                                                                                                                                                                                                                 
ERROR:     throw err;                                                                                                                                                                                                                                                
ERROR:           ^                                                                                                                                                                                                                                                   
ERROR: Error: Cannot find module '../build/Release/binding.node'                                                                                                                                                                                                     
    at Function.Module._resolveFilename (module.js:338:15)                                                                                                                                                                                                           
    at Function.Module._load (module.js:280:25)                                                                                                                                                                                                                      
    at Module.require (module.js:364:17)                                                                                                                                                                                                                             
    at require (module.js:380:17)                                                                                                                                                                                                                                    
    at Object.<anonymous> (/home/root/.node_app_slot/node_modules/bleno/node_modules/bluetooth-hci-socket/lib/native.js:4:15)                                                                                                                                        
    at Module._compile (module.js:456:26)                                                                                                                                                                                                                            
    at Object.Module._extensions..js (module.js:474:10)                                                                                                                                                                                                              
    at Module.load (module.js:356:32)

aaronjessen, 2015-10-04:
Okay, I'm fairly new to Node, but I found out that I could build the repo using `node-gyp rebuild` **from within the Edison** (running `node-gyp rebuild` on Windows created the binding.node file just fine, but it wouldn't run on Linux, due to an issue with ELF headers).

Would it be possible for you to include the Linux-compatible binding.node file in future releases?

sandeepmistry, 2015-10-04:
@aaronjessen no plans in include the built bindings.node in the future. To much variation, each CPU architecture (Intel, ARM, 32-bit vs 64-bit) per OS per Node.js version.

Other modules use node-pre-gyp to have builds that are downloadable: https://www.npmjs.com/package/node-pre-gyp - but I haven't had time to look at the infrastructure needed to set this up.

closed 2015-10-04 by sandeepmistry

PierrickPetain, 2016-02-19:
Hi, 
I'm sorry, I tried to install noble today, and I have the same issue (on Raspbian).
When I try to run "var noble = require('noble');" I get
ERROR: Error: Cannot find module '../build/Release/binding.node'

at Function.Module._resolveFilename (module.js:338:15)

at Function.Module._load (module.js:280:25)

at Module.require (module.js:364:17)

at require (module.js:380:17)

at Object. (/home/root/.node_app_slot/node_modules/bleno/node_modules/bluetooth-hci-socket/lib/native.js:4:15)

at Module._compile (module.js:456:26)

at Object.Module._extensions..js (module.js:474:10)

at Module.load (module.js:356:32) 
  I don't understand how to fix it.
Kind regards

sandeepmistry, 2016-02-21:
@PierrickPetain  you have posted your issue in the wrong place. This issue tracker is for bleno .... not noble.

PierrickPetain, 2016-02-21:
Soory.
 But I still have the issue with noble ;-)

WayneKeenan, 2016-03-12:
I had this error and managed to make it go away by deleting the node_modules folder and repeating the npm install...

I think what happened was I had a 'messy' build tree. 
During the npm install of bleno I added each of these packages one by one:
libhidapi-dev  
libusb-dev  
libusb-1.0-0.dev 

and tried instilling after each, I think things just got messy.  purging the node_modules folder did the trick

BTW, this and noble is such great work Sandeep, thank you so much!

dothebart, 2016-04-28:
got it fixed by:
/usr/lib# ln -s nodejs node

for some reason `bluetooth-hci-socket` installs to `/usr/lib/nodejs`, but as an strace shows its being searched in `/usr/lib/node/` only.

nikoc03, 2017-06-12:
@dothebart I try the command:
ln -s nodejs node in mac os x but I get the following:
ln: node: Operation not permitted

Is there any way to achieve this result in mac since I get the same error? Thanks!

dothebart, 2017-06-12:
It seems that you don't have access to a directory where it tries to add the symlink to - try to run it using sudo ?

nikoc03, 2017-06-13:
achieved it somehow but the problem still exists:/