noble/bleno #141: late disconnect event

danki, 2015-10-09:
Intel edison version 159 and bleno 0.3.0...
Getting disconnect event  pretty late. hcidump  shows disconnect immediately after disconnect but from bleno it can take up to minutes. Not sure if that is expected or a minor bug.

sandeepmistry, 2015-10-10:
@danki can you please provide the output from following commands running it 2 terminal sessions.

1. ```hcidump -t -x```
1. ```DEBUG=* node <file>.js```

It is not expected at all, but might be related to the issues you are seeing in #131.

sandeepmistry, 2015-10-28:
@danki any updates?

danki, 2015-11-06:
sorry missed out this one, I'll get back to you as soon I see this behaviour again.

sandeepmistry, 2015-11-07:
ok, let's close it for now then and re-open if you see it again.

closed 2015-11-07 by sandeepmistry

danki, 2015-11-08:
Yeah.
Was working with the app yesterday again for few hours and didnt really encouter this again, except for when reading characteristics, each read attempt  was triggered every 20 seconds. So 278 bytes with 22byte transmission took allmost 4 min to complete.
Tried that same code on another Edison and it worked there. Could be 1 of my edisons thats messed in some way.

danki, 2015-11-16:
Okay i was able to simulate this again, basically what i did was:

1) Opened bleno app on Edison + hcidump.. after few sec all was configured and fine
2) opened nRF master on phone (Nexus 4, A5.1.0) 
3) scanned for bleno app, found and connected
4) as soon i say my services, i hit the **disconnect** in nrfmaster

Now neither debug=* nor hcidump update. Even when closing nrf master entirely on phone, did hcidump nor debug reveal, so unfortunately i have nothing to share for debugging. :P 

I would guess BLE hardware itself on Edison doesnt seen disconnect to happen.

sandeepmistry, 2015-11-16:
@danki alright, it could also be an issue with Android, right? What happens if you turn bluetooth off on the Android device.

Anyways, it looks like this is out of scope of bleno ...

danki, 2015-11-17:
Yeah this seems to be BLE behavior between adapters. I will try out B restart on phone, will report back on that