noble/bleno #142: Transmitting json as buffer get scrambled

danki, 2015-10-14:
No idea if this is a bug or not, ill lift it to attention anyway in hope for some clarification.

Im doing a simple test with subscriptions, trying to send a JSON over ble...

Here is what i got:

a characteristic for subscription:
```
SimpleCharacteristic.events = new Characteristic({
    uuid:"FE81",
    properties: ["notify"],
    onReadRequest: function (offset, callback) {},
    onWriteRequest: function (data, offset, wor, callback) {},
    onSubscribe: function (mv, callback) {
        console.log(this.uuid, "subscribe to events");
        global.notify_events = callback;
    },
    onUnsubscribe: function () {
        console.log(this.uuid, "un-subscribe to events");
        global.notify_events = null;
    },
    onNotify: function () {}
});
´´´

A interval to notify:
```
var bff; // Buffer
var obj = {}; // Object to convert to JSON
var nn = 0; // simple counter
setInterval(function(){
    if(global.notify_events){
        obj[nn] = nn; // Will be something like: {"0":0, "1":1}
        var json = JSON.stringify(obj);
        bff = new Buffer(json.length);
        bff.write(json)
        transmitToBLE(global.notify_events, bff);
        nn++;
        // Example output: 13 '{"0":0,"1":1}'
        console.log(bff.length, bff.toString()); 
    }
},4000);
´´´

This is a function in nodejs that splits buffers over 31 bytes into packets, and sends those packets:
```
function transmitToBLE(notificationCallback, buff){
    if(notificationCallback) {
        if (buff.length > 31) {

            var packs = Math.ceil(buff.length / 31);
            for (var i = 0; i < packs; i++) {
                var newBufLen = buff.length > (i * 31) + 31 ? 31 : buff.length - i * 31;
                var nbuf = new Buffer(newBufLen);
                buff.copy(nbuf, 0, i * 31, ((i * 31) + 31));
                notificationCallback(nbuf);
            }
        } else {
            notificationCallback(buff);
        }
    }
}
´´´
NOTE: i also tested to use subscriptions callback directly ( instead of splitting the buffer into packets as in above transmitToBLE function AND ive tested with NRF master, i get same issues...

Now here are some outputs, this one from nodejs / bleno, it seems correct
```
4008 subscribe to events
7 '{"0":0}'
13 '{"0":0,"1":1}'
4011 write
19 '{"0":0,"1":1,"2":2}'
25 '{"0":0,"1":1,"2":2,"3":3}'
31 '{"0":0,"1":1,"2":2,"3":3,"4":4}'
4011 write
37 '{"0":0,"1":1,"2":2,"3":3,"4":4,"5":5}'
43 '{"0":0,"1":1,"2":2,"3":3,"4":4,"5":5,"6":6}'
49 '{"0":0,"1":1,"2":2,"3":3,"4":4,"5":5,"6":6,"7":7}'
4011 write
55 '{"0":0,"1":1,"2":2,"3":3,"4":4,"5":5,"6":6,"7":7,"8":8}'
61 '{"0":0,"1":1,"2":2,"3":3,"4":4,"5":5,"6":6,"7":7,"8":8,"9":9}'
69 '{"0":0,"1":1,"2":2,"3":3,"4":4,"5":5,"6":6,"7":7,"8":8,"9":9,"10":10}'
´´´

Heres a output from my app (is not native android app, im using Adobe AIR and BLE ANE, write and read work as expected)
```
[trace] Subscribe; true
[trace] OKAY try-catch 1:  {"0":0}
[trace] OKAY try-catch 2:  {"0":0}
[trace] OKAY try-catch 1:  {"0":0,"1":1}
[trace] OKAY try-catch 2:  {"0":0,"1":1}
[trace] OKAY try-catch 1:  {"0":0,"1":1,"2":2}
[trace] OKAY try-catch 2:  {"0":0,"1":1,"2":2}
[trace] FAIL try-catch 1:  {"0":0,"1":1,"2":2,"
[trace] FAIL try-catch 2:  {"0":0,"1":1,"2":2,"
[trace] FAIL try-catch 1:  {"0":0,"1":1,"2":2,"{"0":0,"1":1,"2":2,"
[trace] FAIL try-catch 2:  {"0":0,"1":1,"2":2,"
[trace] Char write: true
[trace] FAIL try-catch 1:  {"0":0,"1":1,"2":2,"{"0":0,"1":1,"2":2,"{"0":0,"1":1,"2":2,"
[trace] FAIL try-catch 2:  {"0":0,"1":1,"2":2,"
[trace] FAIL try-catch 1:  {"0":0,"1":1,"2":2,"{"0":0,"1":1,"2":2,"{"0":0,"1":1,"2":2,""5":5}
[trace] FAIL try-catch 2:  "5":5}
[trace] FAIL try-catch 1:  {"0":0,"1":1,"2":2,"{"0":0,"1":1,"2":2,"{"0":0,"1":1,"2":2,""5":5}{"0":0,"1":1,"2":2,"
[trace] FAIL try-catch 2:  {"0":0,"1":1,"2":2,"
[trace] FAIL try-catch 1:  {"0":0,"1":1,"2":2,"{"0":0,"1":1,"2":2,"{"0":0,"1":1,"2":2,""5":5}{"0":0,"1":1,"2":2,""5":5,"6":6}
[trace] FAIL try-catch 2:  "5":5,"6":6}
[trace] FAIL try-catch 1:  {"0":0,"1":1,"2":2,"{"0":0,"1":1,"2":2,"{"0":0,"1":1,"2":2,""5":5}{"0":0,"1":1,"2":2,""5":5,"6":6}{"0":0,"1":1,"2":2,"
[trace] FAIL try-catch 2:  {"0":0,"1":1,"2":2,"
[trace] FAIL try-catch 1:  {"0":0,"1":1,"2":2,"{"0":0,"1":1,"2":2,"{"0":0,"1":1,"2":2,""5":5}{"0":0,"1":1,"2":2,""5":5,"6":6}{"0":0,"1":1,"2":2,""5":5,"6":6,"7":7}
[trace] FAIL try-catch 2:  "5":5,"6":6,"7":7}
´´´

Right after 2:nd  object in the json, everything gets messed up. 
I see the read capability in bleno stacks up same JSON object and sends as "packets" ( bleno  sends splitted json and in ANE i get the final object, not as with notify, i get splits of that json).

im not really sure whats going on here, maybe im getting the idea wrong here... that subscribe callback should not be used to burst data into ? If thats the case, i would guess subscribe callback has limitation of buffer-size, any idea how large the buffer can be ? Also worth to mention, my BLE ANE (that is android native) is not storing any data anywhere, it dispatches data to be right as it is received.

It seems the json string always gets messed somewhere around 3:rd-4:th index. Even if i replace values with "Hello" instead of numbers. Something like:
```
1) {"0":"Hello"}
2) {"0":"Hello", "1":"Hello" }
3) {"0":"Hello", "1":"Hello", "2":"Hello" }
4) {"0":"Hello", "1":"Hello", "2":"Hello", "3":"Hel
5) {"0":"Hello", "1":"Hello", "2":"Hello", "3":"Hel{"0":"Hello", "1":"Hello", "2":"Hello", "3":"Hel""4":"Hello"
´´´

sandeepmistry, 2015-10-26:
@danki can we close this?

danki, 2015-11-06:
@sandeepmistry yes

closed 2015-11-06 by danki