noble/bleno #143: Characteristics onReadRequest max 594 bytes ?

danki, 2015-10-18:
Is it limited to 594 bytes only ?
When transmitting a buffer with 600+ bytes in readReq callback, i always end up with offset as 594 and it never goes above it.

In around 50 tests, ive seen offset to increase to the actual buffer size 2 times. But rest of the tests, offset always stops at 594.

Just a sample code:
```javascript
var str = "";
for ( var i = 0; i < 999; i++){
    str += ""+i;
}
var appStatusBuff=new Buffer(str);
CustomChar.status = new Characteristic({
    uuid: "1010dbd1-0000-1000-8000-00805f9b34fb",
    properties: ["read"],
    onReadRequest: function (offset, callback) {
        console.log(this.uuid,"read",
            "\n  Offset:",offset,
            "\n  Total :",packageInfo.length,
            "\n  Remain:",packageInfo.length-offset);

            sendData(offset,appStatusBuff.slice(offset,(offset+31),callback);
    }});
```

This gives this output:
```
  Offset: 550 
  Total : 2887 
  Remain: 2337
  gatt read response: 0d32323032323132323232323332323432323532323632 +1ms
  gatt send: 0d32323032323132323232323332323432323532323632 +0ms
  hci write acl data pkt - writing: 0240001b00170004000d32323032323132323232323332323432323532323632 +1ms
  hci onSocketData: 0240200900050004000c4f003c02 +93ms
  hci 	event type = 2 +0ms
  hci 		cid = 4 +1ms
  hci 		handle = 64 +0ms
  hci 		data = 0c4f003c02 +1ms
  gatt handing request: 0c4f003c02 +0ms

  Offset: 572 
  Total : 2887 
  Remain: 2315
  gatt read response: 0d32373232383232393233303233313233323233333233 +2ms
  gatt send: 0d32373232383232393233303233313233323233333233 +0ms
  hci write acl data pkt - writing: 0240001b00170004000d32373232383232393233303233313233323233333233 +0ms
  hci onSocketData: 0240200900050004000c4f005202 +93ms
  hci 	event type = 2 +1ms
  hci 		cid = 4 +0ms
  hci 		handle = 64 +1ms
  hci 		data = 0c4f005202 +0ms
  gatt handing request: 0c4f005202 +1ms

  Offset: 594 
  Total : 2887 
  Remain: 2293
  gatt read response: 0d34323335323336323337323338323339323430323431 +1ms
  gatt send: 0d34323335323336323337323338323339323430323431 +0ms
  hci write acl data pkt - writing: 0240001b00170004000d34323335323336323337323338323339323430323431 +1ms
```

danki, 2015-10-19:
Digging around, seems read should be maximum 512 bytes, why bleno succeeds with 594 is still unknown.
BUT, even examples transmit max 594 bytes and i still have no clue why i at 2 times were able to transmit more than 600bytes with read.

So i would guess everything is as expected, hopefully we get more clarification on this in future.

This ticket can be closed

sandeepmistry, 2015-10-27:
@danki probably related to the maximum GATT attribute length of 512, closing as you suggested.

closed 2015-10-27 by sandeepmistry