noble/bleno #144: Connection issue on RPI

nsbm60, 2015-11-04:
I'm running a simple test to advertise as an iBeacon.  

```var bleno = require('bleno');
bleno.on('stateChange', function(state) {
    if (state === 'poweredOn') {
        bleno.startAdvertisingIBeacon('e2c56db5dffb48d2b060d0xxxxxxxxxx', 0, 0, -59);
    }
});
```

It works fine on OS/X, but on a Raspberry pi, when I attempt to connect using LightBlue, it crashes with the following error -

```home/pi/node_modules/bleno/lib/hci-socket/gatt.js:509
      var handle = this._handles[i];
                                ^
TypeError: Cannot read property '1' of undefined
    at Gatt.handleReadByGroupRequest (/home/pi/node_modules/bleno/lib/hci-socket/gatt.js:509:33)
    at Gatt.handleRequest (/home/pi/node_modules/bleno/lib/hci-socket/gatt.js:320:23)
    at Gatt.onAclStreamData (/home/pi/node_modules/bleno/lib/hci-socket/gatt.js:267:8)
    at emit (events.js:117:20)
    at AclStream.push (/home/pi/node_modules/bleno/lib/hci-socket/acl-stream.js:26:10)
    at BlenoBindings.onAclDataPkt (/home/pi/node_modules/bleno/lib/hci-socket/bindings.js:190:21)
    at emit (events.js:106:17)
    at Hci.onSocketData (/home/pi/node_modules/bleno/lib/hci-socket/hci.js:389:14)
    at BluetoothHciSocket.emit (events.js:95:17)
```

Any thoughts as to what might be causing this?  I am running with bluez 5.36.

danki, 2015-11-06:
easy workaround for this is to capture entire block of that handle array inside a if-case.

had same problem on Edison and after setting ifcase, errors disappeared.
note this handle array is used in multiple places, so add one of case and run again to get next error to capture in ifcase.

sandeepmistry, 2015-11-07:
@nsbm60 resolved via https://github.com/sandeepmistry/bleno/commit/bed63b0d6ed89689959f7e25dd0b3314ddfafaae

closed 2015-11-07 by sandeepmistry

hungtien, 2015-12-24:
Hi Sir
It's still the same error when i use noble module, trying to use LightBlue to connect the ble gondle in raspberry pi. Is there anything can solve this problem? 
Thank you very much.

sandeepmistry, 2016-01-03:
@hungtien please open a new issue with code snippet and debug logs in here: https://github.com/sandeepmistry/noble/issues

This issue is probably unrelated, as it's for bleno and not noble.