noble/bleno #145: more than 4 Characteristic with iOS and Android problem

johnyso, 2015-11-05:
Hey, 
first of all thanks for this great work! 

I'm struggling with an characteristic issue. I'm not sure if this issue is on bleno or Android side. 
I added 3 characteristics to the "echo" example (altogether 4 characteristics). 
After running the service i used a samsung galaxy tab with nRF Control to connect to the service which works fine. Then i used an iPhone 4s with LightBlue to Connect to the service. This was also successful. But when i tried to connect the android tab again, I got a GATT-Service error 129.
If I use 3 characteristics everything works fine. furthermore using only Android devices it works also with more than 4 characteristics.
After restarting the server the android device can read the services again (until the iPhone comes around). 

Maybe you have an idea, thx.

danki, 2015-11-06:
have had same issue before.
I'm my case turned out I had  badly formed 128bit uuid and silent errors in that characteristics.
also make sure you set advertising uuids and characteristics are defined in that service.
might be helpful to clear bleno cache as well before initiating bleno library

johnyso, 2015-11-06:
Thanks for your answer. 
I checked the uuid's and i think they are correct, but the Issue will not disappear. 
How can i clean the bleno cache (var bleno = require.resolve('bleno');delete require.cache[bleno];)?

Another interesting point: if i use 16bit uuid's like "cc00" for service and characteristic it can handle 3 characteristics without error. An 128bit uuids works only with 1 characteristic.

Above the simple example code: 

var bleno = require('bleno');
var BlenoPrimaryService = bleno.PrimaryService;

var Characteristic = bleno.Characteristic;
console.log('bleno - echo');

bleno.on('stateChange', function(state) {
  console.log('on -> stateChange: ' + state);

  if (state === 'poweredOn') {
	bleno.startAdvertising('echo', ['bae27a80846511e58bcffeff819cdc9f']);
  } else {
    bleno.stopAdvertising();
  }
});

bleno.on('advertisingStart', function(error) {
  console.log('on -> advertisingStart: ' + (error ? 'error ' + error : 'success'));

  if (!error) {
    bleno.setServices([
      new BlenoPrimaryService({
		uuid: 'bae27a80846511e58bcffeff819cdc9f',
        characteristics: [
		new Characteristic({
			uuid: 'bae27dc8846511e58bcffeff819cdc9f', 
			properties: [ 'write' ],
			value: null,
		}),
		new Characteristic({
			uuid: 'bae27f4e846511e58bcffeff819cdc9f', 
			properties: [ 'write' ],
			value: null,
		})
        ]
      })
    ]);
  }
});

danki, 2015-11-07:
Ive had the exact same issues as you bring up, ive been working a long time on my project and to be honest i have forgotten many workaround  to issues. Ive also lost track of which "solved" which problems. Kind of a tight deadline.

For clearing cache i use this one:
````
for ( var i in require.cache ){
                if(i.indexOf("/bleno/") > -1 ){
                    delete require.cache[i];
                }
            }
```
I had issues with 128 uuids (was at the time lack of knowledge) and went for 16 uuids which ive had pretty much no problems since end of 2014.
Ive seen few having issues with broadcasting more than 2-4 characteristics, not really sure what is going on and how to fix. Im always trying  woraround like reseting hci0, pulling it up / down, doing various commands in blutoothctl before bleno is being used on device.
Also restart of BLE on remote device (android in my case) is a mandatory move due cache.

A tip, what im using to "reset" bleno-app is when ever you disconnect from bleno app, restart the blenoapp enirely. In my case it has solved few problems and also makes sure  next time a device connects to bleno app, it gets successful connection since ive had issues with "re-connecting" to same intance/process of bleno app

sandeepmistry, 2015-11-07:
@johnyso can you please provide an example app and HCI dump capture from start to end when you encounter the problem?
```
sudo hcidump -t -x
```

johnyso, 2015-11-09:
@danki thanks for the tips, trying out your hints right now
@sandeepmistry here is the bleno-app and the hcidump when the problem appears (connect an Android device after iOS Device was connected)

The example bleno-app:
```
var bleno = require('bleno');
var BlenoPrimaryService = bleno.PrimaryService;

var Characteristic = bleno.Characteristic;
console.log('bleno - echo');

bleno.on('stateChange', function(state) {
  console.log('on -> stateChange: ' + state);

  if (state === 'poweredOn') {
        bleno.startAdvertising('echo', ['bae27a80846511e58bcffeff819cdc9f']);
  } else {
    bleno.stopAdvertising();
  }
});

bleno.on('advertisingStart', function(error) {
  console.log('on -> advertisingStart: ' + (error ? 'error ' + error : 'success'));

  if (!error) {
    bleno.setServices([
      new BlenoPrimaryService({
                uuid: 'bae27a80846511e58bcffeff819cdc9f',
        characteristics: [
                new Characteristic({
                        uuid: 'bae27dc8846511e58bcffeff819cdc9f', 
                        properties: [ 'write' ],
                        value: null,
                }), 
                new Characteristic({
                        uuid: 'bae27f4e846511e58bcffeff819cdc9f',
                        properties: [ 'write' ],
                        value: null,
                })  
        ]   
      })  
    ]); 
  }
});
```
the hcidump:
```
HCI sniffer - Bluetooth packet analyzer ver 2.4
device: hci0 snap_len: 1028 filter: 0xffffffff
2015-11-09 08:09:03.190434 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 67 reason 0x13
    Reason: Remote User Terminated Connection
2015-11-09 08:09:03.258939 < HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1
  01 
2015-11-09 08:09:03.260504 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertise Enable (0x08|0x000a) ncmd 1
    status 0x00
2015-11-09 08:09:03.343436 > HCI Event: LE Meta Event (0x3e) plen 19
    LE Connection Complete
      status 0x00 handle 66, role slave
      bdaddr 30:C7:AE:01:12:25 (Public)
2015-11-09 08:09:03.343901 < ACL data: handle 66 flags 0x00 dlen 16
    L2CAP(d): cid 0x0005 len 12 [psm 0]
      12 01 08 00 28 00 38 00 00 00 D0 07 
2015-11-09 08:09:03.433472 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 66 packets 1
2015-11-09 08:09:03.483105 > ACL data: handle 66 flags 0x02 dlen 10
    L2CAP(d): cid 0x0005 len 6 [psm 0]
      13 01 02 00 00 00 
2015-11-09 08:09:03.483234 > ACL data: handle 66 flags 0x02 dlen 11
[... 98 lines cut ...]
    handle 66 packets 1
2015-11-09 08:09:04.457103 > ACL data: handle 66 flags 0x02 dlen 11
    ATT: Read By Type req (0x08)
      start 0x000a, end 0x000e
      type-uuid 0x2802
2015-11-09 08:09:04.457613 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 66 packets 1
2015-11-09 08:09:04.462752 < ACL data: handle 66 flags 0x00 dlen 9
    ATT: Error (0x01)
      Error: Attribute not found (10)
      Read By Type req (0x08) on handle 0x000a
2015-11-09 08:09:04.554743 > ACL data: handle 66 flags 0x02 dlen 11
    ATT: Read By Type req (0x08)
      start 0x000a, end 0x000e
      type-uuid 0x2803
2015-11-09 08:09:04.555615 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 66 packets 1
2015-11-09 08:09:04.556777 < ACL data: handle 66 flags 0x00 dlen 48
    ATT: Read By Type resp (0x09)
      length: 21
        handle 0x000b, value 0x08 0x0c 0x00 0x9f 0xdc 0x9c 0x81 0xff 0xfe 0xcf 0x8b 0xe5 0x11 0x65 0x84 0xc8 0x7d 0xe2 0xba 
        handle 0x000d, value 0x08 0x0e 0x00 0x9f 0xdc 0x9c 0x81 0xff 0xfe 0xcf 0x8b 0xe5 0x11 0x65 0x84 0x4e 0x7f 0xe2 0xba 
2015-11-09 08:09:04.749634 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 66 packets 1
```

johnyso, 2015-11-09:
Thanks @danki for the tip to "reset" bleno-app. It does not solve this issue but i can continue developing a prototype. It's possible to connect an Android device after the iPhone 4s was connected to bleno service with more than 1 (128-Bit UUID) or 3 (16-Bit UUID) characteristics. 

This is my "not very good" workaround: 
I use the [foreverjs] (https://github.com/foreverjs/forever) tool to run the app continuously. Once a device disconnects i restart the bleno-app.
Here is the example code: 
```
...
bleno.on('disconnect', function(clientAddress){
	console.log("client Disconnected " + clientAddress + " restart advertisement");
	bleno.stopAdvertising();
	// restart bleno-app
	process.exit(1);
});
...
```

danki, 2015-11-09:
Hey, i found something that made me thing, feel free to correct me if im out of bounds.
When looking your hcidump, it states 
```
HCI sniffer - Bluetooth packet analyzer ver 2.4
```

while mine shows:
```
root@sednew:~# hcidump
HCI sniffer - Bluetooth packet analyzer ver 5.24
device: hci0 snap_len: 1500 filter: 0xffffffff
```

Which (if any?) bluez lib do you use ? Maybe is there the issue lies, not using bluez library.

johnyso, 2015-11-12:
Sorry for late responding. 
Thanks for the tip. Indeed i had an old version. 
After removing all the packages and install bluez-5.36 from this tutorial [RPI Bluetooth LE] (http://www.elinux.org/RPi_Bluetooth_LE) the problem was still there.
Now I tried to setup the raspberry from scratch, i thought maybe the packages was not fully uninstalled. But nothing changes. After the iPhone 4s connects to the service it's impossible for android to connect to it again.

If there is no one else who still has this problem it must be an issue with my raspberry config or bluetooth dongle etc... I'll continue trying out different settings...

here the current hcidump after an iPhone was connected and the android device tries to read the data:
```
root@raspberrypi:~# hcidump -x -t
HCI sniffer - Bluetooth packet analyzer ver 5.36
device: hci0 snap_len: 1500 filter: 0xffffffff
2015-11-12 12:42:11.907449 > HCI Event: LE Meta Event (0x3e) plen 19
    LE Connection Complete
      status 0x00 handle 69, role slave
      bdaddr 14:30:C6:00:2D:9D (Public)
2015-11-12 12:42:11.908005 < ACL data: handle 69 flags 0x00 dlen 16
    L2CAP(d): cid 0x0005 len 12 [psm 0]
      12 01 08 00 28 00 38 00 00 00 D0 07 
2015-11-12 12:42:11.976479 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 69 packets 1
2015-11-12 12:42:12.024247 > ACL data: handle 69 flags 0x02 dlen 10
    L2CAP(d): cid 0x0005 len 6 [psm 0]
      13 01 02 00 00 00 
2015-11-12 12:42:12.073124 > ACL data: handle 69 flags 0x02 dlen 11
    ATT: Read By Group req (0x10)
      start 0x0001, end 0xffff
      type-uuid 0x2800
2015-11-12 12:42:12.075082 < ACL data: handle 69 flags 0x00 dlen 18
    ATT: Read By Group resp (0x11)
      attr handle 0x0001, end group handle 0x0005
      value 0x00 0x18
      attr handle 0x0006, end group handle 0x0009
      value 0x01 0x18
2015-11-12 12:42:12.170758 > ACL data: handle 69 flags 0x02 dlen 11
    ATT: Read By Group req (0x10)
      start 0x000a, end 0xffff
      type-uuid 0x2800
2015-11-12 12:42:12.171497 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 69 packets 1
2015-11-12 12:42:12.173439 < ACL data: handle 69 flags 0x00 dlen 26
    ATT: Read By Group resp (0x11)
      attr handle 0x000a, end group handle 0x000e
      value 0x9f 0xdc 0x9c 0x81 0xff 0xfe 0xcf 0x8b 0xe5 0x11 0x65 0x84 0x80 0x7a 0xe2 0xba
2015-11-12 12:42:12.268396 > ACL data: handle 69 flags 0x02 dlen 11
    ATT: Read By Group req (0x10)
      start 0x000f, end 0xffff
      type-uuid 0x2800
2015-11-12 12:42:12.268519 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 69 packets 1
2015-11-12 12:42:12.270074 < ACL data: handle 69 flags 0x00 dlen 9
    ATT: Error (0x01)
      Error: Attribute not found (10)
      Read By Group req (0x10) on handle 0x000f
2015-11-12 12:42:12.366032 > ACL data: handle 69 flags 0x02 dlen 11
    ATT: Read By Type req (0x08)
      start 0x0001, end 0x0005
      type-uuid 0x2802
2015-11-12 12:42:12.366521 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 69 packets 1
2015-11-12 12:42:12.367979 < ACL data: handle 69 flags 0x00 dlen 9
    ATT: Error (0x01)
      Error: Attribute not found (10)
      Read By Type req (0x08) on handle 0x0001
2015-11-12 12:42:12.463539 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 69 packets 1
2015-11-12 12:42:12.463686 > ACL data: handle 69 flags 0x02 dlen 11
    ATT: Read By Type req (0x08)
      start 0x0001, end 0x0005
      type-uuid 0x2803
[... 53 lines cut ...]
        handle 0x0009, uuid 0x2902 (GATT(desc) Client Characteristic Configuration)
2015-11-12 12:42:13.048247 > ACL data: handle 69 flags 0x02 dlen 11
    ATT: Read By Type req (0x08)
      start 0x000a, end 0x000e
      type-uuid 0x2802
2015-11-12 12:42:13.048615 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 69 packets 1
2015-11-12 12:42:13.050432 < ACL data: handle 69 flags 0x00 dlen 9
    ATT: Error (0x01)
      Error: Attribute not found (10)
      Read By Type req (0x08) on handle 0x000a
2015-11-12 12:42:13.145887 > ACL data: handle 69 flags 0x02 dlen 11
    ATT: Read By Type req (0x08)
      start 0x000a, end 0x000e
      type-uuid 0x2803
2015-11-12 12:42:13.146629 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 69 packets 1
2015-11-12 12:42:13.148372 < ACL data: handle 69 flags 0x00 dlen 48
    ATT: Read By Type resp (0x09)
      length: 21
        handle 0x000b, value 0x08 0x0c 0x00 0x9f 0xdc 0x9c 0x81 0xff 0xfe 0xcf 0x8b 0xe5 0x11 0x65 0x84 0xc8 0x7d 0xe2 0xba 
        handle 0x000d, value 0x08 0x0e 0x00 0x9f 0xdc 0x9c 0x81 0xff 0xfe 0xcf 0x8b 0xe5 0x11 0x65 0x84 0x4e 0x7f 0xe2 0xba 
2015-11-12 12:42:13.243635 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 69 packets 1
```

danki, 2015-11-12:
I know from my experience that sometime when you have connected to iot device from phone, after disconnecting from it it will be unresponsive.

Only thing ive found on that which works for me is to restart the BLE (bleno) app on iot device. That makes sure everything get cleared from "cache?" and all is set up again properly when restarting.

Not sure if there are a official way to accomplish that tho.

Sorry i do have rPI tho no BLE dongle for it to test and run for any help :/
Hope you solve it and / or find a workaround, let us know how it goes :)

danki, 2015-11-13:
What could be a possible workaround...

when ever you get disconnect in bleno, execute ***rfkill block bluetooth***. And maybe exit nodejs app, let a service to restart it or something.
When Bleno app starts, run ***rfkill unblock bluetooth***.

This kind of works for me on Edison, seems rfkill does **hard-reset** in the system on BLE stuff.

johnyso, 2015-11-13:
my current workaround is to stop the node service and restart it again. This works for me. Thanks for the tip. (I use foreverJs)
Should I close this Issue? If i find another solution i will post it, or reopen it?

danki, 2015-11-13:
Awesome, that confirms restart of bleno does release **something** in the system.

I think this issue is not much of Bleno issue, but the workaround could still be implemented in some way into bleno, either if sandeep does so or anyone else with time and knowledge of bleno core.
So leave it open, see how community responds...

sandeepmistry, 2015-11-14:
@johnyso thanks for the logs, I a bit confused with your sample app in https://github.com/sandeepmistry/bleno/issues/145#issuecomment-154996847 it has 2 characteristics vs the 4 mentioned in the title of the issue.

From the output of the HCI dump things look ok to me, the last item:
```
2015-11-12 12:42:13.145887 > ACL data: handle 69 flags 0x02 dlen 11
    ATT: Read By Type req (0x08)
      start 0x000a, end 0x000e
      type-uuid 0x2803
2015-11-12 12:42:13.146629 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 69 packets 1
2015-11-12 12:42:13.148372 < ACL data: handle 69 flags 0x00 dlen 48
    ATT: Read By Type resp (0x09)
      length: 21
        handle 0x000b, value 0x08 0x0c 0x00 0x9f 0xdc 0x9c 0x81 0xff 0xfe 0xcf 0x8b 0xe5 0x11 0x65 0x84 0xc8 0x7d 0xe2 0xba 
        handle 0x000d, value 0x08 0x0e 0x00 0x9f 0xdc 0x9c 0x81 0xff 0xfe 0xcf 0x8b 0xe5 0x11 0x65 0x84 0x4e 0x7f 0xe2 0xba 
2015-11-12 12:42:13.243635 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 69 packets 1
```

Matches with:
```javascript
        new Characteristic({
            uuid: 'bae27dc8846511e58bcffeff819cdc9f', 
            properties: [ 'write' ],
            value: null,
        }),
        new Characteristic({
            uuid: 'bae27f4e846511e58bcffeff819cdc9f', 
            properties: [ 'write' ],
            value: null,
        })
```

I'm wondering if it's MTU related, I don't see Android requesting a MTU request in the HCI dump, so the value from iOS might be used.

Could you try adding the following line here: https://github.com/sandeepmistry/bleno/blob/master/lib/hci-socket/gatt.js#L257

```javascript
this._mtu = 23;
```

This will reset the MTU back to the default 23.

johnyso, 2015-11-17:
@sandeepmistry i tried different settings and i realized that this problem already accurs with two characteristics -128 Bit UUID. Thats why the example has only 2 characteristics. Sorry for the confusion.

Thanks so much for your tip! It works now! I tried it also with 7 Characteristics - 128 UUIDs.

This solves this issue!

sandeepmistry, 2015-11-19:
@johnyso thanks for try it out! I've pushed a commit with this change to master: https://github.com/sandeepmistry/bleno/commit/e7dbeb8c9fa4bf51523a8aa20e76c3a90725f67d

closed 2015-11-19 by sandeepmistry

bfree-github, 2016-10-20:
Hmmm - I've encountered a similar problem.

I had bleno running with 13 characteristics on a Mac, and was able to read them from various devices/platforms.

I recently switched to ubuntu, and I no longer able to read them. The most I could fetch were 4, otherwise it would timeout.  But it was not so much the number, but rather which characteristics - sometimes the most I could get was one, but picking the correct characteristcs, I could get up to 4 working.

I tried up'g the MTU, and that didn't seem to help.

I'll try to send more info later.