noble/bleno #146: hci0 tx timeout

danki, 2015-11-11:
I ran a callback too intensively and now when i restart my app, i get this..

```
Conf init
Initiating ble...
   Load conf
Netconnection: true
[  881.505020] Bluetooth: hci0 command 0x0c01 tx timeout
[  883.504717] Bluetooth: hci0 command 0x2001 tx timeout
[  885.504447] Bluetooth: hci0 command 0x1001 tx timeout
[  887.504047] Bluetooth: hci0 command 0x1009 tx timeout
[  889.503828] Bluetooth: hci0 command 0x0c14 tx timeout
[  891.503554] Bluetooth: hci0 command 0x200a tx timeout
```
Unfortunately, hcidump gives nothing...
I wonder how / where can i capture this "error" so i can reset the BLE adapter ?

Note, i ran DEBUG but that revealed nothing

sandeepmistry, 2015-11-14:
@danki what's generating the "tx timeout" message? The kernel?

danki, 2015-11-15:
@sandeepmistry dont know how / where to find that out, any suggestions ?

When ever i get this warnings, even exiting bleno app does not publish this anymore but when entering back to bleno app, they appear again when bleno initializes.

sandeepmistry, 2015-11-15:
@danki take a look at the output of ```dmesg```, if HCI commands are timing out during transmit and cannot be sent to the Bluetooth adapter not much bleno can do.

Does a "hciconfig hci0 reset" fix the issue?

Any steps to consistently reproduce (including code)?

danki, 2015-11-16:
ahh forgot about dmesg, I'll check it out as soon as I get warnings again.

I tried all kinds of ways to reset BLE adapter, a bit unsure if I tested hci reset (I think I did, as well down).
only one that reseted BLE adapter was 
rfkill block and then unblock

danki, 2015-11-18:
Yeah, disabling bluetooth on phone does kill the connection entirely, Edison does get disconnected so this obviously is a BLE thing itself and not bleno

closed 2015-11-18 by danki