noble/bleno #148: Cannot add custom service

arusu0629, 2015-11-16:
I'm japanese student.
please teach me to add custom service.

```
[pi@raspberry] $ cat testBLE.js                                                                                      [~/testBluetooth]
var bleno = require('./node_modules/bleno');

var name = 'Toru-no-Raspberry Pi2';
var serviceUuids = ['993A0B2F-04D2-49EA-A481-C5BDDB42422A'];

var PrimaryService = bleno.PrimaryService;
var Characteristic = bleno.Characteristic;

var characteristic = new Characteristic({
    uuid: '133F4237-5F57-4151-A269-237003B4D0A7',
    properties: ['read'],
    value: new Buffer('Raspberry Pi2-Toru') // must be buffer class
});

var primaryService = new PrimaryService({
    uuid: 'DC341849-ECA1-47A8-B9B3-A1D8B09DA48A',
    characteristics: characteristic
});

bleno.on('stateChange', function(state) {
    console.log('on -> stateChange: ' + state);

    if (state === 'poweredOn') {
	bleno.startAdvertising(name, serviceUuids);
	console.log('start Advertising');
    } else {
	bleno.stopAdvertising();
	console.log('stop Advertising');
    }
});

bleno.on('advertisingStart', function(error) {
    console.log('on -> advertisingStart: ' + (error ? 'error' + error : 'success'));
    if (!error) {
	bleno.setServices([primaryService]);
    }
});

bleno.on('servicesSet', function(error) {
    console.log('servicesSet: ' + (error ? 'error' + error : 'success'));
});

bleno.on('servicesSetError', function(error) {
    console.log('servicesSetError: ' + (error ? 'error' + error : 'success'));
})
```

And,
```
[pi@raspberry] $ node testBLE.js                                                                                     [~/testBluetooth]
on -> stateChange: poweredOn
start Advertising
on -> advertisingStart: success
servicesSet: success
```

No Error message.

#### I use LightBlue app for debugging BLE.
#### I want to find my custom service, but cannot found it.

[image]

[image]

[image]

[image]

[image]

Please Sir.

danki, 2015-11-16:
This code:
```javascript
var primaryService = new PrimaryService({
    uuid: 'DC341849-ECA1-47A8-B9B3-A1D8B09DA48A',
    characteristics: characteristic
});
```
should be like this
```javascript
var primaryService = new PrimaryService({
    uuid: 'DC341849-ECA1-47A8-B9B3-A1D8B09DA48A',
    // characteristics property is expected to be array
    characteristics: [ characteristic ]
});
```

sandeepmistry, 2015-11-16:
@arusu0629 yes, you'll have to make the change @danki suggested.

Also, if you are using ```bluetoothd``` version of 5.x you'll also have to stop it, as per https://github.com/sandeepmistry/bleno/issues/24.

Here's how to get the version:
```sh
pi@raspberrypi ~ $ bluetoothd --version
5.23
```

Then how to stop it and power up the Bluetooth adapter:
```sh
pi@raspberrypi ~ $ sudo /etc/init.d/bluetooth stop
[ ok ] Stopping bluetooth (via systemctl): bluetooth.service.
pi@raspberrypi ~ $ sudo hciconfig hci0 up
```

After that:
1. On your iOS device, turn bluetooth off and on again
2. Start bleno app
3. Connect using LightBlue

Let me know if this works for you.

arusu0629, 2015-11-17:
@danki, @sandeepmistry 
Thank you for replying message.

I fix those code

```
var primaryService = new PrimaryService({
    uuid: 'DC341849-ECA1-47A8-B9B3-A1D8B09DA48A'
    characteristics: [characteristic] // array
})
```

And, I don't using bluetoothd maybe because of finding command #### bluetoothd

```
[pi@raspberry] $ bluetoothd --version                                                                               [~/testBluetooth]
zsh: command not found: bluetoothd
```

But, I did @sandeepmistry said
```
pi@raspberrypi ~ $ sudo /etc/init.d/bluetooth stop
[ ok ] Stopping bluetooth (via systemctl): bluetooth.service.
pi@raspberrypi ~ $ sudo hciconfig hci0 up
```

After that,
I execute LightBlue app and Raspbery-Pi2 again, I did it!!!
It looks work well!!! Thank you a lot.

closed 2015-11-17 by arusu0629

sandeepmistry, 2015-11-19:
@arusu0629 great! Thanks for following up!

timosadchiy, 2016-05-10:
Hi!

I have a similar problem and has spent quite a while on it. Basically, I cannot set custom service or characteristic UUIDs. I also use bleno on Raspberry Pi 2 with Raspbian. 'bluetoothd --version' gives me
'5.23'. I did 'sudo /etc/init.d/bluetooth stop' and 'sudo hciconfig hci0 up'

For testing I used slightly changed Echo example. Here is my code:

    'use strict';

    var util = require('util'),
        bleno = require('bleno'),
        serviceUUID = '659CF625-F6CE-4D51-A0A3-351B1D2D1A0E',
        characteristicUUID = '22B6DFA8-D6AD-408C-891E-D24F5D646F0F',
        BlenoPrimaryService = bleno.PrimaryService,
        BlenoCharacteristic = bleno.Characteristic;

    var EchoCharacteristic = function () {
        EchoCharacteristic.super_.call(this, {
            uuid: characteristicUUID,
            properties: ['read', 'write', 'notify'],
            value: null
        });

        this._value = new Buffer(0);
        this._updateValueCallback = null;
    };

    util.inherits(EchoCharacteristic, BlenoCharacteristic);

    EchoCharacteristic.prototype.onReadRequest = function (offset, callback) {
        console.log('EchoCharacteristic - onReadRequest: value = ' + this._value.toString('hex'));

        callback(this.RESULT_SUCCESS, this._value);
    };

    EchoCharacteristic.prototype.onWriteRequest = function (data, offset, withoutResponse, callback) {
        this._value = data;

        console.log('EchoCharacteristic - onWriteRequest: value = ' + this._value.toString('hex'));

        if (this._updateValueCallback) {
            console.log('EchoCharacteristic - onWriteRequest: notifying');
            this._updateValueCallback(this._value);
        }

        callback(this.RESULT_SUCCESS);
    };

    EchoCharacteristic.prototype.onSubscribe = function (maxValueSize, updateValueCallback) {
        console.log('EchoCharacteristic - onSubscribe');

        this._updateValueCallback = updateValueCallback;
    };

    EchoCharacteristic.prototype.onUnsubscribe = function () {
        console.log('EchoCharacteristic - onUnsubscribe');
       this._updateValueCallback = null;
    };

    module.exports = function () {
        console.log('bleno - echo');

        bleno.on('stateChange', function (state) {
            console.log('on -> stateChange: ' + state);

            if (state === 'poweredOn') {
                bleno.startAdvertising('echo', [serviceUUID]);
            } else {
                bleno.stopAdvertising();
            }
        });

        bleno.on('advertisingStart', function (error) {
            console.log('on -> advertisingStart: ' + (error ? 'error ' + error : 'success'));

            if (!error) {
                bleno.setServices([
                    new BlenoPrimaryService({
                        uuid: serviceUUID,
                        characteristics: [
                            new EchoCharacteristic()
                        ]
                    })
                ]);
            }
        });

        bleno.on('accept', function (clientAddress) {
            console.log('Connected:', clientAddress);
        });

        bleno.on('disconnect', function (clientAddress) {
            console.log('Disconnected:', clientAddress);
        });

         bleno.on('servicesSet', function (error) {
            console.log('servicesSet: ' + (error ? 'error' + error : 'success'));
         });

         bleno.on('servicesSetError', function (error) {
            console.log('servicesSetError: ' + (error ? 'error' + error : 'success'));
         });
    };

If I connect to my device using the LightBlue App, I cannot find neither service with the UUID '659CF625-F6CE-4D51-A0A3-351B1D2D1A0E' nor the characteristic with the UUID '22B6DFA8-D6AD-408C-891E-D24F5D646F0F'. However, if I write any value to the 'Alert Level' characteristic available in the 'Link Loss' service (both have different UUIDs), the listener set by 'EchoCharacteristic.prototype.onWriteRequest' in the code above reacts on it.

Please help me to find out, what am I doing wrong!

timosadchiy, 2016-05-10:
So, I think I have found the solution to my problem, which was rebooting my iOS device. Found here [#67](https://github.com/sandeepmistry/bleno/issues/67#issuecomment-53963617)