noble/bleno #151: Consider yocto-board as an Intel Edison board

beaufortfrancois, 2015-11-26:
It would be nice if https://github.com/sandeepmistry/bleno/blob/ceb2151ac9656ce96234e58eb5b880be9d120a38/lib/hci-socket/gap.js#L9 could also include `yocto-standard`, not only `edison`.
 
See #131 and  https://gitter.im/sandeepmistry/bleno?at=5656d27acac1354864a7068a

sandeepmistry, 2015-11-27:
Implemented in https://github.com/sandeepmistry/bleno/commit/0c150480320f23bc04ea1618670712a4e9c55f91

@beaufortfrancois can you please try out master: ```npm install sandeepmistry/bleno```?

**Note:** it would be better to get the adapter info via HCI commands (including standard and Broadcom vendor specific), so the workaround is not run on non-Edsion yocto based boards. However, using ```os.release()``` is ok for now ...

closed 2015-11-27 by sandeepmistry

beaufortfrancois, 2015-11-27:
It works perfectly fine now!
Thank you @sandeepmistry

sandeepmistry, 2015-11-28:
@beaufortfrancois great! Thanks for confirming the change.