noble/bleno #153: Changing MAC Adress

paulgavrikov, 2015-11-26:
Is there a way to change the MAC adress that is broadcasted from a peripherial?

sandeepmistry, 2015-11-26:
@paulgavrikov this is not implemented currently.

Could you please provide more details on your use case?

paulgavrikov, 2015-11-27:
I am working on a repeater device, which works by mirroring a peripheral. I know that some chinese devices support changing MAC, so I was wondering if this could work here, too.

sandeepmistry, 2015-11-27:
@paulgavrikov bleno could support using a random BT address on Linux, but not a high priority currently.

For public address, you can use the BlueZ ```bdaddr``` tool can be used to change the address of most CSR and Broadcom adapters.

http://git.kernel.org/cgit/bluetooth/bluez.git/tree/tools/bdaddr.c

paulgavrikov, 2015-11-27:
Thanks! I will try bdaddr!

sandeepmistry, 2015-11-27:
@paulgavrikov no problem!

closed 2015-11-27 by sandeepmistry