noble/bleno #155: No callback when advertising stops

brghena, 2015-12-07:
I'm running into a bad interaction between Bleno and Noble.

Bleno is running in the background and advertising data. When a script using Noble is run, advertisement stops, but the Bleno application doesn't seem to get a callback that the state has changed or that advertisement has stopped.

After the Noble script has stopped, the Bleno application still believes it is running without problems, but there is no advertising. Stopping and restarting the Bleno application manually begins advertisement again.

It would be great to get a callback when the system decides I no longer get to advertise so that I could automatically restart advertisement later.

(Alternatively, being able to advertise with Bleno while scanning with Noble would be the best, but I understand that hardware may not support that.)

sandeepmistry, 2015-12-08:
@brghena do you have a (simple) code sample that I can use to reproduce the issue?

brghena, 2015-12-09:
I created a gist that is available here: https://gist.github.com/brghena/0729612baae067912595

Method to reproduce problem:
    1) Start either script. Functions fine.
    2) Start the other script. Functions fine.
        i) The first will stop functioning (will not see additional devices / no longer advertises respectively)
        ii) No callback will be received in the first script

The bug has occurred on Ubuntu 14.04 with bleno@0.3.3 and noble@1.2.1

brghena, 2015-12-09:
I'm not sure what layer is actually causing the scripts to fail. Following this guide http://stackoverflow.com/questions/16151360/use-bluez-stack-as-a-peripheral-advertiser to use hciconfig to set an advertisement and then running 'sudo hcitool lescan' results scanning for devices and advertising concurrently. So, at least somewhere down the stack advertising and scanning simultaneously are possible with my hardware (USB BLE adapter by Medialink).

sandeepmistry, 2015-12-14:
@brghena thanks for the clarification. Advertisement should not stop when scanning is started, only when a connection occurs using noble.

Could you also provided an HCI dump capture? ```sudo hcidump -t -x```

Have you tried combining both scripts into one?

brghena, 2015-12-14:
Attached is the output from:
    1) Running `sudo hcidump -i hci1 -t -x -w hcidump.out`
    2) Running `sudo ./bleno_app.js`
    3) Running `sudo ./noble_app.js`
    4) Ctrl-C on noble_app.js
    5) Ctrl-C on bleno_app.js
    6) Ctrl-C on hcidump
with about 5 seconds between each step. Note, my BLE adapter is on hci1. The attached file is NOT a txt file, but rather the raw dump from hcidump. (Github limited upload file options)

In the case of what I'm doing right now, it's not possible to combine both scripts into one. I have not tried to see if the same effect would result.

Not the core issue here, but do you have any insight into why advertisement should stop when a connection occurs?

[hcidump.txt](https://github.com/sandeepmistry/bleno/files/60679/hcidump.txt)

sandeepmistry, 2015-12-14:
@brghena thanks!

> Not the core issue here, but do you have any insight into why advertisement should stop when a connection occurs?

With Bluetooth 4.0, there's a simple state machine, you can't be in a connected state and advertising at the same time. See figure 7.4 in this book: http://www.amazon.ca/gp/product/B009XDA1G8

From the HCI dump, I can't see indication of any event that would indicate advertising stopping. According to figure 7.4 above, you can't scan and advertise at the same time, but I'm pretty sure I have got this to work with a Broadcom based adapter.

One new thing that was introduced in noble and bleno was the "Write LE Host Supported" command:

```
2015-12-14 00:33:52.675976 < HCI Command: Write LE Host Supported (0x03|0x006d) plen 2
  01 00 
2015-12-14 00:33:52.678845 > HCI Event: Command Complete (0x0e) plen 4
    Write LE Host Supported (0x03|0x006d) ncmd 1
    00
```

You could go into ```gatt.js``` and try removing sending this command. Maybe it's reseting the adapter somehow.

Fallback would be to use an HCI interface per script, via specifying the interface to use via an environment variable (see read me's).

sandeepmistry, 2016-01-03:
@brghena any updates on this?

brghena, 2016-01-06:
I haven't been able to dig into the noble libraries at all.

The only update I've got is that using hcitool & hciconfig to advertise, and then starting noble stops said advertisement. Running noble and then using hcitool & hciconfig to advertise allows advertisement and noble keeps working. It makes it seem to me that noble does something on startup that stops advertisements over the hci interface.

sandeepmistry, 2016-01-08:
@brghena this might be related to https://github.com/sandeepmistry/noble/issues/287 which has been fixed in both the master versions of noble and bleno.

Can you try those versions?

```
npm install sandeepmistry/noble
npm install sandeepmistry/bleno
```

brghena, 2016-01-08:
Heck yeah, totally solves the problem. Bleno and Noble can function at the same time now. Thanks a bunch for keeping up on this.

closed 2016-01-08 by brghena

sandeepmistry, 2016-01-08:
@brghena sweet!

I'll be publishing a new release of both to npm over the next few days.