noble/bleno #160: Central + Peripheral Simulatenously

popasquat89, 2015-12-18:
Is it possible to have noble connected to other peripheral while Bleno hosts a peripheral connection? I've attempted to get this working, but noble doesn't ever achieve the connections.

Bryce

sandeepmistry, 2015-12-20:
@popasquat89 is this on Linux?

You'll need a Bluetooth 4.1 adapter to achieve this, and most of the adapters I've seen are still Bluetooth 4.0 only. If you have more than one Bluetooth adapter you can specify which one noble and bleno should use via environment variables (see read me's).

The following patch in noble, will now report connection failures, it has not been published to npm yet. https://github.com/sandeepmistry/noble/issues/255

popasquat89, 2015-12-24:
> not been published to npm yet. sandeepmistry/noble#255

Hey sorry for the late response, end of the year is getting busy.

We're running on Debian, with a 4.0 adapter. We found a 4.1 adapter that's
made in the UK, but no such luck in the US as of yet. (LM910) It's not
practical for us to throw in another USB adapter for a couple of reasons,
so I'm hoping I can get my hands on a 4.1 adapter soon to test this theory
out.

@sandeepmistry

Bryce

sandeepmistry, 2016-04-11:
Closing due to lack of discussion and https://github.com/sandeepmistry/noble/issues/377 seems promising :).

closed 2016-04-11 by sandeepmistry