noble/bleno #161: No characteristic is passed on 'characteristicsDiscover' at a central component.

kaippp, 2015-12-19:
Hello Sandeep,
 
I’m very happy using your noble/bleno modules.
 
I was able to use both noble/bleno modules successfully on a raspberry pi with raspbian.
 
But now I have a problem when running the same bleno based peripheral peripher.js on a raspberry with raspbian 4.1.13 + nodejs V4.2.3.
 
A central component (built using noble, or an iOS app, and assuming they are OK) receives an event ‚on characteristicsDiscover‘, but the characteristics passed is empty (length = 0). What could be the reasons for this Problem at the peripheral?
 
I would be very happy if your could help me.

Thank you.

[BlenoIssue.zip](https://github.com/sandeepmistry/bleno/files/67622/BlenoIssue.zip)

sandeepmistry, 2016-01-03:
@kaippp looks ok to me, central OS X 10.11 and peripheral Raspberry Pi.

```
$  node bleno-issue-161.js 
noble = central
on -> stateChange: poweredOn
startScanning 00
on -> scanStart
startScanning 99
on -> discover: {"id":"1861702501ef4f9da58583e21c642dd8","address":"00:19:0e:11:28:17","addressType":"unknown","connectable":true,"advertisement":{"localName":"raspberrypi","serviceData":[],"serviceUuids":["2215f5762aa942e4a300ceb07484b415"]},"rssi":-87,"state":"disconnected"}
connecting
on -> connect
on -> scanStop
on -> peripheral services discovered:
serviceIndex = 0
on -> service characteristics discovered {"uuid":"9666ffd4b3d14985a4051ef2a6dbb283","name":null,"type":null,"properties":["notify"]}
on -> characteristic notify true
```

Also, screenshot from LightBlue on OS X:

[image]

Could you share a screenshot of what LightBlue on iOS shows for the services and characteristics of the bleno peripheral?

kaippp, 2016-01-05:
Hello Sandeep,

thank you for your response.

Attached are the screenshots (upper + lower parts) of the LightBlue app on iPad connecting to peripher.js that runs on a Raspberry pi.
The characteristic was not displayed by the LightBlue app.
Raspberry pi:
- Raspbian Linux 4.1.13-v7+
- Node v4.2.4
- bluez 5.23 (If the bluetoothd process was killed, peripher.js doesn't work at all)

Note: If the same peripher.js runs on the same Raspberry but with the following software, than there is no problem.
- Raspbian Linux 4.1.7-v7+
- Node 0.10.40
- bluez 4.99

kaippp
[LightBlue-peripher.js.zip](https://github.com/sandeepmistry/bleno/files/78492/LightBlue-peripher.js.zip)

sandeepmistry, 2016-01-05:
@kaippp still looks like bluetoothd was still interfering when you are using BlueZ 5.23.

Can you confirm is was not running? ```pgrep bluetoothd```

There are also some instruction here to stop it: https://github.com/sandeepmistry/bleno#linux

kaippp, 2016-01-07:
@sandeepmistry 

pi@raspberrypi:~ $ apt-show-versions bluez
bluez:armhf/jessie 5.23-2+b1 uptodate

pi@raspberrypi:~ $ apt-show-versions bluetooth
bluetooth:all/jessie 5.23-2 uptodate

When bluetoothd is running, peripher.js, missing expected onSubscribe event, shows:

bleno = peripheral
on -> stateChange: poweredOn, address = 00:15:83:d2:07:9d
on -> advertisingStart: success
ctor SampleService
ctor NotifyOnlyCharacteristic
on -> servicesSet: success
on -> accept, client: 45:f1:40:cd:e9:e4
on -> rssiUpdate: -50

After stopping bluetoothd by "sudo service bluetooth stop", peripher.js shows only one line:

bleno = peripheral

On the same raspberry, when bluetoothd is running, a noble program works well; But when bluetoothd is stopped, the noble program doesn't work at all.

It seems that bluetoothd is still required; Without it bleno/noble don't work at all.

sandeepmistry, 2016-01-08:
@kaippp when you stop ```bluetoothd``` it powers off the Bluetooth adapter. You need to use ```sudo hciconfig hci0 up``` to power up manually.

closed 2016-01-08 by kaippp

kaippp, 2016-01-08:
Indeed, the bleno module is absolute OK.
On my raspberry pi with bluez 5.23, I must do the following to solve the problem:
- sudo service bluetooth stop
- sudo hciconfig hci0 up 

@sandeepmistry Thank you very much for your support. I'm happy to have the problem solved.

sandeepmistry, 2016-01-09:
@kaippp great! I've added a note in the read me for this (https://github.com/sandeepmistry/bleno/commit/821eaf64b262ba0ea01f564282f68240d0836e7d).