noble/bleno #165: handleReadByTypeRequest crash

jslawek, 2016-01-16:
Steps to reproduce:

start example echo service (hci0)

```
ja@s ~/node_modules/bleno $ node examples/echo/main
bleno - echo
on -> stateChange: poweredOn
on -> advertisingStart: success
```

gatt client (bluez 5.37), hci1 - reading/writing handles works correctly:

```
ja@s ~ $ gatttool -i hci1 -I -b DC:53:60:43:43:43
[DC:53:60:43:43:43][LE]> connect
Attempting to connect to DC:53:60:43:43:43
Connection successful
[DC:53:60:43:43:43][LE]> characteristics
handle: 0x0002, char properties: 0x02, char value handle: 0x0003, uuid: 00002a00-0000-1000-8000-00805f9b34fb
handle: 0x0004, char properties: 0x02, char value handle: 0x0005, uuid: 00002a01-0000-1000-8000-00805f9b34fb
handle: 0x0007, char properties: 0x20, char value handle: 0x0008, uuid: 00002a05-0000-1000-8000-00805f9b34fb
handle: 0x000b, char properties: 0x1a, char value handle: 0x000c, uuid: 0000ec0e-0000-1000-8000-00805f9b34fb
[DC:53:60:43:43:43][LE]> char-write-req 0x0c aa
Characteristic value was written successfully
[DC:53:60:43:43:43][LE]> char-read-hnd 0x0c 
Characteristic value/descriptor: aa 
[DC:53:60:43:43:43][LE]> 
```
Unfortunatelly while trying to read by uuid:
```
[DC:53:60:43:43:43][LE]> char-read-uuid 0000ec0e-0000-1000-8000-00805f9b34fb
[DC:53:60:43:43:43][LE]> 
(gatttool:2543): GLib-WARNING **: Invalid file descriptor.

```
or 

```
[DC:53:60:43:43:43][LE]> char-read-uuid 0xec0e
[DC:53:60:43:43:43][LE]> 
```

the service crashes:

```
ja@s ~/node_modules/bleno $ node examples/echo/main
bleno - echo
on -> stateChange: poweredOn
on -> advertisingStart: success
EchoCharacteristic - onWriteRequest: value = aa
EchoCharacteristic - onReadRequest: value = aa
/home/ja/node_modules/bleno/lib/hci-socket/gatt.js:661
      var dataLength = Math.min(data.length, this._mtu - 4);
                                    ^
TypeError: Cannot read property 'length' of null
    at Gatt.handleReadByTypeRequest (/home/ja/node_modules/bleno/lib/hci-socket/gatt.js:661:37)
    at Gatt.handleRequest (/home/ja/node_modules/bleno/lib/hci-socket/gatt.js:317:23)
    at Gatt.onAclStreamData (/home/ja/node_modules/bleno/lib/hci-socket/gatt.js:273:8)
    at emit (events.js:129:20)
    at AclStream.push (/home/ja/node_modules/bleno/lib/hci-socket/acl-stream.js:26:10)
    at BlenoBindings.onAclDataPkt (/home/ja/node_modules/bleno/lib/hci-socket/bindings.js:190:21)
    at emit (events.js:118:17)
    at Hci.onSocketData (/home/ja/node_modules/bleno/lib/hci-socket/hci.js:430:14)
    at BluetoothHciSocket.emit (events.js:107:17)

```

Same scenario with other example services (e.g. pizza).

sandeepmistry, 2016-01-17:
@jslawek thanks for the detail issue report and steps to reproduce!

I'll take a closer look later this week.

sandeepmistry, 2016-02-01:
@jslawek I've pushed some changes in https://github.com/sandeepmistry/bleno/commit/b224753c9bb666b18fb3efd22173c0a1b48c44b1 to resolve this.

The previous read by type request handler was only expecting static characteristic values, both static and dynamic are now supported.

You can test it via ```npm install sandeepmistry/bleno``` before a new release is published to npm.

closed 2016-02-01 by sandeepmistry

jslawek, 2016-02-01:
Works great, thanks a lot!

sandeepmistry, 2016-02-02:
@jslawek awesome, thanks for verifying! I've published v0.3.5 of bleno to npm, which includes the fix.

cc/ @InMyOrbit