noble/bleno #166: Android client Characteristic name is already "Unknown"

ducmaxime, 2016-01-18:
Hello, how can I set the characteristic name ?

My function  : 

var GetConfiguration = function()
{
    GetConfiguration.super_.call(this,
    {
        uuid:       '31E98556-045A-4A22-9E88-6BF944352C21',
        properties: ['read', 'write', 'indicate'],
        value:      null
    });

    this._value                 = new Buffer(0);
    this._updateValueCallback   = null;
};
In my android client I receive "Unknown" for the characteristic name 

Thanks,

nickhod, 2016-01-20:
I also would like to know how to set the name of both Characteristics and Services.

Thanks.

sandeepmistry, 2016-01-23:
@ducmaxime this might not be possible on Android, the name might only be set for Bluetooth SIG standard UUID's:

 * https://developer.bluetooth.org/gatt/services/Pages/ServicesHome.aspx?_ga=1.44626528.591292021.1365037697
 * https://developer.bluetooth.org/gatt/characteristics/Pages/CharacteristicsHome.aspx?_ga=1.44626528.591292021.1365037697

You could trying adding a [user description descriptor](https://developer.bluetooth.org/gatt/descriptors/Pages/DescriptorViewer.aspx?u=org.bluetooth.descriptor.gatt.characteristic_user_description.xml) to the characteristic, see:

https://github.com/sandeepmistry/bleno/blob/master/test.js#L17-L21

@don do you have any info on this?

sandeepmistry, 2016-01-30:
I'm closing this, as nothing can be done the bleno side.

closed 2016-01-30 by sandeepmistry