noble/bleno #167: Cannot connect to service

jefBinomed, 2016-01-25:
I have a strange problem probably link to my bluetooth card because I reproduce this problem only on one of my computer.

The usecase is very simple, I copy your sample project with, read / write caracteristic and try to access with application [NRF](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp) on android.

My code is available here [simple gat server](https://github.com/binomed/WebExperiments/blob/master/WebBluetooth/ble_server.js)

here is the result of 
```sh
$sudo hciconfig - a
```

```sh
hci0:	Type: BR/EDR  Bus: USB
	BD Address: 5C:93:A2:9E:CA:94  ACL MTU: 1022:8  SCO MTU: 183:5
	UP RUNNING PSCAN ISCAN 
	RX bytes:1908 acl:29 sco:0 events:151 errors:0
	TX bytes:4019 acl:55 sco:0 commands:84 errors:0
	Features: 0xff 0xfe 0x0d 0xfe 0xd8 0x7f 0x7b 0x8f
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH HOLD SNIFF 
	Link mode: SLAVE ACCEPT 
	Name: 'ChromeLinux'
	Class: 0x0c010c
	Service Classes: Rendering, Capturing
	Device Class: Computer, Laptop
	HCI Version: 4.1 (0x7)  Revision: 0x3101
	LMP Version: 4.1 (0x7)  Subversion: 0x1
	Manufacturer: Atheros Communications, Inc. (69)
```

It don't work on my Lenvo Y50-70 under Ubuntu 15.10. It don't work with a nexus 4 under Lolipop 5.1 and same thing for a Samsung Galaxy S5 under lolipop

The same script works on a Dell under Ubuntu 15.10 and on a RaspberryPi Zero.

On Nrf master control I have the following error : Error 129 (0x91) GATT INTERNAL ERROR

sandeepmistry, 2016-01-26:
@jefBinomed a few suggestions:

1. Have you stopped bluetoothd as per the read me, if you are running BlueZ 5.x
2. You could try adjust the max MTU, to 23. This is the first time I heard of someone using an Atheros based adapter: https://github.com/sandeepmistry/bleno/blob/master/lib/hci-socket/gatt.js#L67 (you'll need to clone the latest from master)

jefBinomed, 2016-01-26:
It works ! But is it normal that when i do a npm install on 0.3.4 version I don't have the same version of code ?  For example, the file gatt.js getting with npm don't have this.maxMtu, i had to download the zip of the repository of bleno and set it to maxMtu = 23 to make it work.

Maybe, you should do a version 0.3.5 ?

but thanks for the tips :+1:

sandeepmistry, 2016-01-30:
> It works ! But is it normal that when i do a npm install on 0.3.4 version I don't have the same version of code ? For example, the file gatt.js getting with npm don't have this.maxMtu, i had to download the zip of the repository of bleno and set it to maxMtu = 23 to make it work.

Yes, I typically don't publish a new version to npm after every commit.

> Maybe, you should do a version 0.3.5 ?

I will eventually ...

What manufacture id does you Bluetooth adapter have? We can update the following section:

https://github.com/sandeepmistry/bleno/blob/master/lib/hci-socket/bindings.js#L123-L128

> Edit : after some trys, it not stable at all :( for a reason I don't understand, the bluetooth adapter of my computer turn on alone an it close my connection :( And If I continue after a moment, I can't start the bluetooth service, I'm forced to reboot the computer :(
>
> So it works but not for a long time and it isn't stable :(

Sorry to hear this, maybe you'd be better off using an USB Bluetooth 4.0 adapter instead?

sandeepmistry, 2016-02-08:
Closing this for now, due to lack of activity. Please re-open with requested info.

closed 2016-02-08 by sandeepmistry