noble/bleno #169: Unable to install bleno (build)

MiguelRipoll23, 2016-02-08:
I've installed on my computer:

- Visual Studio Community
- Visual Studio Express (I think it's using this one, didn't change anything)
- Python 2.7 (npm uses this one, I changed the config)
- Python 3.4 

So node is using Python 2.7 but the it fails when it starts building.

> npm WARN install Couldn't install optional dependency: Unsupported
> 
> > usb@1.1.1 install C:\Users\user\node_modules\usb
> > node-pre-gyp install --fallback-to-build
> 
> 
> C:\Users\user\node_modules\usb>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\nod
> e-gyp.js" clean )  else (node  clean )
> 
> C:\Users\user\node_modules\usb>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\nod
> e-gyp.js" configure --fallback-to-build --module=C:\Users\user\node_modules\usb\src\binding\usb_bindings.node --module_name=usb_bindings --module_path=C:\Users\user\n
> ode_modules\usb\src\binding --python=/path/to/executable/python2.7 )  else (node  configure --fallback-to-build --module=C:\Users\user\node_modules\usb\src\binding\usb
> _bindings.node --module_name=usb_bindings --module_path=C:\Users\user\node_modules\usb\src\binding --python=/path/to/executable/python2.7 )
> 
> C:\Users\user\node_modules\usb>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\nod
> e-gyp.js" build --fallback-to-build --module=C:\Users\user\node_modules\usb\src\binding\usb_bindings.node --module_name=usb_bindings --module_path=C:\Users\user\node_
> modules\usb\src\binding )  else (node  build --fallback-to-build --module=C:\Users\user\node_modules\usb\src\binding\usb_bindings.node --module_name=usb_bindings --mod
> ule_path=C:\Users\user\node_modules\usb\src\binding )
>   core.c
>   descriptor.c
>   hotplug.c
>   io.c
> C:\Users\user\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: 'timespec': 'struct' type redefinition (compiling source file ..\libusb\libusb\io.
> c)C:\Users\user\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: 'timespec': 'struct' type redefinition (compiling source file ..\libusb\libusb\d
> escriptor.c) [C:\Users\user\node_modules\usb\build\libusb.vcxproj]
> 
>   C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declaration of 'timespec' (compiling source file ..\libusb\libusb\io.c)
>   C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declaration of 'timespec' (compiling source file ..\libusb\libusb\descriptor.c
>   )
> ..\libusb\libusb\io.c(1227): warning C4244: '=': conversion from 'time_t' to 'long', possible loss of data [C:\Users\user\node_modules\usb\build\libusb.vcxproj]
> ..\libusb\libusb\io.c(1902): warning C4244: '=': conversion from 'time_t' to 'long', possible loss of data [C:\Users\user\node_modules\usb\build\libusb.vcxproj]
> ..\libusb\libusb\io.c(2435): warning C4244: '=': conversion from 'time_t' to 'long', possible loss of data [C:\Users\user\node_modules\usb\build\libusb.vcxproj]
>   strerror.c
> C:\Users\user\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: 'timespec': 'struct' type redefinition (compiling source file ..\libusb\libusb\hot
> plug.c) [C:\Users\user\node_modules\usb\build\libusb.vcxproj]
> C:\Users\user\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: 'timespec': 'struct' type redefinition (compiling source file ..\libusb\libusb\cor
> e.c)C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declaration of 'timespec' (compiling source file ..\libusb\libusb\hotplug.c)
> sync.c [C:\Users\user\node_modules\usb\build\libusb.vcxproj]
> 
> 
>   C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declaration of 'timespec' (compiling source file ..\libusb\libusb\core.c)
>   poll_windows.c
>   threads_windows.c
> C:\Users\user\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: 'timespec': 'struct' type redefinition (compiling source file ..\libusb\libusb\syn
> c.c) [C:\Users\user\node_modules\usb\build\libusb.vcxproj]
>   C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declaration of 'timespec' (compiling source file ..\libusb\libusb\sync.c)
> C:\Users\user\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: 'timespec': 'struct' type redefinition (compiling source file ..\libusb\libusb\str
> error.c) [C:\Users\user\node_modules\usb\build\libusb.vcxproj]
>   C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declaration of 'timespec' (compiling source file ..\libusb\libusb\strerror.c)
>   windows_usb.c
>   win_delay_load_hook.c
> C:\Users\user\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: 'timespec': 'struct' type redefinition (compiling source file ..\libusb\libusb\os\
> poll_windows.c) [C:\Users\user\node_modules\usb\build\libusb.vcxproj]
>   C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declaration of 'timespec' (compiling source file ..\libusb\libusb\os\poll_wind
>   ows.c)
> C:\Users\user\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: 'timespec': 'struct' type redefinition (compiling source file ..\libusb\libusb\os\
> threads_windows.c) [C:\Users\user\node_modules\usb\build\libusb.vcxproj]
>   C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declaration of 'timespec' (compiling source file ..\libusb\libusb\os\threads_w
>   indows.c)
> ..\libusb\libusb\os\threads_windows.c(194): warning C4244: '=': conversion from 'time_t' to 'long', possible loss of data [C:\Users\user\node_modules\usb\build\libusb
> .vcxproj]
[... 34 lines cut ...]
> node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:185:7)
> node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:821:16)
> node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
> node-pre-gyp ERR! System Windows_NT 10.0.10586
> node-pre-gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\user\\node_modules\\usb\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "install" "--fallba
> ck-to-build"
> node-pre-gyp ERR! cwd C:\Users\user\node_modules\usb
> node-pre-gyp ERR! node -v v5.5.0
> node-pre-gyp ERR! node-pre-gyp -v v0.6.20
> node-pre-gyp ERR! not ok
> Failed to execute 'node-gyp.cmd build --fallback-to-build --module=C:\Users\user\node_modules\usb\src\binding\usb_bindings.node --module_name=usb_bindings --module_pat
> h=C:\Users\user\node_modules\usb\src\binding' (1)
> npm WARN install:usb@1.1.1 usb@1.1.1 install: `node-pre-gyp install --fallback-to-build`
> npm WARN install:usb@1.1.1 Exit status 1
> C:\Users\user
> └── bleno@0.3.5
> 
> npm WARN ENOENT ENOENT: no such file or directory, open 'C:\Users\user\package.json'
> npm WARN EPACKAGEJSON user No description
> npm WARN EPACKAGEJSON user No repository field.
> npm WARN EPACKAGEJSON user No README data
> npm WARN EPACKAGEJSON user No license field.
> npm WARN EBUNDLEOVERRIDE Replacing bundled node_modules\usb\node_modules\node-pre-gyp with new installed version

C:\Users\user>

sandeepmistry, 2016-02-13:
@MiguelRipoll23 the node-usb dependency is failing.

Please open an issue [here](https://github.com/nonolith/node-usb/issues) and let me know if the bleno documentation needs to be updated once you have things working: https://github.com/nonolith/node-usb

closed 2016-02-13 by sandeepmistry

MiguelRipoll23, 2016-02-14:
@sandeepmistry got it, thank you.

ShadaabKhan, 2016-03-08:
I'm trying to install usb npm package seperately since unstalling with the bluetooth-hci-socket npm package wasn't working. And I ran into the following error. I'm using Windows 10.

C:\Users\Shadaab>npm install usb --msvs_version=2015
npm WARN update-linked node_modules\express needs updating to 4.13.4 from 4.13.4 but we can't, as it's a symlink

> usb@1.1.1 install C:\Users\Shadaab\node_modules\usb
> node-pre-gyp install --fallback-to-build

Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  core.c
  descriptor.c
  hotplug.c
  io.c
  strerror.c
  sync.c
  poll_windows.c
  threads_windows.c
C:\Users\Shadaab\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error
 C2011: 'timespec': 'struct' type redefinition (compiling source file ..\libusb
\libusb\sync.c)C:\Users\Shadaab\node_modules\usb\libusb\libusb\os/threads_windo
ws.h(47): error C2011: 'timespec': 'struct' type redefinition (compiling source
 file ..\libusb\libusb\descriptor.c) [C:\Users\Shadaab\node_modules\usb\build\l
ibusb.vcxproj]

  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39):
  note: see declaration of 'timespec' (compiling source file ..\libusb\libusb\s
  ync.c)C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h
  (39): note: see declaration of 'timespec' (compiling source file ..\libusb\li
  busb\descriptor.c)C:\Users\Shadaab\node_modules\usb\libusb\libusb\os/threads_
  windows.h(47): error C2011: 'timespec': 'struct' type redefinition (compiling
   source file ..\libusb\libusb\io.c)

  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39):
  note: see declaration of 'timespec' (compiling source file ..\libusb\libusb\i
  o.c)
..\libusb\libusb\io.c(1227): warning C4244: '=': conversion from 'time_t' to 'l
ong', possible loss of data [C:\Users\Shadaab\node_modules\usb\build\libusb.vcx
proj]
  windows_usb.c
  win_delay_load_hook.c
..\libusb\libusb\io.c(1902): warning C4244: '=': conversion from 'time_t' to 'l
ong', possible loss of data [C:\Users\Shadaab\node_modules\usb\build\libusb.vcx
proj]
..\libusb\libusb\io.c(2435): warning C4244: '=': conversion from 'time_t' to 'l
ong', possible loss of dataC:\Users\Shadaab\node_modules\usb\libusb\libusb\os/t
hreads_windows.h(47): error C2011: 'timespec': 'struct' type redefinition (comp
iling source file ..\libusb\libusb\hotplug.c) [C:\Users\Shadaab\node_modules\us
b\build\libusb.vcxproj]

  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39):
  note: see declaration of 'timespec' (compiling source file ..\libusb\libusb\h
  otplug.c)C:\Users\Shadaab\node_modules\usb\libusb\libusb\os/threads_windows.h
  (47): error C2011: 'timespec': 'struct' type redefinition (compiling source f
  ile ..\libusb\libusb\os\poll_windows.c)

  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39):
  note: see declaration of 'timespec' (compiling source file ..\libusb\libusb\o
  s\poll_windows.c)
C:\Users\Shadaab\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error
 C2011: 'timespec': 'struct' type redefinition (compiling source file ..\libusb
\libusb\strerror.c) [C:\Users\Shadaab\node_modules\usb\build\libusb.vcxproj]
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39):
  note: see declaration of 'timespec' (compiling source file ..\libusb\libusb\s
  trerror.c)
C:\Users\Shadaab\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error
 C2011: 'timespec': 'struct' type redefinition (compiling source file ..\libusb
\libusb\os\threads_windows.c) [C:\Users\Shadaab\node_modules\usb\build\libusb.v
cxproj]
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39):
  note: see declaration of 'timespec' (compiling source file ..\libusb\libusb\o
[... 71 lines cut ...]
npm WARN Shadaab No README data
npm WARN Shadaab No license field.
npm ERR! Windows_NT 10.0.10586
npm ERR! argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "install" "usb" "--msvs_version=2015"
npm ERR! node v5.7.1
npm ERR! npm  v3.6.0
npm ERR! code ELIFECYCLE

npm ERR! usb@1.1.1 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the usb@1.1.1 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the usb package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs usb
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls usb
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     C:\Users\Shadaab\npm-debug.log

sandeepmistry, 2016-03-13:
@ShadaabKhan this is dependent on https://github.com/nonolith/node-usb/issues/109 as far as I can tell.