noble/bleno #170: onIndicate not documented

dfowler7437, 2016-02-16:
How is onIndicate used. I assumed it would be called when the client accepted an update. After the client received a subscribed Characteristic change notify/update... 

I have never seen a call to onIndicate...

I don't see how it's used in the documentation or any sample.

Is there a way to confirm that Characteristic updates have been received at the client? Other then some process where a write from the client is used as an ACK?

sandeepmistry, 2016-02-21:
@dfowler7437 

> I don't see how it's used in the documentation or any sample

Please see brief documentation in the read me: https://github.com/sandeepmistry/bleno#characteristic

Code snippet: https://github.com/sandeepmistry/bleno/blob/master/test.js#L128-L162

> Is there a way to confirm that Characteristic updates have been received at the client? Other then some process where a write from the client is used as an ACK?

This is the purpose of ```onIndicate```, it is only supported with characteristics that have the ```indicate``` property.

closed 2016-02-21 by sandeepmistry

dfowler7437, 2016-03-02:
I don't see documentation. I see a mocked up test case but that gives no indication as to when indicate would be called or how to setup a case where it would be called.  My strong suggestion is to improve the documentation in that area.  For my application I have moved on with a handshake approach.

don, 2016-03-02:
@dfowler7437 When your app sends an indication (as opposed to notification) to the client, your app needs to wait the confirmation from the client before sending another indication. `onIndicate` is the callback that is called when the server (your code) get's the confirmation from the client.

This library generally follows Bluetooth LE conventions. Reading and running the sample code is a great way to figure out how things work. Where things need clarification or improvement please submit pull requests including improvements to the documentation.

dfowler7437, 2016-03-03:
Thank you for the help.  I see now that your Charictistic can be either "notify" or "indicate"

I have implemented an "indicate" and do get the onIndicate call when I change a Charisterict but only once...

I am sending several chunks via the Characteristic. The logic is as follows.
Application sends long message via Characteristic. 
1)break up message into chunks - send first chunk
2)on "indicate" send next chunk - if done flag the completion

step 2 should happen several times.  I see the first chunk transfer, then the "indicate" sends a second chunk, but not an "indicate" event for the second chunk. this stalls the transfer...

Is there a step I need to do to enable the "indicate" for an update call.

Note that I am calling the "updateValueCallback(chunk)" inside the "indicate" handler when there are more chunks to send.

dfowler7437, 2016-03-03:
More on the above...
I was calling the updateValueCallback inside the onIndicate as described. I changed so that I use a timeout to all updateValueCallback instead. With a 0 timeout value. That worked.

So I'm not directly calling updateValueCallback directly inside the onIndicate.

Does this mean there is some reentrant code issue?  Shouldn't I be able to turn around another message the instant I get the handshake?

don, 2016-03-03:
@dfowler7437 I have a similar use case sending multiple indications to transfer data that is too large for one characteristic. In `onIndicate` I check if there are additional chunks of data and call another function that sends it. I've only run this on OS X. The next indication is sent as soon as the first acknowledgement arrives.

uchobby, 2016-03-04:
Thank you Don...

I am doing the exact same thing. I found that I could not call the updateValueCallback from inside the onIndicate callback. That would fail to trigger a follow up onIndicate.

I tried using a timeout to call updateValueCallback with the next chunk to send and that worked. With a time out of 0. I see very slow transfers this way.  I'm guessing the turn around time is based on the advertising interval which is set to 100ms by default. 

I tried setting the intervale to the minimum 20ms and did not see a noticeable increase. Maybe I need to change the interval in the client as well.

Am I seeing normal operation?  With the updateValueCallback and with the long turn around times?

Another question.... In the on subscribe callback...  One of the parameters is "maxValueSize", described as "maximum data size".  I read that BLE has a max size of 20 bytes somewhere so I have been breaking up my messages into 20 byte chunks.  Can / should I use the much larger value reported by "maxValueSize"?  Does bleno or the lower level driver would break up the message and send it all at once in 20byte chunks for me...

don, 2016-03-04:
I haven't changed any of the timings.

20 bytes is the safest characteristic length. Longer data can be sent using a blob. With bleno on OS X I can receive up to 500 bytes. Bleno assembles the chunks automatically. I haven't tested sending blobs from the server to the client, but I expect it works the same.

dfowler7437, 2016-03-05:
Should I post the two items I've found as new issues?
1) Calling updateValueCallback repeatedly to update a Characteristic can crash something in the bluetooth stack. Recovery requires a system reboot, might be OK with a Bluez restart, have not tried that yet.

2)Calling updateValueCallback from inside the onIndicate method does not update the Characteristic in a way that triggers further onIndicate calls.

I don't know what you mean about the "haven't changed any of the timings"  My question was related to why I would need to call the updateVlueCallback via a timeout...  That works even with a timeout of zero MS.  I'm thinking the timeout 0 just drops a call on the message stack so that it's called after onIndicate returns rather then before.

dfowler7437, 2016-03-05:
On #1. by crash.. I mean that the BLE system fails to function correctly. The connection is dropped, and any follow up connections are accepted then dropped a second later.  There is no warning or error, the system seems to fail silently.

mattsson, 2016-10-31:
I'm experiencing exactly the same thing, i.e. I need to set a timeout 0 before calling updateValueCallback from inside the onIndicate method, but I'm only seeing this on Linux, specifically on Raspbian. The same code works fine running on OS X.

sandeepmistry, 2016-10-31:
@mattsson please open a new issue with the following:

* minimal sample code to reproduce the issue
* debug logs: `sudo DEBUG=* node <file>.js`
* HCI dump: `sudo hcidump -t -x`