noble/bleno #171: connection disconnect after pairing

ysknkd, 2016-02-19:
I'm using bleno on rpi (wheezy). 

When the peripheral(bleno) device sends a security request, the central(noble) and the peripheral performs the pairing.
But... After about 4 seconds, the peripheral(maybe not bleno) send a disconnect command, the link is disconnected.
I confirmed the disconnect processing of hci.js. but, disconnect seemed not called.

Is there a way to maintain the connection ?

```
2016-02-19 18:17:31.590611 > ACL data: handle 64 flags 0x02 dlen 7
    ATT: Read req (0x0a)
      handle 0x000c
2016-02-19 18:17:31.609117 < ACL data: handle 64 flags 0x00 dlen 9
    ATT: Error (0x01)
      Error: Insufficient authentication (5)
      Read req (0x0a) on handle 0x000c
2016-02-19 18:17:31.725753 > ACL data: handle 64 flags 0x02 dlen 11
    SMP: Pairing Request (0x01)
      capability 0x03 oob 0x00 auth req 0x01
      max key size 0x10 init key dist 0x00 resp key dist 0x01
      Capability: NoInputNoOutput (OOB data not present)
      Authentication: Bonding (No MITM Protection)
      Initiator Key Distribution:    
      Responder Key Distribution:  LTK  
2016-02-19 18:17:31.726243 < ACL data: handle 64 flags 0x00 dlen 11
    SMP: Pairing Response (0x02)
      capability 0x03 oob 0x00 auth req 0x01
      max key size 0x10 init key dist 0x00 resp key dist 0x01
      Capability: NoInputNoOutput (OOB data not present)
      Authentication: Bonding (No MITM Protection)
      Initiator Key Distribution:    
      Responder Key Distribution:  LTK  
2016-02-19 18:17:31.733848 < ACL data: handle 64 flags 0x00 dlen 11
    SMP: Pairing Response (0x02)
      capability 0x03 oob 0x00 auth req 0x01
      max key size 0x10 init key dist 0x00 resp key dist 0x01
      Capability: NoInputNoOutput (OOB data not present)
      Authentication: Bonding (No MITM Protection)
      Initiator Key Distribution:    
      Responder Key Distribution:  LTK  
2016-02-19 18:17:31.793360 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 2
2016-02-19 18:17:31.860894 > ACL data: handle 64 flags 0x02 dlen 21
    SMP: Pairing Confirm (0x03)
      key bd2550ecae1fe55f486bf2866c3f60cf
2016-02-19 18:17:31.861294 < ACL data: handle 64 flags 0x00 dlen 21
    SMP: Pairing Confirm (0x03)
      key e14d587b0d5b6f4abbbc247e0d33fdfe
2016-02-19 18:17:31.862276 > ACL data: handle 64 flags 0x02 dlen 21
    SMP: Pairing Confirm (0x03)
      key 58f26495006f50509b8b989f1ea243a4
2016-02-19 18:17:31.907986 < ACL data: handle 64 flags 0x00 dlen 21
    SMP: Pairing Confirm (0x03)
      key 0fc94ad1679267ebd53c8b354744cdba
2016-02-19 18:17:31.919278 < ACL data: handle 64 flags 0x00 dlen 21
    SMP: Pairing Confirm (0x03)
      key 8963f6495364f79af02c609b46bfe8f5
2016-02-19 18:17:31.992388 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 1
2016-02-19 18:17:31.997380 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 2
2016-02-19 18:17:32.063654 > ACL data: handle 64 flags 0x02 dlen 21
    SMP: Pairing Random (0x04)
      random f8d83e3954757e99c0063e2f9a9729d3
2016-02-19 18:17:32.063779 > ACL data: handle 64 flags 0x02 dlen 21
    SMP: Pairing Random (0x04)
      random f8d83e3954757e99c0063e2f9a9729d3
2016-02-19 18:17:32.064400 < ACL data: handle 64 flags 0x00 dlen 6
    SMP: Pairing Failed (0x05)
      reason 0x04
      Reason Confirm Value Failed
2016-02-19 18:17:32.065166 > ACL data: handle 64 flags 0x02 dlen 21
    SMP: Pairing Random (0x04)
      random f8d83e3954757e99c0063e2f9a9729d3
2016-02-19 18:17:32.098928 < ACL data: handle 64 flags 0x00 dlen 21
    SMP: Pairing Random (0x04)
      random 30d1b01549141e2bb65301efcf749d13
2016-02-19 18:17:32.113525 < ACL data: handle 64 flags 0x00 dlen 21
    SMP: Pairing Random (0x04)
      random 30d1b01549141e2bb65301efcf749d13
2016-02-19 18:17:32.124897 < ACL data: handle 64 flags 0x00 dlen 21
    SMP: Pairing Random (0x04)
      random 30d1b01549141e2bb65301efcf749d13
2016-02-19 18:17:32.131400 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 2
2016-02-19 18:17:32.333417 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 2
2016-02-19 18:17:32.468437 > HCI Event: LE Meta Event (0x3e) plen 13
    LE Long Term Key Request
    40 00 00 00 00 00 00 00 00 00 00 00 
2016-02-19 18:17:32.468656 < HCI Command: LE Long Term Key Request Reply (0x08|0x001a) plen 18
  40 00 19 BD AF 9A 8A 17 0D F9 61 BC 39 38 97 7A 35 DF 
2016-02-19 18:17:32.470441 > HCI Event: Command Complete (0x0e) plen 6
    LE Long Term Key Request Reply (0x08|0x001a) ncmd 1
    00 40 00 
2016-02-19 18:17:32.617462 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 1
2016-02-19 18:17:32.738468 > HCI Event: Encrypt Change (0x08) plen 4
    status 0x00 handle 64 encrypt 0x01
2016-02-19 18:17:32.745269 < ACL data: handle 64 flags 0x00 dlen 21
    SMP: Encryption Information (0x06)
      LTK 19bdaf9a8a170df961bc3938977a35df
2016-02-19 18:17:32.748683 < ACL data: handle 64 flags 0x00 dlen 15
    SMP: Master Identification (0x07)
      EDIV 0x0000 Rand 0x0000000000000000
2016-02-19 18:17:33.143529 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 2
2016-02-19 18:17:33.346566 > HCI Event: LE Meta Event (0x3e) plen 13
    LE Long Term Key Request
    40 00 00 00 00 00 00 00 00 00 00 00 
2016-02-19 18:17:33.346784 < HCI Command: LE Long Term Key Request Reply (0x08|0x001a) plen 18
  40 00 19 BD AF 9A 8A 17 0D F9 61 BC 39 38 97 7A 35 DF 
2016-02-19 18:17:33.348583 > HCI Event: Command Complete (0x0e) plen 6
    LE Long Term Key Request Reply (0x08|0x001a) ncmd 1
    00 40 00 
2016-02-19 18:17:33.683588 > HCI Event: Encryption Key Refresh Complete (0x30) plen 3
    status 0x00 handle 64
2016-02-19 18:17:33.818008 > ACL data: handle 64 flags 0x02 dlen 7
    ATT: Read req (0x0a)
      handle 0x000c
2016-02-19 18:17:33.860771 < ACL data: handle 64 flags 0x00 dlen 42
    ATT: Read resp (0x0b)
      30 31 32 33 34 35 36 37 38 39 61 62 63 64 65 66 67 68 69 6A 
      6B 6C 6D 6E 6F 70 71 72 73 74 75 76 77 78 79 7A 5F 
2016-02-19 18:17:34.117641 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 1
2016-02-19 18:17:34.560712 > ACL data: handle 64 flags 0x02 dlen 7
    ATT: Read req (0x0a)
      handle 0x000c
2016-02-19 18:17:34.574136 < ACL data: handle 64 flags 0x00 dlen 42
    ATT: Read resp (0x0b)
      30 31 32 33 34 35 36 37 38 39 61 62 63 64 65 66 67 68 69 6A 
      6B 6C 6D 6E 6F 70 71 72 73 74 75 76 77 78 79 7A 5F 
2016-02-19 18:17:34.867732 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 1
2016-02-19 18:17:37.598710 > ACL data: handle 64 flags 0x02 dlen 7
    ATT: Read req (0x0a)
      handle 0x000c
2016-02-19 18:17:37.605180 < ACL data: handle 64 flags 0x00 dlen 42
    ATT: Read resp (0x0b)
      30 31 32 33 34 35 36 37 38 39 61 62 63 64 65 66 67 68 69 6A 
      6B 6C 6D 6E 6F 70 71 72 73 74 75 76 77 78 79 7A 5F 
2016-02-19 18:17:37.676191 < HCI Command: Disconnect (0x01|0x0006) plen 3
    handle 64 reason 0x13
    Reason: Remote User Terminated Connection
2016-02-19 18:17:37.678076 > HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
2016-02-19 18:17:37.868092 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 1
2016-02-19 18:17:37.936102 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x16
    Reason: Connection Terminated by Local Host
```

sandeepmistry, 2016-02-21:
@ysknkd 

> Is there a way to maintain the connection ?

This might be a Linux kernel issue.

Are things more stable without using secure characteristics?

Can you please provide debug output? ```sudo DEBUG=* node <file>.js``` As well as share the code you are running.

ysknkd, 2016-02-28:
Thank you for your reply.

> This might be a Linux kernel issue.

Perhapse, SMP of kernel module is processing "Pairing Request Command". Could you tell me how to stop SMP of kernel module?

> Are things more stable without using secure characteristics?

Yes. Connection is stable without secure characteristics.

> Can you please provide debug output? sudo DEBUG=* node <file>.js As well as share the code you are running.

OK. Please wait for a little more time.

sandeepmistry, 2016-03-13:
Closing due to lack of feedback, please re-open with requested info if you are still interested.

closed 2016-03-13 by sandeepmistry

WayneKeenan, 2016-03-17:
Hi, 

I'm getting the same issue on a Pi for both the battery-service and echo examples on a stock Jessie (2016-02-26) and a updated kernel, via rbi-update.

Versions (post rpi-update):
Node: v4.4.0
Linux raspberrypi 4.1.19-v7+ #858 SMP Tue Mar 15 15:56:00 GMT 2016 armv7l GNU/Linux

To replicate the same base OS I'm using from the 2016-02-26 jessie (full) image:

download: https://downloads.raspberrypi.org/raspbian/images/raspbian-2016-02-29/2016-02-26-raspbian-jessie.zip

sudo apt-get remove nodered nodejs nodejs-legacy npm
sudo apt-get update
curl -sL https://deb.nodesource.com/setup_4.x | sudo bash -
apt-get install nodejs
sudo apt-get install -y build-essential python-dev python-rpi.gpio nodejs libudev-dev libusb-1.0-0.dev
sudo rpi-update  f406502f5628d32e6ca5dadac34ff7ca59f8e27f

Node Debug logs attached.

[EchoService-NodeDebug.txt](https://github.com/sandeepmistry/bleno/files/177457/EchoService-NodeDebug.txt)
[BatterService-NodeDebug.txt](https://github.com/sandeepmistry/bleno/files/177458/BatterService-NodeDebug.txt)