noble/bleno #175: Cannot connect to bleno from Android.

Vijayenthiran, 2016-02-29:
I am using bleno on Raspberry pi 2. I have been testing my bleno codes with LightBlue App on iOS and it works greats with it. The same code is not working with any android application (nRF master control panel and BLE scanner). I even tried the `test.js` and pizza examples. Both the examples are working fine with the iOS but not on android. Here are the details:

_Single Board Computer:_ Raspberry Pi 2
_Bluez version: _5.23-2+b1
_Kernel Version: _Linux raspberrypi 4.1.17-v7+ #838 SMP Tue Feb 9 13:15:09 GMT 2016 armv7l GNU/Linux
_BLE Adapter used:_ Bluetooth 4.0 USB dongle
_Node.js version: _v4.3.1
_Android Verson: _5.1.1
 
_HCI dump while running `test.js` and trying to connect with nRF master control panel (Android):_

```
pi@raspberrypi:~ $ sudo hcidump -t -x
HCI sniffer - Bluetooth packet analyzer ver 5.23
device: hci0 snap_len: 1500 filter: 0xffffffff
2016-02-23 20:09:23.906991 < HCI Command: Set Event Mask (0x03|0x0001) plen 8
    Mask: 0xfffffbff07f8bf3d
2016-02-23 20:09:23.908228 > HCI Event: Command Complete (0x0e) plen 4
    Set Event Mask (0x03|0x0001) ncmd 1
    status 0x00
2016-02-23 20:09:23.910240 < HCI Command: LE Set Event Mask (0x08|0x0001) plen 8
    mask 0x1f00000000000000 (Reserved)
2016-02-23 20:09:23.912203 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Event Mask (0x08|0x0001) ncmd 1
    status 0x00
2016-02-23 20:09:23.912270 < HCI Command: Read Local Version Information (0x04|0x0001) plen 0
2016-02-23 20:09:23.925248 > HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.0 (0x6) HCI Revision: 0x22bb
    LMP Version: 4.0 (0x6) LMP Subversion: 0x22bb
    Manufacturer: Cambridge Silicon Radio (10)
2016-02-23 20:09:23.925405 < HCI Command: Write LE Host Supported (0x03|0x006d) plen 2
  01 00
2016-02-23 20:09:23.927203 > HCI Event: Command Complete (0x0e) plen 4
    Write LE Host Supported (0x03|0x006d) ncmd 1
    00
2016-02-23 20:09:23.927313 < HCI Command: Read LE Host Supported (0x03|0x006c) plen 0
2016-02-23 20:09:23.929197 > HCI Event: Command Complete (0x0e) plen 6
    Read LE Host Supported (0x03|0x006c) ncmd 1
    00 01 00
2016-02-23 20:09:23.929262 < HCI Command: Read BD ADDR (0x04|0x0009) plen 0
2016-02-23 20:09:23.931201 > HCI Event: Command Complete (0x0e) plen 10
    Read BD ADDR (0x04|0x0009) ncmd 1
    status 0x00 bdaddr 00:1A:7D:DA:71:13
2016-02-23 20:09:23.979228 < HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1
  00
2016-02-23 20:09:23.981218 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertise Enable (0x08|0x000a) ncmd 1
    status 0x0c
    Error: Command Disallowed
2016-02-23 20:09:23.981371 < HCI Command: LE Set Advertising Parameters (0x08|0x0006) plen 15
    min 100.000ms, max 100.000ms
    type 0x00 (ADV_IND - Connectable undirected advertising) ownbdaddr 0x00 (Public)
    directbdaddr 0x00 (Public) 00:00:00:00:00:00
    channelmap 0x07 filterpolicy 0x00 (Allow scan from any, connection from any)
2016-02-23 20:09:23.983211 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertising Parameters (0x08|0x0006) ncmd 1
    status 0x00
2016-02-23 20:09:24.006178 < HCI Command: LE Set Scan Response Data (0x08|0x0009) plen 32
  06 05 08 74 65 73 74 00 00 00 00 00 00 00 00 00 00 00 00 00
  00 00 00 00 00 00 00 00 00 00 00 00
2016-02-23 20:09:24.008222 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Response Data (0x08|0x0009) ncmd 1
    status 0x00
2016-02-23 20:09:24.008299 < HCI Command: LE Set Advertising Data (0x08|0x0008) plen 32
  15 02 01 06 11 06 F0 FF FF FF FF FF FF FF FF FF FF FF FF FF
  FF FF 00 00 00 00 00 00 00 00 00 00
2016-02-23 20:09:24.010208 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertising Data (0x08|0x0008) ncmd 1
[... 312 lines cut ...]
    ATT: Write resp (0x13)
2016-02-23 20:11:36.323866 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 69 packets 1
2016-02-23 20:11:37.200890 < ACL data: handle 69 flags 0x00 dlen 11
    ATT: Handle indicate (0x1d)
    18 00 00 00 00 00
2016-02-23 20:11:37.336395 > ACL data: handle 69 flags 0x02 dlen 5
    ATT: Handle CNF (0x1e)
2016-02-23 20:11:37.337001 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 69 packets 1
2016-02-23 20:11:38.204649 < ACL data: handle 69 flags 0x00 dlen 11
    ATT: Handle indicate (0x1d)
    18 00 01 00 00 00
2016-02-23 20:11:38.281635 > ACL data: handle 69 flags 0x02 dlen 5
    ATT: Handle CNF (0x1e)
2016-02-23 20:11:38.282133 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 69 packets 1
2016-02-23 20:11:39.091863 > ACL data: handle 69 flags 0x02 dlen 9
    ATT: Write req (0x12)
      handle 0x0019 value  0x00 0x00
2016-02-23 20:11:39.098466 < ACL data: handle 69 flags 0x00 dlen 5
    ATT: Write resp (0x13)
2016-02-23 20:11:39.227243 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 69 packets 1
```

sandeepmistry, 2016-03-06:
@Vijayenthiran there is no sign of a connection attempt from your Android device in the HCI dump capture you provided. Not sure what can be done on the bleno side, I would suggest you try connecting to another BLE peripheral from your Android device, as well as trying again after toggling Bluetooth off and on again.

Vijayenthiran, 2016-04-06:
Hi Sandeep,

I tried with the following devices:
LG Nexus 5, Sony Xperia Z3 and Samsung Galaxy S5.
On the raspberry pi side, I tried with the following BLE Adapters:
Leoxsys and Kingston BLE adapter.

I am using Raspbian Jessie Version:
`Linux raspberrypi 4.1.17-v7+ #838 SMP Tue Feb 9 13:15:09 GMT 2016 armv7l GNU/Lin                                                                                        ux`
BlueZ Version: `5.23-2+b1 `

I ran the test.js file and for all the combinations, I am able to connect the peripheral to the iOS with the LightBlue application. Screenshots shown below:
[image]

Whereas when I try with nRF Master Control Panel or BLE Scanner on Android, it say 'Connecting' but after few seconds it disconnects and no service is listed. There is no connection attempt on the bleno side also. Screenshots of the app:
[image]
[image]

What could be the possible issue?

sandeepmistry, 2016-04-09:
@Vijayenthiran please provide an HCI dump capture ```sudo hcidump -t -x```.

sandeepmistry, 2016-04-11:
@Vijayenthiran @vdharmon

You could also try tweaking the GAP flags, maybe 0x05 is better than 0x06?

https://github.com/sandeepmistry/bleno/issues/175

vdharmon, 2016-04-11:
If you can point me to exactly what you need me to change in code I will change and test.

vdharmon, 2016-04-11:
I'm assuming you mean change a flag in the gap.js file.

vdharmon, 2016-04-12:
I've updated to the latest firmware for the PI 3  to (4.4.6-v7+) and the code is working for both IOs and Android.

Thank you for your help.

mfferreira, 2016-04-14:
I'm having the same problem.
I can connect to the bleno service from iOS perfectly, but Android can't.
I also have a nrf51 running custom firmware looking for the ble service and it doesn't connect to it either.

sandeepmistry, 2016-04-18:
@mfferreira please try updating the kernel as @vdharmon mentioned in https://github.com/sandeepmistry/bleno/issues/175#issuecomment-208965478.

Also, since your Android device can't connect to the nRF51, seems like something unrelated to bleno and more device specific.

I'm closing this for now, as the original issue seems kernel related.

closed 2016-04-18 by sandeepmistry

mfferreira, 2016-04-18:
@sandeepmistry I meant to say that both the nRF51 and Android are looking for the BLE service advertised by bleno.

sandeepmistry, 2016-04-24:
@mfferreira not sure if I'm misinterpreting the btmon gist (I usually use hcidump), but it looks like a non-LE connection attempt is being made. bleno only supports LE connections.